emsesp/EMS-ESP32 #1: (ESP32) Watchdog 1-WIRE DS18b20

sq4bja, 2021-03-14:
Hi,
I use several ds18b20 thermometers to read boiler temperatures. Sometimes there are problems with the operation of the 1-wire bus. To restore proper operation it is necessary to reset the esp32 module. Is it possible for ems-esp to expose any pin to high state to power dc 3.3v 1-wire sensors? In the absence of a temperature update (watchdog), the 3.3V power supply would be disconnected for 10 seconds to reset the 1-wire bus. This would greatly improve the stability of the system.

Tom

proddy, 2021-03-23:
@sq4bja @MichaelDvP is there stuff we need to do here?

MichaelDvP, 2021-03-23:
There wasn't any feedback to my version with gpio27 switched, i removed it a while ago. The bus_reset()-check is in normal code and will count dallas-errors up.

closed 2021-04-01 by proddy