emsesp/EMS-ESP32 #28: WLAN AP still active even if turned off

FauthD, 2021-03-16:
Hi Proddy, 
just found this in my router (Fritz!Box). The ems-esp is still in AP mode (at least it is listed, even after refreshing the list).

My first idea was that this is an artefact of rebooting the ems-esp32. But after refreshing the list it shows up again.
I did not expect that it is even possible to use the AP mode and work as a client at same time.
AP settings is "never"  and UI says "inactive".

Any ideas?

[image]

[image]

[image]

Dieter Fauth

proddy, 2021-03-16:
oh, ok. Let me check here too. Thanks for reporting!

proddy, 2021-03-17:
I can't reproduce. Could be that the router keeps hold of it. Can you join the AP? Or ping or telnet too 192.168.4.1?

FauthD, 2021-03-17:
Hi,
Could well be a bug  in my frotzbox 7590, v7.25, also seen with 7.21.
With other FB and Android devices I cannot find the AP.
Strange is that the FB shows the MAC of the AP (client mac +1).

Perhaps it apears for a short time after reboot.
Then it should be disapear on reboot of the FB, but it sticks.

Dieter

tp1de, 2021-03-18:
I can't confirm. I am using FB7590 v7.25 as well.
But keep in mind that the mesh network overview in the FB takes some hours to find inactive devices.

proddy, 2021-03-18:
Ok, can I close this and re-open if it happens again?

FauthD, 2021-03-19:
Yes, please close.
So far I could not find the undesired AP with other tools.
Most likely some glitch in the FB.
Dieter :-)