emsesp/EMS-ESP32 #31: expose individual commands via MQTT topics

proddy, 2021-03-21:
As a request from @glitter-ball , since all EMS-ESP commands need to follow the JSON format `{cmd:"", data:"", id:""}` it's hard to send individual commands, like from a component or trigger in HA. For example, when using HA's `platform:mqtt` in a switch component it can only send on/off payloads to a single command topic, like in this screenshot:

[image]

This feature is to expose all commands as individual MQTT topics by subscribing to them as for example:

 [21] (Sub) topic=ems-esp/boiler/comfort
 [22] (Sub) topic=ems-esp/boiler/wwactivated
 [23] (Sub) topic=ems-esp/boiler/wwtapactivated
 [24] (Sub) topic=ems-esp/boiler/wwflowtempoffset
 [25] (Sub) topic=ems-esp/boiler/wwmaxpower
 [26] (Sub) topic=ems-esp/boiler/wwonetime
 [27] (Sub) topic=ems-esp/boiler/wwcircpump
 [28] (Sub) topic=ems-esp/boiler/wwcirculation
 [29] (Sub) topic=ems-esp/boiler/wwcircmode
 [30] (Sub) topic=ems-esp/boiler/flowtemp
 [31] (Sub) topic=ems-esp/boiler/wwsettemp
 [32] (Sub) topic=ems-esp/boiler/heatingactivated
 [33] (Sub) topic=ems-esp/boiler/heatingtemp
 [34] (Sub) topic=ems-esp/boiler/burnmaxpower
 [35] (Sub) topic=ems-esp/boiler/burnminpower
 [36] (Sub) topic=ems-esp/boiler/boilhyston
 [37] (Sub) topic=ems-esp/boiler/boilhystoff
 [38] (Sub) topic=ems-esp/boiler/burnperiod
 [39] (Sub) topic=ems-esp/boiler/pumpdelay
 [40] (Sub) topic=ems-esp/boiler/maintenance
 [41] (Sub) topic=ems-esp/boiler/pumpmodmax
 [42] (Sub) topic=ems-esp/boiler/pumpmodmin

proddy, 2021-03-21:
did a first rough implementation. It will create a lot of additional MQTT subscribes, but I think the ESP32 can handle it.

glitter-ball, 2021-03-21:
OK. Should I look to implement it a better way in HA? Maybe scripting was a better option? Shout if/when you want me to test. Thanks!

proddy, 2021-03-21:
haven't tested with HA but should work with latest dev build.

MichaelDvP, 2021-03-21:
I think for thermostat you should enhance with the tag, now `thermostat/daytemp` works only on default hc, better use `thermostat/hc2/daytemp`

proddy, 2021-03-21:
> I think for thermostat you should enhance with the tag, now `thermostat/daytemp` works only on default hc, better use `thermostat/hc2/daytemp`

yeah, spotted that too. thanks.

MichaelDvP, 2021-03-21:
The `system/pin` without id is also nearly useless (defaults to gpio0). Maybe skip this from direct subscribe?

proddy, 2021-03-21:
also true. Adding the tag is going to be tricky since for the thermostat this is done dynamically when a HC is detected. And these subscribes are created when a device is added. So I think I'll just limit it to a few functions, either
a) by adding a parameter to the register_mqtt_cmd() to choose whether or not to create a new subscribe topic
or b) just hard-code it for the Boiler only and forget the other devices like system, thermostat, solar etc...

what shall I do?

MichaelDvP, 2021-03-21:
I think thermostat is the most used device to set a value. Imho stay with default hc is better than skip thermostat.
Option c: always add all 4 hc to thermostat.
The other devices solar, mixer have no commands for now. System commands are not used frequently used over mqtt, i think that can be skipt.

proddy, 2021-03-22:
ok skipping 'system' and thermostat will use the default hc. Adding all the hc's 1-4 can create up to 14*4 additional subscribes.

MichaelDvP, 2021-03-22:
> Adding all the hc's 1-4 can create up to 14*4 additional subscribes.

For my RC35 it's 4x22 subscribes for the hc. I tested this and heap (free/max.alloc) drops from 96.968 / 84.384 bytes without subscribes to 81.968 / 55.140 bytes with boiler, thermostat  and all 4 hc. No issues so far, the esp32 can handle it, but it's a lot of memory allocated.

 In [#739](https://github.com/emsesp/EMS-ESP/issues/739) there is someone with 3 active hc now switching to esp32. I don't know if he needs the direct subscriptions.

Maybe add a setting to mqtt-settings to activate the subscriptions?

MichaelDvP, 2021-03-22:
The high memory usage is due to register all subscriptions also as mqtt-subfunctions. I skip this, only use `queue_subscribe_message(topic);` and strip the topic in `on_message` to the known function. With all hc subscriptions heap is now: 93.572 / 67.728 bytes.

proddy, 2021-03-22:
ok! Let's go with that

MichaelDvP, 2021-03-22:
Ok, i need to clean up my test codes a bit, than i can push it. I've also added a mqtt setting below boolean format as `subscribe format` with selection `general device topics`, `individual topics, main heating circuit`, `individual topics, all heating circuits`. Rename it if you find a better naming.

MichaelDvP, 2021-03-22:
Ok, pushed, please check. With thermostats like RC20 with only a single heatingcircuit the setting does not have a effect, there will not /hc1/ tags added (can be changed by flagging the commands in thermostat).

proddy, 2021-03-22:
ok, I'll check. I also changed a few web files so it'll probably merge clash somewhere

proddy, 2021-04-06:
tested and works. nice.

closed 2021-04-06 by proddy