emsesp/EMS-ESP32 #32: For info: replace Wemos D1 ESP866 with ESP32 on BBQKees board v0.9

balk77, 2021-03-21:
I wanted to share my experience with upgrading to an ESP32 board

**Old situation**
Wemos D1 ESP8266 on a BBQKees v0.9 board

**Desired situation**
Nodemcu ESP32 on a BBQKees v0.9 board

**Steps**

1. Flash EMS-ESP32 to ESP32 board and configure WiFi
2. Power off and disconnect the ESP8266 from the EMS board
3. Set jumper J4 from 100E to 4k7
4. Connect GND, power and RXTX (on EMS board) to any set of pins on the ESP32. I have chosen 22 and 23. Make sure to connect RX on the EMS board to TX on the ESP32.

I took a while before I realized that step 3 was required. The ESP was in a bootloop. Just wanted to share this for the greater benefit.

[image]

bbqkees, 2021-03-23:
Strange it goes into a boot loop with 100Ohm. Most ESP32 GPIO can source absolute max 40mA per pin, and 100Ohm@3,3V is 33mA. But 4k7 is the better value anyway.

proddy, 2021-03-28:
I'll add this to the documentation