emsesp/EMS-ESP32 #33: Junkers FW200 wrong heating circuit mode reported

FredericMa, 2021-03-21:
**Bug description**
Heating circuit mode and modetype of Junkers FW200 are reported incorrectly.

**Steps to reproduce**
The problem showed up after upgrading from 2.1.0 ESP8266 to 3.0.0 ESP32.

**Expected behavior**
The example provided in the screenshots show the toggling from manual-nofrost to manual-heat for HC2.
EMS-ESP however shows auto-eco while it should be manual-nofrost. I changed to heating mode to manual-heat and EMS-ESP only shows auto. It looks like the modetype has an invalid state and is not shown in the console anymore. It is also not reported in MQTT.

**Screenshots**
Should be MANUAL-NOFROST for HC2 (and actually also HC1):
[image]
Should be MANUAL-HEAT for HC2:
[image]

**Device information**
```json
{
  "System": {
    "version": "3.0.0",
    "uptime": "000+01:32:46.780"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 6505,
    "#read requests sent": 1012,
    "#write requests sent": 59,
    "#incomplete telegrams": 0,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:23.04)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x26 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.14)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x20 ProductID:102, Version:20.08)",
      "handlers": "0x10C"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x21 ProductID:102, Version:20.08)",
      "handlers": "0x10C"
    },
    {
      "type": "Solar",
      "name": "Junkers ISM1 (DeviceID:0x30 ProductID:101, Version:23.04)",
      "handlers": "0x103 0x101"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:23.04)",
      "handlers": ""
    }
  ]
}
```

**Additional context**
This issue reminds me of a previous issue reported in a long thread here:
https://github.com/emsesp/EMS-ESP/issues/410#issuecomment-651672722
I guess it was fixed in this PR:
https://github.com/emsesp/EMS-ESP/pull/417
(or maybe this one):
https://github.com/emsesp/EMS-ESP/pull/419

proddy, 2021-03-21:
Hi @FredericMa , so it worked in 2.2.x on the ESP8266 but not on the ESP32 v3?

FredericMa, 2021-03-21:
That's correct, I would guess it is related to the software and no to the hardware since all other parameters are reported correctly as before.

FredericMa, 2021-03-22:
I can confirm that the fix of @MichaelDvP fixed the issue. Thanks!

closed 2021-03-22 by FredericMa

proddy, 2021-03-22:
thanks Michael, well spotted!

commit 70033017 (2021-03-22): fix #33, Junkers mode names sorted