emsesp/EMS-ESP32 #35: Align naming of values, MQTT keys and command names

proddy, 2021-03-23:
From https://github.com/emsesp/EMS-ESP32/issues/29

I propose:
- keep the parameter/value names consistent with the English translations from the EMS manuals
- keep the command names the same as the values (which are used in the MQTT JSON)
- limit the names to all lower-case

proddy, 2021-03-23:
Starting with Boiler, need to complete the table:

| command name | MQTT name | proposed new name |
| - | - | - |
comfort | wWComfort
wwactivated | wWActivated
wwtapactivated
wwflowtempoffset | wWFlowTempOffset
wwmaxpower | wWMaxPower
wwonetime | wWOneTime
wwcircpump | wWCircPump
wwcirculation  |  ?
wwcircmode | wWCircPumpMode
flowtemp | ?
wwsettemp | wWSetTemp
heatingactivated
heatingtemp
burnmaxpower
burnminpower
boilhyston
boilhystoff
burnperiod
pumpdelay
maintenance
pumpmodmax
pumpmodmin

bbqkees, 2021-03-23:
How about a dot notation? Like ww.comfort and boiler.hyst.off (can imagine parsing issues though).

proddy, 2021-03-23:
Nice idea, if we don't add the device name. Reason being the `<cmd>` is used in the Console like `call boiler ww.temp 60` and also in the API `http://ems-esp/api?device=boiler&cmd=ww.temp&data=60&id=1`

proddy, 2021-03-23:
> Could be a problem with parsing json.
> Parsing `boiler_data.ww.temp` will search for `boiler_data={"ww":{"temp":60}}` instead of `boiler_data={"ww.temp":60}`

that's a very good point. didn't think of that.

MichaelDvP, 2021-03-24:
I've extracted commands and names-texts from actual v3-dev and put them in a csv (github does not accept csv, i renamed to txt, import to calc/excel for formatting). Maybe a good time to move them to local_EN or a new file. Then people can also make their own local_KM and name mqtt like km200 outputs ;-)
[names.txt](https://github.com/emsesp/EMS-ESP32/files/6196671/names.txt)
[commands.txt](https://github.com/emsesp/EMS-ESP32/files/6196673/commands.txt)

proddy, 2021-03-24:
nice @MichaelDvP . I agree, we should move these to a new file as we did with `device_library.h` with a structure like `{ device_type, key, full_name, uom }`. 

The key is a string that is also used as the MQTT key name. And all in lower case.

Then the function `register_device_value()` would be simpler because we can look up the values. So a line like

`register_device_value(TAG_BOILER_DATA_WW, &wWSelTemp_, DeviceValueType::UINT, nullptr, F("wWSelTemp"), F("selected temperature"), DeviceValueUOM::DEGREES)`

could become

`register_device_value(TAG_BOILER_DATA_WW, &wWSelTemp_, DeviceValueType::UINT, nullptr, F("wwseltemp")`

MichaelDvP, 2021-03-24:
I thought of a simple name-file like the `locale_EN.h` and only move the strings to it, but for mqtt-name and full-name.
like: 
```
register_device_value(TAG_BOILER_DATA_WW, &wWSelTemp_, DeviceValueType::UINT, nullptr, F_(wWSelTemp), F_(wWSelTemp_), DeviceValueUOM::DEGREES)
```
and the file contains both strings for mqtt-publish and web/telnet output, commands can also point to the same string `F_(wWSelTemP)` to make sure there isn't a mismatch.
```
MAKE_PSTR(wWSelTemp, "wwseltemp")
MAKE_PSTR(wWSelTemp_, "selected temperature")
```
than someone can change only the file and adapt mqtt and web output, like:
```
MAKE_PSTR(wWSelTemp, "dhw_setpoint")
MAKE_PSTR(wWSelTemp_, "temperature of DHW setpoint")
```
or translations. (the `warm water` tag name should also go there).

If you use the mqtt-name as key, you can't change it in a single file.

proddy, 2021-03-24:
ok, that's cleaner and less error prone. And one day I'll get back to finishing the translation code.

tp1de, 2021-03-26:
It's good to hear that you work on these tables. I prepared for the moment my own ones for ioBroker integration.

There are two more points to consider: 

1. mqtt returns **values** for the field values. 
E.g.: for thermostat_data building: 1,2 or 3 with the following definitions: 1:light,2:medium,3:heavy. 
I started to use a seperate column for these definitions to read it for ioBroker.
2. commands need to define the allowed value range  -> minimum and maximum allowed values. E.g. building 1-3.

tp1de, 2021-03-26:
I use for testing the following "translation table" with units, max and min values and possible states. MAybe thats a help for you as well.
In orange I marked in excel the fields where I do not understand the content / values yet.
[ems1.xlsx](https://github.com/emsesp/EMS-ESP32/files/6212825/ems1.xlsx)

MichaelDvP, 2021-03-28:
@proddy I've tested a bit and think it's better to have both strings in one structure. Without need to refactor i test:
local_EN.h:
`MAKE_PSTR_LIST(selFlowTemp, F("selflowtemp"), F("selected flow temperature"))`

boiler.cpp:
```
register_device_value(TAG_BOILER_DATA, &selFlowTemp_, DeviceValueType::UINT, nullptr, FL_(selFlowTemp), DeviceValueUOM::DEGREES);
```
and for commands
```
register_mqtt_cmd(FL_(selFlowTemp)[1], MAKE_CF_CB(set_flow_temp));
```
with emsdevices.cpp addition:
```
void EMSdevice::register_device_value(uint8_t tag, void * value_p, uint8_t type, const __FlashStringHelper * const * options, const __FlashStringHelper * const * name, uint8_t uom) {
    register_device_value(tag, value_p, type, options, name[1], name[2], uom);
}
```
What do you think about this?

proddy, 2021-03-28:
looks fine. Perhaps create some macros for name[1] & [2] like SHORT_NAME/FULL_NAME to make the code easier to read.

I'm going to make some hopefully final changes to the Ethernet code today so if you're pushing today either make a PR or wait until after midnight!

MichaelDvP, 2021-03-28:
For now i have only copied all strings to local_EN and testes with a few boiler values. To replace all `register_device_value` is a lot of manual work. It takes some more time.

proddy, 2021-04-06:
When I was implementing https://github.com/emsesp/EMS-ESP32/issues/18#issuecomment-814271194 with the Web Commands, I noticed the difference between `temp` and `seltemp` on a Thermostat as commands, so implemented also seltemp as a new command action. I'd prefer to just remove `temp` but this will be a breaking change

tp1de, 2021-04-07:
For me it would be very important to have a mqqt datafield for the temporay temperature setting per hc. (auto, off, temperature). Valid until next time switchpoint - at least for rc300/310. And then mqtt datafield and command should be equal. Is there any problem to keep the temp command in parallel to avoid a breaking change?

proddy, 2021-04-21:
This has been implemented. The API changes will be covered in a different GH issue

closed 2021-04-21 by proddy