emsesp/EMS-ESP32 #37: EMS-ESP32 Ver. 3.0.0. - how to send MQTT commands

SpaceTeddy, 2021-03-24:
Hi guy's,

I updated to Ver. 3.0.0 and try to send mqtt commands to the gateway. Basically I saw, that there is a discussing about mqtt structure, but I tried two ways to enable the boiler with one time WW charging:
a) ems-esp/boiler/wwonetime 1
b) ems-esp/boiler {"cmd": "wwonetime" ,"data":"on", "hc":1}

may you privide me some examples to set boiler wwonetime and thermostat day/night temperatur for hc1?

thx,
Chris

proddy, 2021-03-24:
I just tested with the latest dev and those commands work (wwonetime).

MichaelDvP, 2021-03-24:
`wwonetime`was always a bit tricky. There are a lot of discussions how it works depending on boiler, boiler settings, temperature-levels, thermostat, etc. The syntax is ok, the needless hc-settings is ignored, data in both topics can be `"on"`, `1`, `"1"`, `"true"`, or `true`.

SpaceTeddy, 2021-03-25:
Hi all,
works with latest dev v3.0.1b3. 

you can close the issue