emsesp/EMS-ESP32 #38: Missing Homeassistant entities

mrfixit1, 2021-03-25:
I'm having trouble restoring the shower timer and last ems data update info in HA after upgrading from esp8266 v194 to esp32 v3. Anyone have a suggestion where to look?

Config: bbqkees v0.8 interface, esp32, usb power, shower timer in ems UI enabled.

The ems-esp upgrade was very simple, thanks to the excellent readme and web UI: built and uploaded the firmware using platformio, configured the wifi and mqtt in AP mode, swapped the boards and updated a few entities in HA's ui-lovelace.yaml to their current names. The only items that confused me for a few minutes were not enabling Nested format together with HA MQTT Discovery, and disabling Invert/Hide LED to turn the blue LED off. 

All of my dashboard entries in HA are present and correct, except that the last ems data update date+time and last shower date+time are stuck on the date+time of the last core restart, while the shower duration stays blank.

I spent time going through the states list but can't find what I need. I deleted all 3 mqtt devices as recommended in esp8266 issue #643, which cleared a couple of double entries in the entity list but didn't fix my issue. Changing the lovelace mode in HA configuration from yaml to storage mode makes a big mess of my dashboard and I'm not keen on starting again from a fresh HA installation and reconfiguring the ems and other devices from scratch. 

What have I missed?

proddy, 2021-03-25:
Nice to hear the upgrade to ESP32 went smooth, even on an old 0.8 gateway! On which ESM-ESP firmware version are you on?

Is the MQTT with the shower_data correct? This is what HA is using. If it is, then the problem is likely with a script. I'll need to check when I'm home to see what I have.

mrfixit1, 2021-03-25:
v3.0.0

I'm seeing the expected shower data in HA's MQTT listener - see screenshot.
[image]

proddy, 2021-03-25:
Those last updated scripts were from v2 (ESP8266) and will not work with v3. You can basically delete all those ems-esp entries in the yaml files. Use Integrations->MQTT to add the Devices to your lovelace UI. The only thing you might want to add to a yaml file are custom names or icons.

Then for the shower I only have this:

I have this
sensors.yaml
```yaml

- platform: template
  sensors:
    showertime_time:
      friendly_name: 'Last shower at'
      value_template: '{{ as_timestamp(states.sensor.last_shower_duration.last_updated) | int | timestamp_custom("%H:%M on %a %-d %b") }}'

- platform: mqtt
  name: 'Last shower duration'
  state_topic: "ems-esp/shower_data"
  value_template: "{{ value_json.duration | is_defined }}"
```

mrfixit1, 2021-03-27:
Thanks for your quick answer and tips.

After removing all of the ems-esp entries from the .yaml files, I spent a couple of hours down a rabbit hole trying to restore the dashboard with lovelace in storage mode, but gave up as I couldn't find a way to get the thermostat card back.

I then tried v2 on an esp8266, which had the same shower timer issue, but worked fine otherwise. This suggests to me that the breaking change was somewhere in v1.9.5+ and MQTT discovery, not in the switch from v2 to v3.

When time permits I'll try a fresh installation on a test VM (don't have a test pi yet).

mrfixit1, 2021-03-31:
Tried a fresh HA installation on a Win10 test VM using MQTT discovery. 

The thermostat card was still empty and the shower data were still missing with ems-esp v3.0.0, but both reappeared after updating the ems-esp to v3.0.1.