emsesp/EMS-ESP32 #41: run web interface standalone with a mock API server

proddy, 2021-04-01:
When developing and testing the web interface, it's handy not to bother with reflashing an ESP32 each time. The idea is to mimic the ESP using a mock/stub server that responds to HTTP POST and GET requests. 

To run this, d
```sh
% cd interface
% npm run dev
```

The hardcoded data is limited at the moment and can be easily extended by changing the file `/mock-api/server.js`

bbqkees, 2021-04-02:
I tried it and it looks great.

One thing: I got an error during the build; had to install the express module, and then it worked fine.

To install Express:
`npm install express`

proddy, 2021-04-03:
fixed that and updated the doc. https://github.com/emsesp/EMS-ESP32/blob/ft_mockapi/mock-api/README.md

proddy, 2021-04-05:
merged into dev

closed 2021-04-05 by proddy