emsesp/EMS-ESP32 #43: Please return command flowtemp

levran, 2021-04-09:
Please return command flowtemp for boiler in branch v3.0.2b1.

I use this command for turnoff heating for Junkers.

proddy, 2021-04-10:
@levran are you building the firmware yourself? If so take the latest from branch `ft_webcallcmd` and the command is `selflowtemp`

levran, 2021-04-10:
Unfortunately I am having difficulty compiling firmware. 
It is much easier for me to use the already compiled firmware. 
Tell me, what is the reason for removing the flowtemp command from the 3 version?

SpaceTeddy, 2021-04-11:
Hi Vlad,
I compiled the latest version of this branch for you. 
It compiles with SUCCESS, but npm gave me some errors. Hope that has no influences.
```
erbose mode can be enabled via `-v, --verbose` option
Building interface with npm
npm install
npm WARN @babel/plugin-bugfix-v8-spread-parameters-in-optional-chaining@7.13.12 requires a peer of @babel/core@^7.13.0 but none is installed. You must install peer dependencies yourself.
npm WARN create-react-context@0.3.0 requires a peer of react@^0.14.0 || ^15.0.0 || ^16.0.0 but none is installed. You must install peer dependencies yourself.
```

[firmware.bin.zip](https://github.com/emsesp/EMS-ESP32/files/6291339/firmware.bin.zip)

proddy, 2021-04-11:
@levran the flowtemp command was just renamed. We wanted to align the naming of the commands to the MQTT values. From https://github.com/emsesp/EMS-ESP32/issues/35

levran, 2021-04-11:
Many thanks!!!
Today I do not have time to try it, but tomorrow night is a must!
The problem is that the heatingactivated command is not working on my junkers. And I have to use flowtemp with a temperature of 0.

levran, 2021-04-11:
It looks like everything is working! :)
Thank you so much!
I will figure out what's new.

levran, 2021-04-12:
All working fine!

closed 2021-04-12 by levran