emsesp/EMS-ESP32 #45: Extension of boiler data with current power and temperatures

Archidus, 2021-04-17:
Attached is an Excel file showing the information menu structure off an HPC400 and the values at that time and mapping to the values from telegrams 0x48d and 0x48f. The data is for a boiler identified as "Enviline/Compress 6000AW/Hybrid 7000iAW".

In 0x48d is the current compressor power
0x48f shows the temperatures of the outdoor unit.

[image]
[image]
[image]
[image]
[EMS data.xlsx](https://github.com/proddy/EMS-ESP/files/6095094/EMS.data.xlsx)

proddy, 2021-04-25:
@MichaelDvP 12 more boiler params, from a heatpump. I think we can fit this into the boiler MQTT payload? Any thoughts of how to add these since you're busy with this part of the code now. I'm tinkering with SSL tokens and the new API still.

MichaelDvP, 2021-04-25:
Ok, i'll add them, i also think it fits into boiler_data. Also it is heatpump specific,  if boiler_data gets to large we can add a section boiler_data_hp.

MichaelDvP, 2021-04-25:
Ok, i've pushed the values. I'm not sure about the positions, most of the temperatures are on even positions, but 3 with offset 24 not. 
@Archidus Please check if the values are correct and shown in mqtt and web. For the mqtt-names i used `hp` with the given temperature shortcuts. `hptc0`, etc.

Archidus, 2021-04-26:
PL1, PH1 and TR7 are at the wrong position. 

The following positions should fit:

PL1 | Niedrigdruckfühler - div10 - in C
offset 24, highbyte at position 2 of the data block, lowbyte 3 at position of the data block

PH1 | Hochdruckfühler - div10 - in C
offset 24, highbyte at position 4 of the data block, lowbyte 5 at position of the data block

TR7 | Temperaturfühler Kältemittel (Gas) - div10 - in C
offset 24, highbyte at position 6 of the data block, lowbyte 7 at position of the data block

proddy, 2021-07-23:
@MichaelDvP did you make these last changes so it can go into v3.2 ?

MichaelDvP, 2021-07-23:
I think this was [this commit.](https://github.com/emsesp/EMS-ESP32/commit/de9ff6a3a1eb311ea612c83094c8bec60cdc9c0e)

proddy, 2021-07-24:
ok closing then. we can always re-open

closed 2021-07-29 by proddy