emsesp/EMS-ESP32 #47: The DS18B20 sensor stopped working after upgrade to v3.0.1

milleroff, 2021-04-20:
Hi,

I have a Gateway S32 (Standard Edition) from @bbqkees
It came preinstalled with a v3.x.x beta firmware (sorry, i don't remember the specific version).
After i upgraded the firmware to v3.0.1 the DS18B20 sensor is not detected anymore.

I get the following message:

> no external temperature sensors were detected

It was working well on the preinstalled firmware.

**Device information**
```
{
  "System": {
    "version": "3.0.1",
    "uptime": "001+13:29:35.280"
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 49778,
    "#read requests sent": 11264,
    "#write requests sent": 1,
    "#incomplete telegrams": 15,
    "#tx fails": 3,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Topline/GB162 (DeviceID:0x08 ProductID:115, Version:03.06)",
      "handlers": "0x10 0x11 0x14 0x15 0x16 0x18 0x19 0x1A 0x1C 0x26 0x2A 0x33 0x34 0x35 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)",
      "handlers": ""
    }
  ]
}
```

Thanks.

MichaelDvP, 2021-04-20:
Please check your settings for the new `board-profile` if it fit's your board.

milleroff, 2021-04-20:
It's set to the correct board.
But to be sure, i pressed save and rebooted the board. The sensor is still not detected.

[image]

milleroff, 2021-04-20:
@bbqkees can you please provide the link to the firmware that was preinstalled with the gateway ?
I could install it and check if the sensor works.

The order number is 6096.

MichaelDvP, 2021-04-21:
Try with board-profile `NodeMCU`. If this does not work, try `Custom` and set port for dallas as in older version (if you do not remember try out).

bbqkees, 2021-04-21:
I see there is a small mistake in the board profile:
`    if (board_profile == "S32") {
        data = {2, 3, 23, 5, 0}; // BBQKees Gateway S32`

https://github.com/emsesp/EMS-ESP32/blob/4841e42286f4ff1093aed27d54c31dd1faf55975/src/system.cpp#L1039-L1040

The S32 has the same pinout as the MH-ET and the NodeMCU-32S. So it should be 18.

MichaelDvP, 2021-04-21:
I'll fix it for the dev-build (and rename "MT-ET" to "MH-ET")

bbqkees, 2021-04-21:
Ok thanks Michael.

milleroff, 2021-04-21:
Indeed, changing the profile to "MT-ET Live D1 Mini" fixed the problem.
Thanks !

commit a633225a (2021-04-21): Fix #47 Gateway S32 board profile