emsesp/EMS-ESP32 #48: move system-commands to main context?

MichaelDvP, 2021-04-20:
ems-esp has the feature to open up to 3 telnet sessions, but closing/leave system-context removes the commands from all open sessions. It's useless to have more than one opened, because the other instances have no commands if one instance closes context or telnet.

@proddy I want to move all commands to main context (and console.cpp) and, once loaded, leave them loaded. Esp32 has enough memory to hold them. I have to remove double commands from system like `set` and `show`, replacing them by `show system`. 
For v2 i think it's better to stay with system context, but reduce possible telnet-sessions to one. 
What do you think? 

V3 looks like this:
```
ems-esp2:/# help
call [device] [cmd] [data] [id|hc]
exit
format
help
log [level]
passwd
read <device ID> <type ID> [offset]
restart
scan devices [deep]
set
set board_profile <name>
set bus_id <device ID>
set hostname <name>
set master thermostat <device ID>
set timeout <n>
set tx_mode <n>
set wifi password
set wifi ssid <name>
show
show commands
show devices
show ems
show mqtt
show system
show users
show values
su
watch [off | on | raw | unknown] [ID]
wifi reconnect
ems-esp2:/# show system
Uptime: 002+07:10:28
SDK version: v3.3.5-1-g85c43024c
CPU frequency: 160 MHz
Free heap:  191632 bytes
Free block: 113792 bytes

WiFi: Connected
SSID: HZTD67Fr45d
BSSID: 44:4E:6D:AF:6F:8F
RSSI: -69 dBm (62 %)
MAC address: 3C:61:05:06:6C:40
Hostname: ems-esp2
IPv4 address: 192.168.0.115/255.255.255.0
IPv4 gateway: 192.168.0.5
IPv4 nameserver: 192.168.0.5

Ethernet: disconnected

Syslog:
Host: 192.168.0.8
Port: 514
Log level: info
Mark interval: 0s

ems-esp2:/#
```

proddy, 2021-04-20:
yes! That's quite some work you did there. I've been wanting to do this for a while and actually started on it a few weeks back but never finished it. So all the contexts have gone and also the override functions that handle 'exit' and offloading? Great! Check it in!

MichaelDvP, 2021-04-21:
For now i've moved only all commands to main-context and to console.cpp file and leave context logic unused but not delete the functions. I you are sure not to use them in future we can clean up.

MichaelDvP, 2021-04-23:
Closing this, we can reopen if there are any complains.

closed 2021-04-23 by MichaelDvP