emsesp/EMS-ESP32 #50: Adopt the OpenAPI 3.0 standard for the REST API

proddy, 2021-04-29:
The current REST API is a bit sloppy and inconsistent. I'd like to use an industry standard for interfacing with EMS-ESP with these requirements in mind:
- it's secure. Using Access Tokens for commands and eventually HTTPS with self-signed certificates
- it's readable and easy to understand and adapt, with the specification publically available on Postman
- it could possibly be used to replace MQTT when interfacing with home automation systems, like Home Assistant

I choose OpenAPI, but also looked at RAML and JsonAPI.

proddy, 2021-04-29:
## EMS-ESP REST API OpenAPI Description

OpenAPI (formaly swagger) is an open standard specification for describing REST APIs. 

The EMS-ESP REST API will eventually be fully described in an OpenAPI 3.0 compliant document.

## What is OpenAPI?

From the [OpenAPI Specification](https://github.com/OAI/OpenAPI-Specification):

> The OpenAPI Specification (OAS) defines a standard, programming language-agnostic interface description for HTTP APIs, which allows both humans and computers to discover and understand the capabilities of a service without requiring access to source code, additional documentation, or inspection of network traffic. When properly defined via OpenAPI, a consumer can understand and interact with the remote service with a minimal amount of implementation logic. Similar to what interface descriptions have done for lower-level programming, the OpenAPI Specification removes guesswork in calling a service.

## Specification

- All data is sent and received as JSON.
- The URL is http://<hostname>/api. The default hostname is `ems-esp`. Note that it may also be ems-esp.local on some environments.
- HTTPS/SSL will be supported soon.
- To test use cURL from the command line, or `GET` or `POST` commands on Linux. `curl -i` will also show the header information.
- Read's are done via `HTTP GET` and are public. Any parameters not specified in the path can be passed as an HTTP query string parameter. e.g.

    ```bash
    $ curl -I http://ems-esp/api/boiler?name=wwtemp
    or
    $ curl -I http://ems-esp/api/boiler -d '{ "name": "wwtemp" }'
    ```

- Write's are done via `HTTP POST` and require authentication with the access token sent in the header or as a query parameter. Tokens, also known as JWT, can be created from the Security tab in the Web UI and are strings 152 characters in length. They do not expire (yet). e.g.

    ```bash
    $ curl -i -H "Authorization: Bearer {ACCESS_TOKEN}" -X POST http://ems-esp/api/boiler
    $ curl -i -H "Authorization: Bearer {ACCESS_TOKEN}" -d '{ "name": "wwtemp", "value":60}' http://ems-esp/api/boiler
    ```

## Error handling

- Requests that require authentication will return `403 Forbidden`
- Authenticating with invalid credentials will return `401 Unauthorized`

    ```html
    HTTP/1.1 401 Unauthorized
    {"message": "Bad credentials"}
    ```

- Sending invalid JSON will result in a `400 Bad Request` response.

    ```html
    HTTP/1.1 400 Bad Request
    {"message":"Problems parsing JSON"}
    ```

- Sending the wrong type of JSON values will result in a `400 Bad Request` response.

    ```html
    HTTP/1.1 400 Bad Request
    {"message":"Body should be a JSON object"}
    ```

- Sending invalid fields will result in a `422 Unprocessable Entity` response. `code` can be missing, missing_field, invalid or unprocessable. For example when selecting an invalid heating circuit number.

    ```html
    HTTP/1.1 422 Unprocessable Entity
    {
       "message": "Validation Failed",
       "errors": [
         {
           "field": "title",
           "code": "missing_field"
         }
       ]
    }
    ```

# Device Endpoints

## `<URL>/api/{device}`

`{device}` is `boiler` `thermostat` `mixer` `solar` `heatpump` `dallassensor`

`{name}` is the short name, e.g. **curflowtemp**

`GET /api/{device}` to get all device details and values (formally the info command). The full-names are used here. Example:

```json
example here
```

`GET /api/{device}/{name}` to return a specific value. Example:

```json
example here
```

`POST /api/{device}/{name}` to set a specific value, requires a JSON message (value is mandatory, id is optional and used to identify heating circuits)

```json
{
   "value": <value>,
   "id": <id>
}
```

example:

```json
{
  "name": "curflowtemp",
  "fullname": "current flow temperature",
  "value": 45.8,
  "type": "number",
  "min": 0,
  "max": 3200,
  "unit": "°C",
  "writeable": false
}
```

`POST /api/{device}` to set a specific value, requires a JSON message (name and value are 

mandatory, id is optional and used to identify heating circuits)

```bash
{
 "name": "<name>",
 "value": <value>,
 "id": <id>
}

example:

$ curl -i -H "Authorization: Bearer {ACCESS_TOKEN}" -d '{ "name": "wwtemp", "value":60}' http://ems-esp/api/boiler
```

# System Endpoints

## `<URL>/api/system`

`GET <URL>/api/system` returns all system information (previously the info command)

The following are all POST messages and require an access token from a user with Admin rights.

`POST <URL>/api/system/settings` returns the current system settings, passwords excluded

`POST <URL>/api/system/pin` turn a GPIO on or off, with data:

```json
{ "pin":<pin>, "value":<0, 1, on, off, true, false> }
```

`POST <URL>/api/system/send` sends a telegram with data:

```json
{ "telegram" : <string> }
```

`POST <URL>/api/system/publish` forces an MQTT publish, optionally to a specific device, with data:

```json
{ "name" : < <device> | "ha"> } 
```

`POST <URL>/api/system/fetch` forces a fetch of all EMS data from all attached devices

`POST <URL>/api/system/restart` forces a restart of the EMS-ESP

Note this can also be invoked using

```bash
$ curl -i -X POST http://ems-esp/rest/restart\?access_token\={ACCESS_TOKEN}
or
$ curl -i -H "Authorization: Bearer {ACCESS_TOKEN}" -X POST http://ems-esp.local/
rest/restart
```

proddy, 2021-04-29:
One thing I'm still pondering about is whether to include the heating circuit in the path, like `api/thermostat/hc1/....`

tp1de, 2021-04-29:
@proddy 
I just finalized the ioBroker adapter with the enhancements done in #46 .
Some part you mentioned above seems a bit to complex for me (e.g. access tokens -  especially if they expire)

The ioBroker adapter can now read & write data from km200 as well from ems-esp.
I still find the ems-esp access structure confusing and too complex and would recommend to make it similar to km200.
http get and post to the full adress structure - id like hc1 is included in url command - e.g. for http get:

`km200: http://{km200-ip}/heatingCircuits/hc1/temperatureLevels/comfort2`

`ems today: http//ems-esp//api?device=thermostat"&cmd="comforttemp&id=1` **id needs to be 1 not hc1 !!**
`ems-esp proposal: http://ems-esp/thermostat/hc1/comforttemp` **use hc1 within url** 

http returns the full nested json similar to what we have implemented in #46. (allowedStates/enum, units .... and actual value)
km200 http post just needs the new value in payload - that's what I would recommend for ems-esp as well.

km200 need a quite complex rijndael-128 ecb en-/decryption since it is connected to the Bosch cloud for app access.
For local lan access of the km200 this is needed as well and has to be implemented in any adapter to access the km200.

For ems-esp a simple username / password type of security should be enough since it is not connected to any public cloud.
This could be done e.g. in ioBroker while using the ioBroker cloud adapter while connecting to IFTT or Alexa.

MichaelDvP, 2021-04-30:
@proddy 
> One thing I'm still pondering about is whether to include the heating circuit in the path, like api/thermostat/hc1/....

Make it optional, a lot of people have ony one hc and then there is no need to add it.

How this is:
GET `http://emsesp.local/api/thermostat` 
return info.

GET `http://emsesp.local/api/thermostat/hc2/daytemp` 
GET `http://emsesp.local/api/thermostat/daytemp` 
returns the value-info, in second line for the default hc.

POST `http://emsesp.local/api/thermostat/hc2/daytemp`  {"value":21}
POST `http://emsesp.local/api/thermostat/daytemp`  {"value":21, "id":2}
POST `http://emsesp.local/api/thermostat/daytemp`  {"value":21}
are equivalent if there is only one hc (in this case hc2).

@tp1de 
> I just finalized the ioBroker adapter with the enhancements done in #46 .

Really nice, works good. Only dallassensors were missing. I added `Dallassensor` as device in System/info and the value-infos.
Now your adapter reads the sensors without changes. Very clever to scan the devices and get the info.

Should we add a `ioBroker`page to the wiki and link to your adapter?

I suggest a small page like for openHAB, Loxone, etc. [see here](https://emsesp.github.io/docs/#/openHAB): 
```
![logo](_media/logo/ioBroker-logo.png ':size=40%')
@tp1de has an adapter for EMS-ESP-32 to install by url which you can find [here](https://github.com/tp1de/iobroker.ems-esp).
You can also integrate EMS-ESP-8266 into ioBroker by using the ioBroker-mqtt-adapter.
```

BTW: in current API the commands check for a prefix that creates or overwrites the id. `&cmd=hc2.daytemp` is the same as `&cmd=daytemp&id=2`, `&cmd=hc2:daytemp&id=1` uses hc2. Separator between hc and command can be any single character (in mqtt it is always `/`) .

proddy, 2021-04-30:
thanks both for the feedback. I'll make the changes. @tp1de I really don't want to use username/password on the URL as it's not at all secure, hence the non-expiring Access Tokens. The token is only needed when using POST/PUT/PATCH to update values and can be either added to the HTML Header or as a query parameter in the URL. I'll provide some examples soon.

tp1de, 2021-04-30:
> I really don't want to use username/password on the URL as it's not at all secure, hence the non-expiring Access Tokens

@proddy understood but I assume that the ems-esp will be used only within local lan / wlan. My other Lan components just use htpps with username password as well.

Access tokens are fine for me as well, I just have quite some negative experience with Gardena/Husquana Access Token towards their Smartgarden Cloud Interface.

> Should we add a `ioBroker`page to the wiki and link to your adapter?

Yes - but give me a couple of more days to test under different Linux and Nodejs versions.
I develop and test in WIN10 with VSC and a local ioBroker Test Instance. 

My ioBroker production system runs as a multihostsystem with master in Docker on a Synology Diskstation with nodejs v12.22.1 and a slave with a Raspberry pi4 (4GB) with ioBroker under nodejs 14.16.1.
ioBroker States are maintained in Redis DB (Docker), mySQL DB, Influx DB & Grafana (all under Docker).
I want to test all variants and see if recordings sql statements will work with influxdb as well.

I realized that the iobroker test scripts in Github failed since I missed some try / catch statements within the http request when ems-esp or km200 adresses are not reachable in the test environment (I am not an IT guy.. and still have to learn a lot)
I already did some code adjustments, but is not 100% complete yet.

Now I have to learn how to tag Github for different versions and how to build the npm module for ioBroker.

By the way: The ioBroker adapter transforms the ems-esp state-tree to the km200 tree to secure that ems-esp and km200 datapoints are within same substructure. When having only hc1 the state-tree will be heatingCircuits.hc1.xxxx therefore thermostat.hc1 will become heatingCircuits.hc1. 
All ww... fields in ems-esp will be under dhwCuircuits.dhw1 for boiler or just  dhwCuircuits for the thermostat values. This is fixed for the moment but I will make it selectable to keep the old device structure.

tp1de, 2021-04-30:
@proddy 
just to add for security:
iobroker rest api:  https username password
homematic xlm api: https username password
mqtt server : https username password
Router: https username password

For security reasons ems-esp does not need access tokens. Maybe easier to implement or use?

proddy, 2021-04-30:
EMS-ESP doesn't use https yet (SSL) so all usernames/passwords are sent as encrypted plain text. Anyone sniffing the wifi can easily pick them up.

proddy, 2021-05-01:
It's not about whether using username or passwords to connect is ok or not, it's about whether they are exposed. In EMS-ESP the Telnet console and Web-UI are not sending username/passwords in plain text so it's safe. This is also the same as in your examples like the MQTT server, Router and homematic (I just read their API doc). The open API at the moment is dangerous - you can send bad commands and literally blow up your boiler/heatpump.

I will try and keep it backwards compatible but it's highly likely a few things will not work the same. When changing values you will need to supply the access token for example. I could make an option to leave it all open and make this configurable in the WebUI but rather not.

tp1de, 2021-05-01:
> I will try and keep it backwards compatible but it's highly likely a few things will not work the same. When changing values you will need to supply the access token for example. I could make an option to leave it all open and make this configurable in the WebUI but rather not.

I put the ioBroker adapter productive since some days and switched off the standard km200 adapter on ioBroker.
I am just a bit concerned that changes on the api might kill the actual adapter since I can't overlook the changes needed.
(I do not have a second ems-api board with esp32 for testing yet.)

Please tell in advance when this changes will be pushed to github and what needs to be changed on the api calls.

tp1de, 2021-05-01:
@MichaelDvP 
> Really nice, works good. Only dallassensors were missing. I added `Dallassensor` as device in System/info and the value-infos.
> Now your adapter reads the sensors without changes. Very clever to scan the devices and get the info.

Michael have you already tested on on an ioBroker installation?

MichaelDvP, 2021-05-01:
> Michael have you already tested on on an ioBroker installation?

Sure, running on raspberryPi4, rasbian lite 5.10.17-v7l, node14.16.1, npm6.14.12.

[image]

tp1de, 2021-05-01:
@MichaelDvP 
I am happy to see that the adapter is already working  not just for me ... this was the first ioBroker Adapter I made :)
Have you tested state changes and is anything not working?

Wit km200 and emsp-esp in parallel it looks like this:
[image]

MichaelDvP, 2021-05-01:
> Have you tested state changes and is anything not working?

For now i've only checked a few values, boiler wwseltemp and thermostat hc2.seltemp Both are ok and changeable.
Only small thing i noticed is in "Info->Meine Adapter"  a empty field for this adapter. 
I'm also running mqtt in parallel without issues so far.
If i find an issue i'll tell it on [your repo](https://github.com/tp1de/iobroker.ems-esp/issues), ok?

tp1de, 2021-05-01:
> If i find an issue i'll tell it on [your repo](https://github.com/tp1de/iobroker.ems-esp/issues), ok?

Yes fine for me, I just do not really understand yet how to use github with versioning. 
I use VSC do do changes and test on WIN10 and then upload changes to github. 

But I have not understood how to tag changes for github. 
Do I need to register a token in npm and use npm? I understood for ioBroker I have to do this.
Michael maybe you can help ... ?

tp1de, 2021-05-01:
> Only small thing i noticed is in "Info->Meine Adapter" a empty field for this adapter.

Yes, but 70% of my adapters installed are not listed there as well.
The adapter does not show a short description on the adapter overview - just the graphic symbol.
May this comes with npm registering and adding to the iobroker repository ? 
I am still looking for a short description how to do this part ...

proddy, 2021-05-01:
in the meantime while you're having fun with ioBroker I've updated the API spec (https://github.com/emsesp/EMS-ESP32/blob/ft_https/doc/EMS-ESP32%20API.md) and will include the old method so it's backwards compatible with how it works now.

proddy, 2021-05-02:
checked in first implementation. Will still need to test all scenarios before merging into dev. The doc is up to date.

proddy, 2021-05-04:
merged into dev 3.1.1

tp1de, 2021-05-04:
@proddy 
I can't upload new dev 3.1.1 on the webui to test. Where to report issues on REST API 3.0? (issue #)

proddy, 2021-05-04:
You can use this issue to report back. I'll test here.

tp1de, 2021-05-04:
@proddy Findings for V3 REST API in v3.1.1 b0

The v2 style commands for detailled info doesn't work with v3.1.1 anymore:
e.g.  `http://ems-esp/api?device=boiler&cmd=wwworkm`

The new style commands for devices return long names. e.g. `http://ems-esp/api/boiler` 
How do I get the short names with v3?

V3: http://ems-esp/api/boiler/{name} doesn't work as well I tried with short and long name (I need to to use short names !)

So how do I get default info's per datafield in V3? V2 style doesn't work anymore as well.

proddy, 2021-05-04:
what do you mean "doesn't work". nothing comes back? What are you using to test. Via a browser it'll use GET so you'll need to pass the access_token or turn off the security in the settings. If testing on the command line with `curl` use the `-X` parameter. It's all documented in the doc.

tp1de, 2021-05-04:
I test from javascript. Access token switched off in settings. http get requests.
For devices I get the json with long names and values. How to get short names?

http get request with device and name returns invalid command. Anything special needed in the header? Please provide some examples. V2 command structure works for devices hut not for fields.

MichaelDvP, 2021-05-05:
@tp1de I think i've  fixed the old api, it worked only for commands, noot for values. 
I havn't tested the new api yet, i first want to understand the code and how it works. A quick check with normal browser-get redirects me to main page.

proddy, 2021-05-05:
I'll take a look too. I may have inadvertently missed something when adding the horrible v2 backwards compatibility. Please try and keep the spec and not introduce any 'hacks' like using id=0 for shortnames. If Thomas wants short names we'll create a new endpoint as discussed before

tp1de, 2021-05-05:
I can adjust the adapter to new API if short names and field info are available. Can't you return one json with short name, long name and value for devices? And for state initialization I need the field info with enum, units etc like in V2. If backward compatibility to V2 is too difficult then skip it.

proddy, 2021-05-05:
The new API behaves 100% the same as the old one, only with the addition of security and path names. It's just a silly bug I probably introduced without thinking. I'll take a look later...very busy with work these days

MichaelDvP, 2021-05-05:
> Please try and keep the spec 

I've only removed the command filter to allow the value-info again. This filter broke the spec, not my mod!

> and not introduce any 'hacks' like using id=0 for shortnames.

Using the id>0 for single hc and 0 for all was introduced a long time ago in common sense (search v2 issues). It was not a hack. Yes, you removed the id and swich to long names with refactoring. I thought id was only forgotten and add it back, sorry if it was intended to remove it.

tp1de, 2021-05-05:
I loaded last FW from this morning and recognized for http GET:

V2: cmd=id0 works for devices --> returns short names | cmd= {field} --> returns `{ "message": "Problems parsing elements" } `

V3: http://ems-esp/api/{device}/hc returns short names (similar to id=0) - but "hc" looks strange when used with boiler 
V3: http://ems-esp/api/thermostat/wwcirmode" --> returns `{ "message": "Problems parsing elements" }`

BTW: I will be out of my home from tomorrow onwards for a couple of days - so I can't test anymore

proddy, 2021-05-05:
This is what I feel needs changing

- `http://ems-esp/api/{device}` shows short names by default. Just like https://api.github.com/orgs/emsesp/repos 
- `http://ems-esp/api/{device}/hc` or `http://ems-esp/api/{device}/wwc` should only work for Thermostat or Mixing
- there is no `hc0`. That should give an error

The later to add
- another endpoint to give back a list of available shortnames for each EMS device. Perhaps `http://ems-esp/api/{device}/catalog` ? Would not work with System.
- modify the internal Command structure so we get rid of the mandatory `data` and `hc/id`. Each command can define its own json body. For example when changing a GPIO pin the arguments should be `gpio` and `value` instead of `data` and `id` now. I'm taking inspiration again from GitHub https://docs.github.com/en/rest/reference/repos#branches

the `V3: http://ems-esp/api/thermostat/wwcirmode" --> returns { "message": "Problems parsing elements" }` is a bug, not sure why but either me or Michael will look into it.

MichaelDvP, 2021-05-05:
>    http://ems-esp/api/{device} shows short names by default. 

That's easy, only set last parameter false in EMSESP::command_info  `generate_values_json` 
What do you mean with "by default"? Should there be a non-default case? Or a seperate command for long names?
e.g `info` for shortnames, `information` gives long names?

>   http://ems-esp/api/{device}/hc or http://ems-esp/api/{device}/wwc should only work for Thermostat or Mixing
>   there is no hc0. That should give an error

Same routine, two lines to add when checking the TAG. The id is not needed if we always use the prefix. i.g.
`http://ems-esp/api/thermostat/hc2/mode` or `http://ems-esp/api?device=thermostat&cmd=hc2/mode`
That already working.

I'll change the dev to shortname and valid hcs.

> the V3: http://ems-esp/api/thermostat/wwcirmode" --> returns { "message": "Problems parsing elements" } is a bug, not sure why but either me or Michael will look into it.

I think the bug is the missing `c` in `wwcirmode`.

tp1de, 2021-05-05:
> I think the bug is the missing `c` in `wwcirmode`.

Michael you are right ...... sorry.

> `http://ems-esp/api/thermostat/hc2/mode` or `http://ems-esp/api?device=thermostat&cmd=hc2/mode`
> That already working.

Yes I can confirm, but syntax for second version is slightly different then before with cmd=mode&id=2

> That's easy, only set last parameter false in EMSESP::command_info `generate_values_json`

What does it mean exactly. Has this to be set in the header? ....and how? Could you give an example?

MichaelDvP, 2021-05-05:
> Yes I can confirm, but syntax for second version is slightly different then before with cmd=mode&id=2

Sure, but both ( id or prefix) works in old api syntax.

> What does it mean exactly. 

That means i'll change line 996 in emsesp.cpp to 
`has_value |= emsdevice->generate_values_json(json, tag, (id == -1), false);`
And the info gives short names.

proddy, 2021-05-05:
seems to work nicely now - thanks for jumping in Michael and fixing it.

tp1de, 2021-05-05:
I can confirm V3 Get work now. V2 command structure is not 100% available but I will change my code.

How will changes now need to be postet:

- without token
- with token

Can you provide an example (header url payload)

MichaelDvP, 2021-05-05:
> V2 command structure is not 100% available but I will change my code.

What's missing? Anything else than the blocked id=0?

tp1de, 2021-05-05:
I think nobody uses this part with v3. I am on the way to change to V3. So skipp V2 compatibilty ...
What about http post? Is this code ok when url is: `ems-esp/api/{device}/{field} `?

```
async function ems_put(url,value) {return new Promise(function(resolve,reject) {
	request.post({headers: {"Content-Type": '"application/json"',"Bearer":ems_token},url: url, body: {"value":value}},
		function(error, response, body){if (error) {return reject(error);} resolve(response);});
});
```

proddy, 2021-05-05:
There's an example in the [API doc](https://github.com/emsesp/EMS-ESP32/blob/dev/doc/EMS-ESP32%20API.md). You're missing the Authorization key. So probably add `headers: { "Authorization": 'Bearer ' +ems_token }`

tp1de, 2021-05-05:
I seem to be too stupid to understand the documentation. ... The argument body doesn't work
I will stop for tonight. Since I am travelling next days it has to wait ...

proddy, 2021-05-06:
I wrote a quick python script to show how to include the header
```
# test_api.py
import requests
import json

url = "http://ems-esp.local/api/thermostat/temp"
token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWUsInZlcnNpb24iOiIzLjEuMWIwIn0.qeGT53Aom4rDYeIT1Pr4BSMdeWyf4_zN9ue2c51ZnM0"
body = json.dumps({ "value":22 })
headers = {  'Content-Type': 'application/json',  'Authorization': 'Bearer ' + token }

response = requests.post(url, headers=headers, data=body, verify=False)
print(response.json())
```

and running it:
```
% python3 test_api.py
{'message': 'OK'}
```

and in the EMS-ESP Telnet console
`000+00:24:23.101 I 18: [thermostat] Setting thermostat temperature to 22.0 for heating circuit 1, mode auto`

hope that helps.

proddy, 2021-05-07:
> http://ems-esp/api/{device} shows short names by default.
>> That's easy, only set last parameter false in EMSESP::command_info generate_values_json
What do you mean with "by default"? Should there be a non-default case? Or a seperate command for long names?
e.g info for shortnames, information gives long names?

@MichaelDvP  I was thinking of `/api/{device}` shows shortnames as default as it does now, and `/api/{device}/info` shows the verbose long names. This also matches `/api/system/info`.  What do you think?

tp1de, 2021-05-07:
@proddy Thanks for your request example with python. In javascript it looks like this and it is working now. 
I tested with two headers ( with / without token):

var request = require('request');
var token = "eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWUsInZlcnNpb24iOiIzLjEuMWIwIn0.hQDJaIIiq7nlagPkHk6cqwrDbXdLa8JDSaS1wMA3gxs";
var body =JSON.stringify({ "value":52 })
var url = "http://ems-esp/api/boiler/wwsettemp ";

var headers = {"Content-Type": "application/json"};
var headers_token = {'Content-Type': 'application/json',  'Authorization': 'Bearer ' + token };

var response = request.post({url, headers: headers_token, body});
console.log(response);

In setup access token was required. Value only changes when access token is send (headers_token), but I would expect an error message in response, when no token is send within header (header). That does not seem to be the case. Why?

**response without token but token required:**
` {'uri':{'protocol':'http:','slashes':true,'auth':null,'host':'ems-esp','port':80,'hostname':'ems-esp','hash':null,'search':null,'query':null,'pathname':'/api/boiler/wwsettemp','path':'/api/boiler/wwsettemp','href':'http://ems-esp/api/boiler/wwsettemp'},'method':'POST','headers':{'Content-Type':'application/json','host':'ems-esp','content-length':12}}
`
**response with token:**
`{'uri':{'protocol':'http:','slashes':true,'auth':null,'host':'ems-esp','port':80,'hostname':'ems-esp','hash':null,'search':null,'query':null,'pathname':'/api/boiler/wwsettemp','path':'/api/boiler/wwsettemp','href':'http://ems-esp/api/boiler/wwsettemp'},'method':'POST','headers':{'Content-Type':'application/json','Authorization':'Bearer eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VybmFtZSI6ImFkbWluIiwiYWRtaW4iOnRydWUsInZlcnNpb24iOiIzLjEuMWIwIn0.hQDJaIIiq7nlagPkHk6cqwrDbXdLa8JDSaS1wMA3gxs','host':'ems-esp','content-length':12}}`

proddy, 2021-05-08:
> In setup access token was required. Value only changes when access token is send (headers_token), but I would expect an error message in response, when no token is send within header (header). That does not seem to be the case. Why?

correct, when calling `/api//system/settings` and `/api/system/info` or `/api/system` without a token it should report an error. It used to, so that logic must have been moved with all the changes.

proddy, 2021-05-09:
I added a new command called `commands` which lists all the available commands, and also tidied up the `call` function in the Console.
```sh
% curl http://ems-esp.local/api/boiler/commands
{
  "boilhystoff": "hysteresis off temperature",
  "boilhyston": "hysteresis on temperature",
  "burnmaxpower": "burner max power",
  "burnminperiod": "burner min period",
  "burnminpower": "burner min power",
  "commands": "list all commands",
  "heatingactivated": "heating activated",
  "heatingtemp": "heating temperature",
  "info": "list all values",
  "maintenance": "maintenance scheduled",
  "pumpdelay": "pump delay",
  "pumpmodmax": "burner pump max power",
  "pumpmodmin": "burner pump min power",
  "reset": "reset messages",
  "selburnpow": "burner selected max power",
  "selflowtemp": "selected flow temperature",
  "wwactivated": "activated",
  "wwcirc": "circulation active",
  "wwcircmode": "circulation pump frequency",
  "wwcircpump": "circulation pump available",
  "wwcomfort": "comfort",
  "wwdisinfectiontemp": "disinfection temperature",
  "wwflowtempoffset": "flow temperature offset",
  "wwmaxpower": "max power",
  "wwonetime": "one time charging",
  "wwsettemp": "set temperature",
  "wwtapactivated": "turn on/off DHW by going into maintenance mode"
}
```

MichaelDvP, 2021-05-10:
> @MichaelDvP I was thinking of /api/{device} shows shortnames as default as it does now, and /api/{device}/info shows the verbose long names. This also matches /api/system/info and also consistent when calling the command from the Console. What do you think?

I'm not sure. I think long names are nice on web, for api and mqtt is see not real usecase. On console i would prefer short names, but understand if other like the long names more.
The new implementation with hidden and verbose flags set in many places are very complicated. Also it does not work for dallassensor and system. Also for thermostat/info for a given hc does not give short names, 

I'll push my solar changes now and make a PR with some small changes to a central command check, rejecting a id=0-input, but use it internally. And some other things i've seen. The long names should always check the WW-flag, and the nice command::show_all on the console should give a hint to hc-specific settings.

Ok, or should i wait with the solar and make a complete PR?

proddy, 2021-05-10:
Yes, it's getting crazy. 

This is what I really wanted to do:

- keep commands for actual commands and actions only, like `publish`, `send`. These I expect will mainly be system commands.
- do not make commands from device values (e.g. seltemp). It was a bit of a hack.
- there is no need for hidden commands anymore, so that can be removed
- commands can still be executed from the Console CLI use `call <cmd>` or via a POST URL `/ems-esp/api/command?cmd=<cmd>` or just `/ems-esp/api/command` with a JSON body.
- drop support for the `data=` and `id=` in the old API URL. Use only the newer API with path parameter, e.g. `/thermostat/hc1/seltemp` with GET to retrieve and POST to set a value(s).
- In the Console we can have `show` and `show verbose` to switch between short names and full names
- Remove the `info` and `info_short` command as they are no longer needed
- Fix the bug where `http://ems-esp.local/api/thermostat/hc1` doesn't work (should display short names)
- I like the idea of having 'hidden' device values, like hamode, hatemp, temp so they don't get displayed in the web, console or can be queried on in the API. Now we hack this with having an empty description that works but not clean code.
- No need to keep backwards compatibility. I'm not sure many people use it and @tp1de has already moved over to the new format.

**I think before we make any more changes we should discuss this, perhaps on a GitHub Discussion page?** 

The solar stuff you can check-in straight into the dev branch, but let's not make any more major changes to console or the API until we have an agreement on the specification.

closed 2021-05-15 by proddy