emsesp/EMS-ESP32 #51: Missing Data from Solar Module

mbenkler, 2021-05-03:
i have a SM10 Solar Module and EMS-ESP does not show all Data the Module has: 

in console i see this: 

Solar: SM10 (DeviceID:0x30, ProductID:73, Version:02.00)
  collector temperature (TS1): 42.7 °C
  tank bottom temperature (TS2): 18.5 °C
  pump modulation (PS1): 100 %
  pump (PS1): on
  valve status: on
  tank heated: on
  collector shutdown: on
  pump working time: 617 days 19 hours 31 minutes

but on the RC35 Display i can see 3 different temperatures and the energy of the module like this
Energy Last Hour
Energy Today
Energy Total

am i missing something ? 

i have installed the latest stable release 3.01

proddy, 2021-05-03:
The SM10, from our understanding, doesn't support energy. You can test using console if anything comes back from a `read 30 38E`. It it could be hidden in another telegram that we don't know about.

What are the 3 different temperatures called on the RC35 and do you recognize the values?

MichaelDvP, 2021-05-03:
Afair the 3rd temperature is the ww-temperature from boiler, the SM10 has only 2 sensors. There was a discussion [here in #77](https://github.com/emsesp/EMS-ESP/issues/77). In telegram 97 the first 2 bytes and the last 2 are unknown, maybe one of this is actual energy? In the #77 thread it was always zero, but the sun wasn't shining.

mbenkler, 2021-05-03:
nothing comes back after `read 30 38E`
sorry i did not see the other discussion. and yes the 3rd temperature seems to be the ww-temperature. 

can you tell me how i can get the information of telegram 97  `send 0B B0 02 00 03` like in the mentioned discussion doesn`t work on my esp ?

MichaelDvP, 2021-05-03:
The discussion was on a v1.5 version, the syntax is a bit different now, but `watch 30` is good.
Seems the first data-byte in 97 is a status bitfield, always 08 in your case, i think this bit 3 is collector shutdown, the collector temp is 127°C. Check again if the solar system is running and you have gain, then we check if the last two zeros change.

Or wait a few days, i have a standalone-solar with SC20 for now, but ordered yesterday a cheap, used SM10 on ebay.  In some days i can also check the data from/to the module.

mbenkler, 2021-05-03:
> 00+04:31:04.948 N 322: [telegram] Rx: 30 0B 97 00 08 00 04 9E 00 00 DB 00 0D 93 D0 00 00 6B
000+04:31:09.629 N 323: [telegram] Rx: 30 00 97 00 08 00 04 9E 00 00 DB 00 0D 93 D0 00 00 66
000+04:31:09.849 N 324: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:32:04.697 N 325: [telegram] Rx: 30 0B 97 00 08 00 04 A9 00 00 DB 00 0D 93 D0 00 00 E8
000+04:32:09.809 N 326: [telegram] Rx: 30 00 97 00 08 00 04 AA 00 00 DB 00 0D 93 D0 00 00 B3
000+04:32:10.035 N 327: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:33:05.070 N 328: [telegram] Rx: 30 0B 97 00 08 00 04 B1 00 00 DB 00 0D 93 D0 00 00 6A
000+04:33:09.995 N 329: [telegram] Rx: 30 00 97 00 08 00 04 B3 00 00 DB 00 0D 93 D0 00 00 03
000+04:33:10.220 N 330: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:34:04.694 N 331: [telegram] Rx: 30 0B 97 00 08 00 04 AC 00 00 DB 00 0D 93 D0 00 00 12
000+04:34:10.118 N 332: [telegram] Rx: 30 00 97 00 08 00 04 AB 00 00 DB 00 0D 93 D0 00 00 81
000+04:34:10.343 N 333: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:35:04.942 N 334: [telegram] Rx: 30 0B 97 00 08 00 04 96 00 00 DB 00 0D 93 D0 00 00 E2
000+04:35:10.241 N 335: [telegram] Rx: 30 00 97 00 08 00 04 93 00 00 DB 00 0D 93 D0 00 00 15
000+04:35:10.467 N 336: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:36:04.691 N 337: [telegram] Rx: 30 0B 97 00 00 00 04 71 64 00 DB 03 0D 93 D0 00 00 2B
000+04:36:10.427 N 338: [telegram] Rx: 30 00 97 00 00 00 04 6C 64 00 DC 03 0D 93 D0 00 00 87
000+04:36:10.652 N 339: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:37:05.313 N 340: [telegram] Rx: 30 0B 97 00 00 00 04 48 64 00 DC 03 0D 93 D1 00 00 50
000+04:37:10.674 N 341: [telegram] Rx: 30 00 97 00 00 00 04 43 64 00 DC 03 0D 93 D1 00 00 82
000+04:37:10.900 N 342: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:38:04.939 N 343: [telegram] Rx: 30 0B 97 00 00 00 04 23 64 00 DC 03 0D 93 D2 00 00 B9
000+04:38:10.796 N 344: [telegram] Rx: 30 00 97 00 00 00 04 20 64 00 DC 03 0D 93 D2 00 00 E2
000+04:38:11.022 N 345: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:39:04.745 N 346: [telegram] Rx: 30 0B 97 00 00 00 03 FE 64 00 DD 03 0D 93 D3 00 00 D4
000+04:39:10.918 N 347: [telegram] Rx: 30 00 97 00 00 00 03 FB 64 00 DD 03 0D 93 D3 00 00 23
000+04:39:11.144 N 348: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:40:04.492 N 349: [telegram] Rx: 30 0B 97 00 00 00 03 D9 64 00 DE 03 0D 93 D4 00 00 80
000+04:40:11.164 N 350: [telegram] Rx: 30 00 97 00 00 00 03 D4 64 00 DD 03 0D 93 D4 00 00 3E
000+04:40:11.389 N 351: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:41:04.738 N 352: [telegram] Rx: 30 0B 97 00 00 00 03 B6 64 00 DE 03 0D 93 D5 00 00 A9
000+04:41:11.409 N 353: [telegram] Rx: 30 00 97 00 00 00 03 B2 64 00 DE 03 0D 93 D5 00 00 6C
000+04:41:11.635 N 354: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:42:04.551 N 355: [telegram] Rx: 30 0B 97 00 00 00 03 94 64 00 DE 03 0D 93 D6 00 00 D7
000+04:42:11.593 N 356: [telegram] Rx: 30 00 97 00 00 00 03 90 64 00 DE 03 0D 93 D6 00 00 12
000+04:42:11.818 N 357: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:43:04.918 N 358: [telegram] Rx: 30 0B 97 00 00 00 03 7D 64 00 DE 03 0D 93 D7 00 00 0A
000+04:43:11.713 N 359: [telegram] Rx: 30 00 97 00 00 00 03 7C 64 00 DE 03 0D 93 D7 00 00 35
000+04:43:11.939 N 360: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:44:04.726 N 361: [telegram] Rx: 30 0B 97 00 00 00 03 7F 64 00 DE 03 0D 93 D8 00 00 52
000+04:44:11.896 N 362: [telegram] Rx: 30 00 97 00 00 00 03 80 64 00 DE 03 0D 93 D8 00 00 21
000+04:44:12.121 N 363: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:45:04.472 N 364: [telegram] Rx: 30 0B 97 00 00 00 03 95 64 00 DE 03 0D 93 D9 00 00 D9
000+04:45:12.703 N 365: [telegram] Rx: 30 00 97 00 00 00 03 98 64 00 DE 03 0D 93 D9 00 00 A7
000+04:45:12.928 N 366: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:46:04.530 N 367: [telegram] Rx: 30 0B 97 00 00 00 03 B3 64 00 DD 03 0D 93 DA 00 00 AF
000+04:46:12.948 N 368: [telegram] Rx: 30 00 97 00 00 00 03 B9 64 00 DE 03 0D 93 DA 00 00 8F
000+04:46:13.173 N 369: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:47:04.713 N 370: [telegram] Rx: 30 0B 97 00 00 00 03 D4 64 00 DE 03 0D 93 DB 00 00 CF
000+04:47:13.068 N 371: [telegram] Rx: 30 00 97 00 00 00 03 D9 64 00 DE 03 0D 93 DB 00 00 B1
000+04:47:13.293 N 372: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:48:04.396 N 373: [telegram] Rx: 30 0B 97 00 00 00 03 F5 64 00 DE 03 0D 93 DC 00 00 F7
000+04:48:13.313 N 374: [telegram] Rx: 30 00 97 00 00 00 03 FB 64 00 DE 03 0D 93 DC 00 00 DF
000+04:48:13.538 N 375: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:49:04.953 N 376: [telegram] Rx: 30 0B 97 00 00 00 04 1A 64 00 DE 03 0D 93 DD 00 00 A3
000+04:49:13.563 N 377: [telegram] Rx: 30 00 97 00 00 00 04 1E 64 00 DE 03 0D 93 DD 00 00 66
000+04:49:13.783 N 378: [telegram] Rx: 30 08 35 00 00 00 00 CE
000+04:50:04.449 N 379: [telegram] Rx: 30 0B 97 00 00 00 04 35 64 00 DF 03 0D 93 DE 00 00 EE
000+04:50:13.740 N 380: [telegram] Rx: 30 00 97 00 00 00 04 39 64 00 DF 03 0D 93 DE 00 00 A2
000+04:50:13.965 N 381: [telegram] Rx: 30 08 35 00 00 00 00 CE

so the pump is running now and there are the new telegrams

MichaelDvP, 2021-05-03:
Sorry, there is nothing new in this telegrams, the last bytes stay zero, we read all information that are in this telegrams. 
Maybe the RC35 calculate the gain? What unit do the gain have? Maybe temperature, not energy?
To calculate the solar energy you can use the temperature increase of the warm-water-buffer and the buffer-capacity, Or you can calculate with the flowrate of solar fluid and the temperature difference of collector and buffer. I think SM10 and RC35 do not know the buffer-capacity and solar-flowrate. Inside the solar station there is a small flow-indicator, you can check it at different pump modilation and make a calibration curve for own calculation.

> Perhaps you find more if your SM10 will be delivered ;)

I hope so. Now i use two dallassensors for solar flowtemp in/out of the buffer. But can not read the pump modulation from SC20.

mbenkler, 2021-05-04:
yeah you are right, there ist nothing new in it. 
yeah i also think, that the gain is calculated in the RC35.

MichaelDvP, 2021-05-07:
I've get the SM10, installed it and could get some minutes of sun today. What i see is:
- solar gain in RC35 seems to be the temperature increase of warm water. It is calculated by RC35 and not published.
- the unknown fields in 0x97 keep unknown. But the last two bytes changes sometimes, i've seen 0001, 0002, 0003 and 0806, but have no idea what it means.
- there is a telegram 0x96 for the settings. `SM10Config(0x96), data: FF 18 1E 0A 02 55 28 0A 05 2D 1E 0F 64 28 0A` that i try to decode: 
  - 0: solar modul activated (?)
  - 1: ?
  - 2: min pump modulation (tested)
  - 3: max collector temp in 10degrees (?) 
  - 4: min collector temp in 10degrees (?)
  - 5: max tank temp (tested)
  - 6: min warm water temp (tested)
  - 7: pump turnon difference: (?)
  - 8: pump turnoff difference: (?)
  - 9-14: ? 12 seems to be a percent,  i can write lower values but not higher than 100 (0x64)
The RC35 allows only to change the tested values. The other i have to change and see what happen. For now i tested the turnon/turnoff and it seems right.

I've also added a value to enter a max flowrate to calculate actual power (updated per minute) and energyLastHour as sum of last 60 power values. I assume that flow is linear with pump modulation and solar fluid heat capacity is like water. Buderus/Bosch solarstation recommends ~1 l/min per panel as flow, but there is also a flowindicator inside the solarstation.

I'll test a bit more and than push the solar to dev.
Looks like this now (a few minutes after update, energyLastHour is not for a hour):
[image]

BTW: also add the missing commands for SM100/200, for Junkers-ISM i can not find anything about settings or an example telegram 0x0101, in code there is only tankmaxtemp.

MichaelDvP, 2021-05-11:
Ok, it's now in the [dev-code](https://github.com/emsesp/EMS-ESP32/releases/tag/latest), @mbenkler please try and report if you see something wrong.
For the energy i use now heat capacity of 40%glycole/water, a -25°C solarfluid. I think the calculation error is low if someone uses a different mixture.

@hpanther I added some of the values and settings for SM100, that i found in the code, There are some settings like `doubleMatchFlow_; // double Match Flow, 00=off`, I assume that this is boolean and on is `01`, but need someone with SM100/200 to test. (ems1.0 often uses 0x00/0xFF, but ems+ often 00/01 for booleans).

proddy, 2021-07-19:
@mbenkler can we close this?

closed 2021-07-20 by proddy