emsesp/EMS-ESP32 #52: get_uptime_ms overflows

proddy, 2021-05-04:
https://github.com/emsesp/EMS-ESP32/blob/ee5b1b8c346c08ed61e4c6899a9bd756271f6940/lib/uuid-common/src/get_uptime_ms.cpp#L33

@MichaelDvP I'm trying to understand why you changed this in dev? millis() returns an `unsigned long` which is `uint32_t` and used for all time stamps. The original function handles overflow to a `uint64_t`  so the uptime doesn't reset after 48 days. I remember you made a change for the syslog printing but think we need to revert your change back. Also I want to go back to using millis() in all functions for short-times now we can on ESP32

MichaelDvP, 2021-05-04:
millis() is an arduino function returning 32 bits, but esp32 handles the uptime in a single 64 counter and can be read by esp_timer_get_time(). Millis() in arduino lib also calls the same esp_timer_get_time() and cuts off the high 32 bits. I find it useless to to call millis and add extra code to handle the overflow, instead i directly call the 64 bit timer. But both functions give the same low 32 bits back.

MichaelDvP, 2021-05-04:
[See here](https://github.com/espressif/arduino-esp32/blob/7856de7a57420e494176c16c5138174fe2c1dad0/cores/esp32/esp32-hal-misc.c#L168)

proddy, 2021-05-04:
ok, then probably no need for uuid-common::loop() then which calls `set_time`. All we need is the single function get_uptime_ms().

MichaelDvP, 2021-05-04:
Yes, but now you have the same timestamp for all messages within one loop, if you change every message has a own, more exactly timestamp.
Depends on what you preferre. I think functional it makes no difference.

proddy, 2021-05-04:
ok!

closed 2021-05-04 by proddy