emsesp/EMS-ESP32 #53: New Feature: Configuration service

hpanther, 2021-05-04:
This ticket is to solicit comments on offering configuration services for EMS devices.

Specifically, I would like to implement a configuration page for the solar module MS-100(SM100). The approach could then be extended to other devices that require configuration.

The solar module MS-100 has a number of parameters that must be set for it to work properly (Collector Size, Climate Zone, Maximum Tank Temperature, and more). I would like to add a section "solar configuration" to the web UI settings. Any setting made here would then be stored on the gateway. On device startup, these settings would be programmed to the solar module.

MichaelDvP, 2021-05-10:
@hpanther I've worked a bit on SM10 solar module (see #51) and added some commands for it. I see that you have added a lot of SM100 parameters and values some time ago that are not diplayed yet. @proddy have added the value-edit function to the webinterface, we can read and set values direct from devices-page. 
Is your planned configuration service something other, or can we simply add the values with commands to webinterface?

proddy, 2021-07-19:
Closing this.

closed 2021-07-19 by proddy