emsesp/EMS-ESP32 #55: HA - mqtt - still some warnings in log for value_json.id

giovanne123, 2021-05-04:
**ESP32: v3.1.1b0
HA: 2021.4.6
MQTT:  mosquitto.service - Mosquitto MQTT v3.1/v3.1.1 Broker**

Related: https://github.com/emsesp/EMS-ESP/issues/756

Because I also had lots of warnings in the log, similiar to #756 - today I updated my ESP32 to latest updates.

But I still have some log entries related to "value_json.id".

I already have deleted integrations in HA, deleted in mosquito and restarted mosquito, restarted also EMS-ESP32.

But still no luck :-(

**Someone an idea if there is still a bug or were do I maybe miss to delete/clean something?**

**Looks like I get the warning for every of the five EMS devices/mqtt integration at startup within HA log.
When HA is running I only get it for one device, I think only for device "EMS-ESP Mixer" because for debugging I have set the "Mixer Publish Intevall" to 30 seconds (instead of 10 secondes for the others) in EMS-ESP32 settings.**
see below:

> --> started HA
> 2021-05-04 17:28:29 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
> 2021-05-04 **17:28:48** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:28:48** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:28:48** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:28:48** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:28:48** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 
> 2021-05-04 **17:29:07** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:29:37** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:30:07** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
> 2021-05-04 **17:30:37** WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'

HA integration/devices:
[image]

Some infos for Mixer:
HA device - Mixer02:
**(no ID !?)**
[image]

for e.g. Thermostat there is an "ID":
[image]

MQTT Mixer:
Discover:
[image]
Data:
[image]

KNXBroker, 2021-05-04:
Same for me, warnings are still there..

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1335
First occurred: 20:09:44 (172 occurrences)
Last logged: 20:13:34

Template variable warning: 'dict object' has no attribute 'wWDisinfecting' when rendering '{{value_json.wWDisinfecting}}'
Template variable warning: 'dict object' has no attribute 'wWCharging' when rendering '{{value_json.wWCharging}}'
Template variable warning: 'dict object' has no attribute 'wWRecharging' when rendering '{{value_json.wWRecharging}}'
Template variable warning: 'dict object' has no attribute 'wWTempOK' when rendering '{{value_json.wWTempOK}}'
Template variable warning: 'dict object' has no attribute 'wWActive' when rendering '{{value_json.wWActive}}'
```

giovanne123, 2021-05-04:
@KNXBroker, 
have you deleted the integrations in HA, deleted ems-esp in mosquito and restarted mosquito, and restarted als EMS-ESP32 ? 
Because all the other warnings I was able to solve but only for **"value_json.id"** I still get the warnings, like in my log snippet above.

KNXBroker, 2021-05-05:
Just saw that I have looked in the wrong repository... I have an ESP8266, if my understanding is correct this version will not have this issue solved.

MichaelDvP, 2021-05-05:
@giovanne123 The id is only in mqtt messages, used for a fixed HA-template. But mixer is a bit special because there are multiple devices in one message and the id is in the nest. I (hopefully) fixed it in the dev. 

@KNXBroker I think it is also solved in v2, with a little trick and delayed publish of HA-config. The mixer issue is also in v2 (with type instead of id), but for the values you mentioned it should work. 

BTW: Funny, the issue is one month old and the fix seems to work, but a few hours after release it pops up again.

giovanne123, 2021-05-05:
fine now in dev for the id.

But when restarting EMS-ESP32 I get all the other warnings in HA (once) in startup phase. Currently don't now why.
As it is only at startup and everything so far seems to work (what I need) for me, I will stay with it. It isnt flooting my HA log when running ;-)

BTW: For me have tested it a month ago and afterwards as workaround deactivated the warnings in HA configuration. Was no time to test it with every change in the meantime.

For info, when restarting EMS-ESP32:
> 2021-05-05 07:17:11 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
> 2021-05-05 07:17:29 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform command_line is taking over 10 seconds.
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'selflowtemp' when rendering '{{value_json.selflowtemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'selburnpow' when rendering '{{value_json.selburnpow}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingpumpmod' when rendering '{{value_json.heatingpumpmod}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'outdoortemp' when rendering '{{value_json.outdoortemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'curflowtemp' when rendering '{{value_json.curflowtemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'rettemp' when rendering '{{value_json.rettemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'syspress' when rendering '{{value_json.syspress}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'boiltemp' when rendering '{{value_json.boiltemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'flamecurr' when rendering '{{value_json.flamecurr}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingtemp' when rendering '{{value_json.heatingtemp}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'pumpmodmax' when rendering '{{value_json.pumpmodmax}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'pumpmodmin' when rendering '{{value_json.pumpmodmin}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'pumpdelay' when rendering '{{value_json.pumpdelay}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'burnminperiod' when rendering '{{value_json.burnminperiod}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'burnminpower' when rendering '{{value_json.burnminpower}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'burnmaxpower' when rendering '{{value_json.burnmaxpower}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'boilhyston' when rendering '{{value_json.boilhyston}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'boilhystoff' when rendering '{{value_json.boilhystoff}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'curburnpow' when rendering '{{value_json.curburnpow}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'burnstarts' when rendering '{{value_json.burnstarts}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'burnworkmin' when rendering '{{value_json.burnworkmin}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatworkmin' when rendering '{{value_json.heatworkmin}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'ubauptime' when rendering '{{value_json.ubauptime}}'
> 2021-05-05 07:18:41 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'lastcode' when rendering '{{value_json.lastcode}}'
[... cut ...]
> 2021-05-05 07:19:31 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'program' when rendering '{{value_json.hc2.program}}'
> 2021-05-05 07:19:31 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'pause' when rendering '{{value_json.hc2.pause}}'
> 2021-05-05 07:19:31 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'party' when rendering '{{value_json.hc2.party}}'

MichaelDvP, 2021-05-05:
> But when restarting EMS-ESP32 I get all the other warnings in HA (once) in startup phase.

Yes, the values are registered in HA already and ems-esp has to collect them from the bus after restart, depending on publish-time you get some mqtt-messages with not all parameters. The only way to avoid this is not sending messages for some minutes after start up. But how often do you restart?

giovanne123, 2021-05-05:
Its ok for me, because normally I only restarting when I'm at testing/debugging. My intention was only to report it ;-)

closed 2021-05-05 by giovanne123