emsesp/EMS-ESP32 #56: HA no longer silently accepts templates referring to undefined variables.

glitter-ball, 2021-05-09:
**Bug description**
HA core log growing rapidly, every 9 seconds, with errors of the form:
```
2021-05-09 14:06:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wwMixTemperature' when rendering '{{value_json.wwMixTemperature}}'
2021-05-09 14:06:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wwBufferTemperature' when rendering '{{value_json.wwBufferTemperature}}'
2021-05-09 14:06:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWStorageTemp1' when rendering '{{value_json.wWStorageTemp1}}'
2021-05-09 14:06:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'mixerTemp' when rendering '{{value_json.mixerTemp}}'
2021-05-09 14:06:36 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'tankMiddleTemp' when rendering '{{value_json.tankMiddleTemp}}'
```

**Steps to reproduce**
Connect 3.0.1b1 to HA 2021.4 or later with auto-discovery enabled.

**Expected behavior**
No logging of errors like this.

**Additional context**
Other reports of similar issues online, mainly from people using Zigbee2mqtt connector. HA, apparently, no longer accepts template definitions which refer to undefined variables, which may be a consequence of auto-discovery. More info at...

https://github.com/home-assistant/core/issues/48851
https://github.com/Koenkk/zigbee2mqtt/issues/6987

Looks like Zigbee2mqtt code had to be modded to stop this - maybe something similar required for EMS-ESP?

glitter-ball, 2021-05-10:
OK... I've upgraded to 3.1.0 and rebooted HA. That didn't fix it.

Shut the gateway down and deleted all topics in MQTT Explorer and restarted the gateway. 

Restarted HA. Now I have lots of broken bits and errors still logging...

[image]

```

2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'serviceCodeNumber' when rendering '{% if value_json.serviceCodeNumber == 200 %} CH active {% elif value_json.serviceCodeNumber == 201 %} HW active {% elif value_json.serviceCodeNumber == 202 %} CH anti cycle {% elif value_json.serviceCodeNumber == 203 %} Stand by {% elif value_json.serviceCodeNumber == 204 %} CH cooling {% elif value_json.serviceCodeNumber == 208 %} Service test {% elif value_json.serviceCodeNumber == 265 %} Low CH load {% elif value_json.serviceCodeNumber == 268 %} Component test {% elif value_json.serviceCodeNumber == 270 %} Power up {% elif value_json.serviceCodeNumber == 283 %} Burner start {% elif value_json.serviceCodeNumber == 284 %} Ignition {% elif value_json.serviceCodeNumber == 305 %} HW anti cycle {% elif value_json.serviceCodeNumber == 357 %} Air purge {% elif value_json.serviceCodeNumber == 358 %} Valve kick {% else %} {{ value_json.serviceCodeNumber }} {% endif %}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWCurTemp' when rendering '{{ value_json.wWCurTemp }}'
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'serviceCodeNumber' when rendering '{% if value_json.serviceCodeNumber == 200 %} CH active {% elif value_json.serviceCodeNumber == 201 %} HW active {% elif value_json.serviceCodeNumber == 202 %} CH anti cycle {% elif value_json.serviceCodeNumber == 203 %} Stand by {% elif value_json.serviceCodeNumber == 204 %} CH cooling {% elif value_json.serviceCodeNumber == 208 %} Service test {% elif value_json.serviceCodeNumber == 265 %} Low CH load {% elif value_json.serviceCodeNumber == 268 %} Component test {% elif value_json.serviceCodeNumber == 270 %} Power up {% elif value_json.serviceCodeNumber == 283 %} Burner start {% elif value_json.serviceCodeNumber == 284 %} Ignition {% elif value_json.serviceCodeNumber == 305 %} HW anti cycle {% elif value_json.serviceCodeNumber == 357 %} Air purge {% elif value_json.serviceCodeNumber == 358 %} Valve kick {% else %} {{ value_json.serviceCodeNumber }} {% endif %}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWCurTemp' when rendering '{{ value_json.wWCurTemp }}'
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'serviceCodeNumber' when rendering '{% if value_json.serviceCodeNumber == 200 %} CH ac [... cut ...]
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-10 12:21:01 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-10 12:21:01 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
```

Have there been point changes between 3.0.1b1 and 3.1.0 that could cause the broken stuff or is something else going on? Maybe I should go back to 3.0.1b1 and see what happens?

glitter-ball, 2021-05-10:
Clarifying the broken stuff...

* Some points missing - yellow triangle
* Some points showing 'Unknown'
* Friendly names for lots of points set up in HA GUI now lost

Oops.

glitter-ball, 2021-05-10:
Restored config from a recent snapshot and restarted the gateway which has fixed some stuff, but it's still pretty sick.

* Outside temp shows as 0C on the dashboard but 13C in the history!
* Loads of points now just blank
* Hot water climate entity broken
* Core log absolutely rammed with "Template variable warning: 'dict object' has no attribute" errors.

proddy, 2021-05-10:
and for those elements that mysteriously vanished, let us know and we'll look into the code. We made some changes to only detect and scan for know values on the EMS bus depending on brand and model. Some boilers don't have all the 80 values we support and it was wasteful in memory and cpu

MichaelDvP, 2021-05-10:
v3.1 uses only lowercase valuenames, all error-messages shown missing values for camelcase, this ist not from HA-dicovery, it is stored somewhere in mqtt or HA.

glitter-ball, 2021-05-10:
A check in Configuration/Entities finds loads of duplicate points ending in '_2'. Not sure, at the moment, how they got there or how to sort it out!

glitter-ball, 2021-05-10:
> 
> 
> v3.1 uses only lowercase valuenames, all error-messages shown missing values for camelcase, this ist not from HA-dicovery, it is stored somewhere in mqtt or HA.

@MichaelDvP - I've got some hand-crafted YAML in sensor.yaml with mixed-case point names, e.g.
```
- platform: mqtt
  state_topic: 'ems-esp/boiler_data_ww'
  name: 'Diverter'
  icon: 'mdi:call-split'
  value_template: "{% if value_json.wWHeat == 'off' %} CH {% else %} HW {% endif %}"
```
Is the upshot of your comment that I need to change value_json.wWHeat to value_json.wwheat? Thanks.

glitter-ball, 2021-05-11:
Looked in MQTT Explorer tonight...
[image]

Several things jump out...
* The binary sensor definitions are duplicated in homeassistant/binary_sensor/ems-esp - one has mixed case name (heatingActivated), one has lower case name (heatingactivated)
* There's a thirdentry in sensor/ems-esp/boiler_heatingActivated

How did it get like this? How do I fix it? I thought I'd deleted all the MQTT points in MQTT Explorer. When I go in an manually delete all three, two of them disappear, homeassistant/binary_sensor/ems-esp/boiler_heatingActivated then re-appears. Is that the right one? If all points in EMS-ESP are now in lower case, why is my gateway still publishing topics in mixed case, when I'm running 3.1.0? Confused (very).

proddy, 2021-05-11:
using MQTTExplorer delete `homeassistant` and reboot EMS-ESP. You don't need to do anything in HA.

glitter-ball, 2021-05-11:
Had something like 230 topics under homeassistant. Deleted the tree and they gradually disappeared over about 2 minutes. Rebooted EMS-ESP and eventually 103 topics / 103 messages have reappeared, so quite a reduction.

But, the errors are still coming thick and fast...
```
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'serviceCodeNumber' when rendering '{% if value_json.serviceCodeNumber == 200 %} CH active {% elif value_json.serviceCodeNumber == 201 %} HW active {% elif value_json.serviceCodeNumber == 202 %} CH anti cycle {% elif value_json.serviceCodeNumber == 203 %} Stand by {% elif value_json.serviceCodeNumber == 204 %} CH cooling {% elif value_json.serviceCodeNumber == 208 %} Service test {% elif value_json.serviceCodeNumber == 265 %} Low CH load {% elif value_json.serviceCodeNumber == 268 %} Component test {% elif value_json.serviceCodeNumber == 270 %} Power up {% elif value_json.serviceCodeNumber == 283 %} Burner start {% elif value_json.serviceCodeNumber == 284 %} Ignition {% elif value_json.serviceCodeNumber == 305 %} HW anti cycle {% elif value_json.serviceCodeNumber == 357 %} Air purge {% elif value_json.serviceCodeNumber == 358 %} Valve kick {% else %} {{ value_json.serviceCodeNumber }} {% endif %}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWCurTemp' when rendering '{{ value_json.wWCurTemp }}'
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'serviceCodeNumber' when rendering '{% if value_json.serviceCodeNumber == 200 %} CH active {% elif value_json.serviceCodeNumber == 201 %} HW active {% elif value_json.serviceCodeNumber == 202 %} CH anti cycle {% elif value_json.serviceCodeNumber == 203 %} Stand by {% elif value_json.serviceCodeNumber == 204 %} CH cooling {% elif value_json.serviceCodeNumber == 208 %} Service test {% elif value_json.serviceCodeNumber == 265 %} Low CH load {% elif value_json.serviceCodeNumber == 268 %} Component test {% elif value_json.serviceCodeNumber == 270 %} Power up {% elif value_json.serviceCodeNumber == 283 %} Burner start {% elif value_json.serviceCodeNumber == 284 %} Ignition {% elif value_json.serviceCodeNumber == 305 %} HW anti cycle {% elif value_json.serviceCodeNumber == 357 %} Air purge {% elif value_json.serviceCodeNumber == 358 %} Valve kick {% else %} {{ value_json.serviceCodeNumber }} {% endif %}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWCurTemp' when rendering '{{ value_json.wWCurTemp }}'
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'heatingActivated' when rendering '{{ value_json.heatingActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWActivated' when rendering '{{ value_json.wWActivated }}'
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWComfort' when rendering '{{ value_json.wWComfort }}'
[... cut ...]
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
2021-05-11 20:35:02 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'wWSelTemp' when rendering '{{ value_json.wWSelTemp }}'
2021-05-11 20:35:02 ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from 
```

Presumably, I now need to go through my YAML files and sort out all the ones with mixed case and convert them to lower case only?

glitter-ball, 2021-05-11:
Getting there. Key findings...

- Anything in YAML with mixed case point names needed changing to lower case only.
- Quite a few points which were in boiler_data are now in boiler_data_ww, so the names need updating to reflect that.
- Loads of stuff in the heartbeat tree which didn't match the current map.

Dealing with this has fixed all the errors in the core log - hooray!

Still quite a few broken points where the point name or name format has changed. I can probably fix these in the Entities view. A job for tomorrow.

glitter-ball, 2021-05-14:
Thanks, all, for your help. Normal service now resumed. Closing...

closed 2021-05-14 by glitter-ball