emsesp/EMS-ESP32 #63: Improve HA icons

proddy, 2021-05-12:
The device values can have a UOM (Unit of Measure) which associates with an icon with Home Assistant. Currently there are just 10 defined but it's easy to extend or modify the default icons. Icons are from https://materialdesignicons.com

```txt
emsdevice.h:

MAKE_PSTR(icontemperature, "mdi:coolant-temperature")
MAKE_PSTR(iconpercent, "mdi:percent-outline")
MAKE_PSTR(iconfire, "mdi:fire")
MAKE_PSTR(iconfan, "mdi:fan")
MAKE_PSTR(iconflame, "mdi:flash")
MAKE_PSTR(iconvalve, "mdi:valve")
MAKE_PSTR(iconpump, "mdi:pump")
MAKE_PSTR(iconheatpump, "mdi:water-pump")
MAKE_PSTR(iconclock, "mdi:clock-outline")
MAKE_PSTR(iconmemory, "mdi:memory")
```

PhillyGilly, 2021-05-12:
A few for starters.
On EMS-ESP
 Wifi Signal Strength  "mdi:access-point" (as a rule I think the % in UOM is enough)
 System Uptime, "mdi:clock-check-outline"
MQTT system fails "mdi:message-alert-outline"
MQTT tx "mdi:message-arrow-right-outline"
MQTT rx "mdi:message-arrow-right-outline"

proddy, 2021-05-15:
and while we're at it, add UOM to system variables as @gliiter-ball mentioned "Quite a few of the EMS-ESP system points are formatted as text, not numbers in Recorder : TX/RX fail count, MQTT fail count etc. So they all end up as a bar of numbers rather than a nice graph."

The free mem was already set to 'kb', just need to do the others

proddy, 2021-05-16:
currently, EMS-ESP maps the Unit of Measure/UOM (like percentage, kb, time etc) to an icon. Support for setting an icon for each sensor/entity isn't implemented. So this will be a bit of extra work

proddy, 2021-05-17:
re-opening to provide some further clarification.
- Each device value can be assigned a UOM
- This UOM is used when displaying the values in the Console and Web
- The UOM is also used in Home Assistant (to also set the correct formatting) and used to determine which icon to use
- The UOM used in EMS-ESP are: 
```
DeviceValueUOM:
    NONE,
    DEGREES,
    PERCENT,
    LMIN,
    KWH,
    WH,
    HOURS,
    MINUTES,
    UA,
    BAR,
    KW,
    W,
    KB,
    SECONDS,
    DBM,
    NUM
```
 
The HA used are
```
mdi:coolant-temperature     // DeviceValueUOM::DEGREES
mdi:percent-outline         // DeviceValueUOM::PERCENT
mdi:clock-outline           // DeviceValueUOM::SECONDS MINUTES & HOURS
mdi:memory                  // DeviceValueUOM::KB
mdi:water-boiler            // DeviceValueUOM::LMIN
mdi:transmission-tower      // DeviceValueUOM::KWH & WH
mdi:flash-circle            // DeviceValueUOM::UA
mdi:gauge                   // DeviceValueUOM::BAR
mdi:omega                   // DeviceValueUOM::KW & W
mdi:wifi-strength-2         // DeviceValueUOM::DBM
mdi:counter                 // DeviceValueUOM::NUM
```

It is possible to extend the code to assign an HA icon to every Device Value. It's extra work, and takes up extra memory and only useful to those that customize Home Assistant. Also icons are very personal and my approach here would be to create a customize.yaml file, which is persisted in HA, with any custom naming and icons for each entity.

PhillyGilly, 2021-05-17:
If you use the icon as a visual clue (e.g. in HA glance card) then having the icon based on UOM doesn't always help. For example in Pump Modulation (sensor.boiler_heating_pump_modulation) I would chose mdi:pump rather than mdi:percent-outline.  My proposed alternative is a small number of icons which are linked to the entity name e.g.  time, temperature, fan, valve, memory, wifi, burner, etc. So the look-up would be on DeviceName rather than DeviceValueUOM.

proddy, 2021-05-17:
Right, so to add to the enhancement list would be the possibility to include an optional HA icon for each device value. If none is included it will take the one mapping to it's UOM.

PhillyGilly, 2021-05-17:
Not really. I'm happy that HA users who are into heavy customisation can do it in HA. The enhancement would be that the default mapping should be to the name backed up by UOM if there isn't an icon keyword in the name.

proddy, 2021-05-17:
think we're saying the same thing. There are 200 device values in EMS-ESP at the moment so I'll need your help to do the mapping. fun!

PhillyGilly, 2021-05-17:
OK. Let's do it!

PhillyGilly, 2021-05-19:
Can you post a full list of the devices which I can export to excel to save me rootling through all the header files? TIA.

proddy, 2021-06-17:
```

src/devices/boiler.cpp:
   36:         register_device_value(TAG_HS1 + hs, &burnWorkMin_, DeviceValueType::TIME, nullptr, FL_(burnWorkMin), DeviceValueUOM::MINUTES);
   41:         register_device_value(TAG_HS1 + hs, &selFlowTemp_, DeviceValueType::UINT, nullptr, FL_(selFlowTemp), DeviceValueUOM::DEGREES);
   42:         register_device_value(TAG_HS1 + hs, &selBurnPow_, DeviceValueType::UINT, nullptr, FL_(selBurnPow), DeviceValueUOM::PERCENT);
   43:         register_device_value(TAG_HS1 + hs, &curFlowTemp_, DeviceValueType::USHORT, FL_(div10), FL_(curFlowTemp), DeviceValueUOM::DEGREES);
   44:         register_device_value(TAG_HS1 + hs, &curBurnPow_, DeviceValueType::UINT, nullptr, FL_(curBurnPow), DeviceValueUOM::PERCENT);
   98:         register_device_value(TAG_BOILER_DATA, &heatingActive_, DeviceValueType::BOOL, nullptr, FL_(heatingActive), DeviceValueUOM::BOOLEAN);
   99:         register_device_value(TAG_BOILER_DATA, &tapwaterActive_, DeviceValueType::BOOL, nullptr, FL_(tapwaterActive), DeviceValueUOM::BOOLEAN);
  100:         register_device_value(TAG_BOILER_DATA, &selFlowTemp_, DeviceValueType::UINT, nullptr, FL_(selFlowTemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_flow_temp));
  101:         register_device_value(TAG_BOILER_DATA, &selBurnPow_, DeviceValueType::UINT, nullptr, FL_(selBurnPow), DeviceValueUOM::PERCENT, MAKE_CF_CB(set_burn_power));
  102:         register_device_value(TAG_BOILER_DATA, &heatingPumpMod_, DeviceValueType::UINT, nullptr, FL_(heatingPumpMod), DeviceValueUOM::PERCENT);
  103:         register_device_value(TAG_BOILER_DATA, &heatingPump2Mod_, DeviceValueType::UINT, nullptr, FL_(heatingPump2Mod), DeviceValueUOM::PERCENT);
  104:         register_device_value(TAG_BOILER_DATA, &outdoorTemp_, DeviceValueType::SHORT, FL_(div10), FL_(outdoorTemp), DeviceValueUOM::DEGREES);
  105:         register_device_value(TAG_BOILER_DATA, &curFlowTemp_, DeviceValueType::USHORT, FL_(div10), FL_(curFlowTemp), DeviceValueUOM::DEGREES);
  106:         register_device_value(TAG_BOILER_DATA, &retTemp_, DeviceValueType::USHORT, FL_(div10), FL_(retTemp), DeviceValueUOM::DEGREES);
  107:         register_device_value(TAG_BOILER_DATA, &switchTemp_, DeviceValueType::USHORT, FL_(div10), FL_(switchTemp), DeviceValueUOM::DEGREES);
  108:         register_device_value(TAG_BOILER_DATA, &sysPress_, DeviceValueType::UINT, FL_(div10), FL_(sysPress), DeviceValueUOM::BAR);
  109:         register_device_value(TAG_BOILER_DATA, &boilTemp_, DeviceValueType::USHORT, FL_(div10), FL_(boilTemp), DeviceValueUOM::DEGREES);
  110:         register_device_value(TAG_BOILER_DATA, &exhaustTemp_, DeviceValueType::USHORT, FL_(div10), FL_(exhaustTemp), DeviceValueUOM::DEGREES);
  111:         register_device_value(TAG_BOILER_DATA, &burnGas_, DeviceValueType::BOOL, nullptr, FL_(burnGas), DeviceValueUOM::BOOLEAN);
  112:         register_device_value(TAG_BOILER_DATA, &flameCurr_, DeviceValueType::USHORT, FL_(div10), FL_(flameCurr), DeviceValueUOM::UA);
  113:         register_device_value(TAG_BOILER_DATA, &heatingPump_, DeviceValueType::BOOL, nullptr, FL_(heatingPump), DeviceValueUOM::BOOLEAN);
  114:         register_device_value(TAG_BOILER_DATA, &fanWork_, DeviceValueType::BOOL, nullptr, FL_(fanWork), DeviceValueUOM::BOOLEAN);
  115:         register_device_value(TAG_BOILER_DATA, &ignWork_, DeviceValueType::BOOL, nullptr, FL_(ignWork), DeviceValueUOM::BOOLEAN);
  116:         register_device_value(TAG_BOILER_DATA, &heatingActivated_, DeviceValueType::BOOL, nullptr, FL_(heatingActivated), DeviceValueUOM::BOOLEAN, MAKE_CF_CB(set_heating_activated));
  117:         register_device_value(TAG_BOILER_DATA, &heatingTemp_, DeviceValueType::UINT, nullptr, FL_(heatingTemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_heating_temp));
  118:         register_device_value(TAG_BOILER_DATA, &pumpModMax_, DeviceValueType::UINT, nullptr, FL_(pumpModMax), DeviceValueUOM::PERCENT, MAKE_CF_CB(set_max_pump));
  119:         register_device_value(TAG_BOILER_DATA, &pumpModMin_, DeviceValueType::UINT, nullptr, FL_(pumpModMin), DeviceValueUOM::PERCENT, MAKE_CF_CB(set_min_pump));
  120:         register_device_value(TAG_BOILER_DATA, &pumpDelay_, DeviceValueType::UINT, nullptr, FL_(pumpDelay), DeviceValueUOM::MINUTES, MAKE_CF_CB(set_pump_delay));
  121:         register_device_value(TAG_BOILER_DATA, &burnMinPeriod_, DeviceValueType::UINT, nullptr, FL_(burnMinPeriod), DeviceValueUOM::MINUTES, MAKE_CF_CB(set_burn_period));
  122:         register_device_value(TAG_BOILER_DATA, &burnMinPower_, DeviceValueType::UINT, nullptr, FL_(burnMinPower), DeviceValueUOM::PERCENT, MAKE_CF_CB(set_min_power));
  123:         register_device_value(TAG_BOILER_DATA, &burnMaxPower_, DeviceValueType::UINT, nullptr, FL_(burnMaxPower), DeviceValueUOM::PERCENT, MAKE_CF_CB(set_max_power));
  124:         register_device_value(TAG_BOILER_DATA, &boilHystOn_, DeviceValueType::INT, nullptr, FL_(boilHystOn), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_hyst_on));
  125:         register_device_value(TAG_BOILER_DATA, &boilHystOff_, DeviceValueType::INT, nullptr, FL_(boilHystOff), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_hyst_off));
  126:         register_device_value(TAG_BOILER_DATA, &setFlowTemp_, DeviceValueType::UINT, nullptr, FL_(setFlowTemp), DeviceValueUOM::DEGREES);
  127:         register_device_value(TAG_BOILER_DATA, &setBurnPow_, DeviceValueType::UINT, nullptr, FL_(setBurnPow), DeviceValueUOM::PERCENT);
[... cut ...]
  2289:         register_device_value(tag, &hc->tempautotemp, DeviceValueType::UINT, FL_(div2), FL_(tempautotemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_tempautotemp));
  2290:         register_device_value(tag, &hc->noreducetemp, DeviceValueType::INT, nullptr, FL_(noreducetemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_noreducetemp));
  2291:         register_device_value(tag, &hc->remotetemp, DeviceValueType::SHORT, FL_(div10), FL_(remotetemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_remotetemp));
```

closed 2021-08-01 by proddy