emsesp/EMS-ESP32 #64: Home assistants MQTT show numerous entities twice

swa72, 2021-05-13:
I have recently upgraded to  https://github.com/emsesp/EMS-ESP32/releases/tag/v3.1.0, restarted the EMS-ESP device and had to configure it manually again. Didn't take too long. I noticed that there were a bunch of entities renamed - that's ok. But I found numerous entities that show up twice (with a "_2" postfix), here are two examples
- sensor.boiler_flame_current
- sensor.boiler_flame_current_2

Anyone an idea how to get rid of them?  Maybe related to another issue. HA is complaining of

```
Logger: homeassistant.helpers.template
Source: helpers/template.py:1367
First occurred: 15:53:03 (43019 occurrences)
Last logged: 16:55:14

Template variable warning: 'dict object' has no attribute 'status' when rendering '{{value_json.hc3.status}}'
Template variable warning: 'dict object' has no attribute 'id' when rendering '{{value_json.id}}'
Template variable warning: 'dict object' has no attribute 'flowSetTemp' when rendering '{{value_json.hc2.flowSetTemp}}'
Template variable warning: 'dict object' has no attribute 'flowTemp' when rendering '{{value_json.hc2.flowTemp}}'
Template variable warning: 'dict object' has no attribute 'status' when rendering '{{value_json.hc2.status}}'
```

proddy, 2021-05-13:
This came up earlier so #62 was created. In the meantime use MQTTExplorer to delete everything under `homeassistant/*/ems-esp` and restart EMS-ESP. HA will clean itself up.

swa72, 2021-05-13:
Hhhm, I did run MQTTExplorer. removed everything   

[image]

restarted EMS-ESP but the "_2" entities are still populated with data. Odd.

proddy, 2021-05-13:
oh. not good. I just did a quick google on this and found 
https://community.home-assistant.io/t/duplicate-entries-with-2/181362/16
https://community.home-assistant.io/t/how-to-prevent-same-entity-to-be-created-again-but-with-2-after-name/160683/9
https://community.home-assistant.io/t/edit-core-entity-registry/83228

Looks like the solution is to delete the file `.storage/core_entity_registry` and restart HA. All valid entities will be recreated. Make sure you make a backup first though just in case!

swa72, 2021-05-13:
Here is what I did

1. disabled MQTT on the EMS-ESP
2. removed everything from `homeassistant/*/ems-esp`
3. removed `.storage/core_entity_registry`
4. restarted HA 
5. enabled MQTT on the EMS-ESP

proddy, 2021-05-13:
great, if that worked I'll add it to the wiki. Thanks for trying it out

swa72, 2021-05-14:
Well, there is one drawback if your remove the entity registry - all your entity names are re-created. If you have renamed them in the past, you'll have to do it again. And that is a PITA ;-/

swa72, 2021-05-15:
Everything is back to normal - except "climate.thermostat_hc1" is missing (also hc2 and hc3). MQTT discovered all devices, I clicked the "ADD TO LOVELACE" button. Everything shows up except the climate controls. Per "https://bbqkees-electronics.nl/wiki/gateway/home-assistant-configuration-heating-circuits.html" I thought that I may need to add stuff for 2 and 3 manually. But 1?

proddy, 2021-05-15:
that script from BBQKees is old. In HA you want to show all 3 climate controls and control them independently?

swa72, 2021-05-15:
Exactly. They were present before the update to 3.0 but magically
disappeared.

proddy, 2021-05-15:
ok, I'll take a look.

proddy, 2021-05-16:
@swa72 you don't see the thermostat in HA and MQTTExplorer like this?

[image]

[image]

swa72, 2021-05-16:
Odd that. MQTT explorer shows them now

[image]

I just need to get them back into HA.  Checked that and ... they are magically back. Huh? 

Other than a couple of HA restarts I didn't do anything magic to make them reappear. I assume there must be some time for all MQTT message and hence devices to show up in HA before one does the "add to lovelace" click.