emsesp/EMS-ESP32 #66: Incorrect Thermostat mode mapping in Home Assistant (RC310)

proddy, 2021-05-15:
As reported in Discord by @ghost2021

When RC310 is in mode 'auto' the hamode in the MQTT payload is 'heat', when it should be 'auto'.
When RC310 is in mode 'heat' the hamode in the MQTT payload is 'off', when it should be 'heat'

ecs-ghost, 2021-05-15:
as example I can provide Mqtt payload with two hc

{"id":158,"errorcode":"(0)","datetime":"12:18:11 15/05/2021","floordry":"off","dampedoutdoortemp":27.8,"floordrytemp":0,"building":"heavy","wwsettemp":60,"wwmode":"own_prog","wwsettemplow":45,"wwcircmode":"own_prog","wwextra1":0,"hc1":{"seltemp":24,"currtemp":24.5,"hatemp":24.5,"hamode":"heat","mode":"auto","modetype":"comfort","ecotemp":21,"manualtemp":20.5,"comforttemp":24,"summertemp":15,"designtemp":60,"offsettemp":0,"minflowtemp":25,"maxflowtemp":65,"roominfluence":3,"nofrosttemp":5,"targetflowtemp":0,"heatingtype":"radiator","summermode":"on","controlmode":"optimized","program":1},"hc2":{"seltemp":25,"currtemp":23.6,"hatemp":23.6,"hamode":"off","mode":"manual","modetype":"comfort","ecotemp":20,"manualtemp":25,"comforttemp":23,"summertemp":15,"designtemp":75,"offsettemp":0,"minflowtemp":25,"maxflowtemp":80,"roominfluence":3,"nofrosttemp":5,"targetflowtemp":32,"heatingtype":"radiator","summermode":"off","controlmode":"optimized","program":1}}

both HC (1 and 2) "hamode" is not correct

proddy, 2021-05-15:
I found the error. working on the most elegant fix for it.

MichaelDvP, 2021-05-15:
@proddy I think you had a elegant solution in mind when creating the long enum_hamode list. 

For that stupid ha-thermostat a nonelegant but logical methode is to register `hc->hamode` and add in the monitor telegrams 
RC20, RC20_2, RC30/RC35, Junkers:  `hc->hamode = hc->mode;`
RC300:  `hc->hamode = hc->mode + 1;`
CRF200:  `hc->hamode = 2 - 2 * hc->mode;`

proddy, 2021-05-15:
didn't think of that. I did a dumb solution instead. It works, and will check-in with a few other fixes.

proddy, 2021-05-16:
Michael's idea was much smarter than mine, so implemented that way. In 3.1.1b3

proddy, 2021-05-16:
@ecs-ghost please test and re-open if it doesn't work

closed 2021-05-16 by proddy

ecs-ghost, 2021-05-16:
Compiled and uploaded 
Seems like working correctly now 
I'll test it a little bit more