emsesp/EMS-ESP32 #69: build interface error

MichaelDvP, 2021-05-17:
Since 3 days the interface does not build anymore on windows systems. 
It throws more than 1000 messages of type 
` Line 38:4:   Delete â\x90\x8d  prettier/prettier`
and after a a large number an error.
```
 Line 38:4:   Delete `â\x90\x8d`  prettier/prettier
*** Error 1
  Line 39:2:   Delete `â\x90\x8d`  prettier/prettier
```
The deleteted char is the `cr` from windows file system, i think on linux it works, but it should also build on windows systems.

My system:
```
Visual Studio Code 1.56.2
Node.js: 14.16.0
NPM: 7.11.2
Windows x64 10.0.19042
```

proddy, 2021-05-17:
I'll take a look. Also, unrelated, the automatic GitHub build action is also not working.

proddy, 2021-05-17:
I made a quick fix. Please try.

By the way, I know I've said this before, I do all my coding on a Win10 desktop but run VSCode in WSL2 with an Ubuntu virtual environment. It's only a few lines to install from Windows and the advantage is that it compiles and builds EMS-ESP in roughly twice that speed because of the gcc libraries. Plus I can use 'make' and the mock-api to do most of the development without ever having to upload to an real ESP32. The disadvantage is that it does eat up a lot of system memory (5GB) at run time but you can turn if off by just doing WindowKey-R `wsl --shutddown`.

MichaelDvP, 2021-05-17:
Thanks, it works now.

closed 2021-05-17 by MichaelDvP