emsesp/EMS-ESP32 #71: Show realtime debug log in WebUI

proddy, 2021-06-09:
Was wondering if it's at all useful at all to show the debug log (what is shown in the Console and SysLog) to the WebUI? It's fairly easy to implement using eventsource and consist of a new tab in the System section of the WebUI which shows the messages coming in. And the ability to change the level (off, debug, all etc).

@MichaelDvP what do you think? Waste of time or useful?

MichaelDvP, 2021-06-09:
Usefull if you add a button for logging to file. When asking for a telnet log we see a lot of screenshots with only a few lines, because people don't find the log option in telnet. It can be fixed to log_all and maybe a fixed timeout of 3-5 min to close the file automatically.

bbqkees, 2021-06-10:
If its not much effort it could be worthwhile to implement for those not technical enough for Syslog.

proddy, 2021-06-16:
First implementation done. Need to run some stability tests and keep an eye on memory and bandwidth usage.

[image]

proddy, 2021-06-17:
@MichaelDvP let me know what you think. If ok, I'll merge into dev.

I did think about a Save button (and implemented it) but cut&paste is just as easy

MichaelDvP, 2021-06-18:
> let me know what you think. If ok, I'll merge into dev.

Works good. Only two comments:
- the log-selector should use the same names as in settings-syslog: replace  `TRACE` with `ALL` (not in console, only in selector)
- on your big screen there is a large console area, but on a small Laptop-screen it's only 12 lines. If you move the note about buffering beside the selector there is one more line for the console.

proddy, 2021-06-18:
thanks. I'll change TRACE to ALL and move the text to next to the pull-down menu.
I could also change the font-size? it's 14pt now. Making it 12pt will give a lot more records.

closed 2021-06-18 by proddy

proddy, 2021-07-27:
I made some improvements to the log. The level wasn't working great before. looks like this now:

[image]