emsesp/EMS-ESP32 #72: ESP32 EMS Connection Version 3.1

stmb-100, 2021-06-11:
Since version 3.1.0 I can no longer establish a stable connection with my EMS. The problem also exists with version v3.1.1b6
With the last 3.0.1 version everything still works stable.
TX-Mode = EMS, BusID = ServiceKey, TX Start delay = 2, Board Profile, BBQKees Gateway S32.

proddy, 2021-06-11:
that's not good. What happens exactly? Are you getting many Rx errors?

stmb-100, 2021-06-11:
[image]
[image]

stmb-100, 2021-06-11:
[image]

stmb-100, 2021-06-11:
With Version 3.01

[image]

proddy, 2021-06-12:
Which board are you using? I suspect it's a problem with migration to 3.1 and the new Board Profiles. Go to the "EMS-ESP Help" tab and click "To export your system settings click here" and look at `System.tx_gpio`. This GPIO pin should match what you had before.

stmb-100, 2021-06-12:
The Board is a ESP32 WROOM.32 D1 Mini

blob:file:///4a05c4aa-9b87-4be6-9bfd-1b1738a84c96

stmb-100, 2021-06-12:
Version 3.1
  "Settings": {
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "master_thermostat": 0,
    "shower_timer": true,
    "shower_alert": false,
    "rx_gpio": 23,
    "tx_gpio": 5,
    "dallas_gpio": 18,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "notoken_api": true,
    "analog_enabled": true,
    "pbutton_gpio": 0,
    "board_profile": "S32"
  }
 
Version 3.0
  "Settings": {
    "tx_mode": 1,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "master_thermostat": 0,
    "shower_timer": true,
    "shower_alert": true,
    "rx_gpio": 23,
    "tx_gpio": 5,
    "dallas_gpio": 18,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": false,
    "api_enabled": true,
    "analog_enabled": true,
    "pbutton_gpio": 0,
    "board_profile": "S32"
  }

proddy, 2021-06-12:
that is strange, same settings. Although you should probably change your board profile to be "MH-ET Live D1 Mini" and see if that makes a difference.

proddy, 2021-06-12:
and if you're not using a KM200 set tx delay to 0

MichaelDvP, 2021-06-13:
The first picture makes me wonder. It shows 'bus connected' with 87 messages received, no errors and no messages sent.
- If there are no more messages received, the bus should show disconnected, there is a timeout for that. 
- Normally we get around 730 messages per minute on the bus (my statistics), which means the bus is online since a few seconds.
- It takes more than a few seconds to register the devices. 

So the pictures shows a normal state for a few seconds after reboot.
Is the emsesp constantly rebooting? What does the uptime show?

stmb-100, 2021-06-14:
Hello it looks like the system does a reboot every 10-20 seconds. How can I activate a debug mode?

stmb-100, 2021-06-14:
[image]

proddy, 2021-06-14:
We need to find what is causing the crashes. Take out the ESP32, connect it via USB to a PC and then open a serial connection to the COM port using ptty or something similar. See the logs come in and see if it still crashes. If not then it's something to do with the EMS data coming in. While still connected to the PC go to the Web UI and disable MQTT, SysLog, NTP just to isolate those from being a possible root cause. Now connect the ESP32 back into the Gateway circuit. If possible keep the USB connected so you can see the log data coming in.

Just to be sure, it worked fine on 3.0 but crashes on 3.1?

MichaelDvP, 2021-06-14:
It seems to me there is something wrong with filesystem. It shows 320k, but we have 64k since 3.0.0b3 and before 192k (afair). 
Does it help to do a factory reset and reconfigure? 
@stmb-100 Do you have experience with flashing via usb/esptool?

stmb-100, 2021-06-14:
yes 3.0 no Problems.

It may be a problem with the power supply. The V 3.1 requires more power. On PC-USB I already have an UpTime of 8 minutes 

[image]

stmb-100, 2021-06-14:
Do you have experience with flashing via usb/esptool?
Yes i flashed with this tool.

MichaelDvP, 2021-06-14:
> Yes i flashed with this tool.

Because of the wrong partiition table i suggest a complete flash with partitioning and erase as described in the wiki:
`esptool.py --chip esp32 --port "COM3" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader_dio_40m.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 <firmware.bin>`
and add `-e` or `--erase-all` option.

stmb-100, 2021-06-14:
I reflash the ESP32
[image]

stmb-100, 2021-06-14:
It's really a power problem. Does not work with USB power supplies up to 5 V 2 A, only with PC-USB. Strange behavior with version 3.1

stmb-100, 2021-06-14:
blob:file:///3f5adb16-67ba-4b4b-abfd-59d52391f730

MichaelDvP, 2021-06-14:
Some usb-chargers are only chargers with large rippe, that needs the battery on the output for smoothing the voltage. They can not be used as power supply. Try another one or use the jack-power.  ESP32 works good with jack-power.

stmb-100, 2021-06-14:
If I take the USB supply from a raspberry Pi, the problem is the same.

MichaelDvP, 2021-06-14:
The soldering on GND and 5V dont look good. Is it a contact issue? 
BTW: the filesystem shows now 192k, have you flashed the 3.0 first via usb and updated OTA to v3.1. The actual partition table should give 64k filesystem and a 64k more flash for code and ota.

stmb-100, 2021-06-14:
[image]

It doesn't work with Jack Power, if I add PC-USB it works again. 
I will re-solder and use a new ESP32 D1 Mini
First I flashed it with the Windows Flasher (ESP-Flasher-Windows-x64)

MichaelDvP, 2021-06-14:
I have a very similar configuration and it works with jackpower (Gateway 1.5, MH-ET32, GB125, RC35, MM10, SM10), i think you should resolder the lower two pins on right side (GND/5V) and try again with jack-power.

MichaelDvP, 2021-06-14:
> First I flashed it with the Windows Flasher (ESP-Flasher-Windows-x64)

Ah, that's why i ask for esptool. Most GUI-flashers do not partition correctly.

stmb-100, 2021-06-14:
I have now plugged the jack cable directly into the heater, without a 10 meter extension and now the connection works via jack power supply with V 3.1 and also with the USB power supply. The fault was the 10 meter extension of the jack connection.

proddy, 2021-06-15:
right, so false alarm. Do you want to close this issue?

bbqkees, 2021-06-18:
Jack cables are usually intended for audio signals, the wires inside are very thin and cheap and may give powering issues when used with the Gateway. You may lose several volts over that 10m cable, and intorduce other interferences in the EMS signal.

The jack cable that is supplied with the Gateway therefore has thicker copper wires inside.

closed 2021-06-29 by proddy