emsesp/EMS-ESP32 #74: IO-Broker MQTT

stmb-100, 2021-06-14:
Does anyone also have the problem that no values are displayed in IO broker. 

Configuration MQTT EMS-ESP
[image]

[image]

[image]

stmb-100, 2021-06-14:
[image]

MichaelDvP, 2021-06-14:
These are command subscriptions only and do not publish values. The values are published as json in the `_data` topics.
I have a [experimental version](https://github.com/emsesp/EMS-ESP32/files/6649049/firmware.zip) that also publishes single values on change, or you can use the the api with the [ioBroker-Adapter from tp1de](https://github.com/tp1de/ioBroker.ems-esp).

proddy, 2021-07-23:
@stmb-100 close this?