emsesp/EMS-ESP32 #75: SO Bus for IN-Z61

stmb-100, 2021-06-15:
Could it be possible to implement an "S0 bus" on 1 or 2 free pins, e.g. for the gas meter pulse pick-up IN-Z61. It could be implemented in a similar way to ArduCounter.

[image]

bbqkees, 2021-06-15:
I think it's easier to just add a second ESP board near the meter with Tasmota, ESPEasy etc.
In most houses the gas boiler is far away from the actual meter so I don't think a lot of people can use it this way.

SpaceTeddy, 2021-06-19:
I use a Tasmota device to get the gas meter counter. works great.

proddy, 2021-07-20:
I think an external method would be the easiest

closed 2021-07-20 by proddy