emsesp/EMS-ESP32 #77: Enums without Zero Option

tp1de, 2021-06-28:
- some enums does not have a known zero option. RC300 building is 1-3 for light, meduim, heavy, in RC35 the same settings are 0-2. Maybe RC300 have a `off` option, but no RC300 user have reported it. This confusing numbering is from the Bosch-telegram, not from emsesp. With the latest changes RC300 will show a empty option `["","light","medium","heavy"]`.

_Originally posted by @MichaelDvP in https://github.com/emsesp/EMS-ESP32/issues/76#issuecomment-864515167_

Are we sure to keep the empty zero options in enum for the future?
If I choose 0 for thermostat/building it is replaced by 2 (medium) in the reply. 

So far so good, but I would feel more save if enum table will be only filled with valid content.

tp1de, 2021-07-09:
Since I am preparing my ioBroker Adapter for V2 API as well, I recognized that values for those enum fields which do not start with zero options are different in V2 vs. V3 API.

In V2 thermostat.building has value 1 for medium 
In V3 thermostat.building has value 2 for medium

As @MichaelDvP explained the V3 values are comming 1:1 from the telegrams.

MichaelDvP, 2021-07-09:
Yes, v2 manipulates the values for publishing, v3 has a different datastructure and stores as read.
[See v2 here](https://github.com/emsesp/EMS-ESP/blob/c56dc7b3286645e81df4be3730c3bf75480294b8/src/devices/thermostat.cpp#L434) and [v3 here](https://github.com/emsesp/EMS-ESP32/blob/59913cdc4b2d23bc91b1b36b367168ed14761c7f/src/devices/thermostat.cpp#L2135)
In v3 the options with no zero value starting with a blank, this are for now: [building for RC300 controllers](https://github.com/emsesp/EMS-ESP32/blob/59913cdc4b2d23bc91b1b36b367168ed14761c7f/src/locale_EN.h#L312), [mode4  for Junkers-Controllers](https://github.com/emsesp/EMS-ESP32/blob/59913cdc4b2d23bc91b1b36b367168ed14761c7f/src/locale_EN.h#L324), [modetype4 for Junkers](https://github.com/emsesp/EMS-ESP32/blob/59913cdc4b2d23bc91b1b36b367168ed14761c7f/src/locale_EN.h#L330), and [collectortype for SM100/200 solarmodule](https://github.com/emsesp/EMS-ESP32/blob/59913cdc4b2d23bc91b1b36b367168ed14761c7f/src/locale_EN.h#L344).

tp1de, 2021-07-09:
> In v3 the options with no zero value starting with a blank

Yes I am aware and within the initialization of the ioBroker Adapter I ignore/skip the blank content for V3.

Now for V2 I was thinking how to implement the missing info for writeable, units, min/max and enums.

I tested to make a http write to each single field to find out if writeable -> The heatingsystem went into error state: Bad idea
For the units I implemented to be "°C" whenever "temp" is within name

Now I think about generating a csv-file with the attributes per device/field with the attributes missing, or maybe just missing it and make each device-field writeable without further attribute.

@proddy @MichaelDvP What would you recommend?

MichaelDvP, 2021-07-09:
I'm thinking of recalcualting the values in telegram-reading to have all enums starting from zero. It's only these mentioned few we have to touch. Output is the same as in v2, but the way is different. It is a breaking change for v3 if using bool-format numbers.
@proddy: my suggestion:
change these values from `telegram->read_value()` to `telegram->read_enumvalue(.., 1)` with:
```
    bool read_enumvalue(uint8_t & value, const uint8_t index, uint8_t start = 0) const {
        if ((index < this->offset) || ((index - this->offset) >= this->message_length)) {
            return false;
        }
        uint8_t val = value;
        value = this->message_data[index - this->offset] - start;
        if (val != value) {
            return true;
        }
        return false;
    }

```

MichaelDvP, 2021-07-09:
>  I tested to make a http write to each single field to find out if writeable -> The heatingsystem went into error state: Bad idea

If you send the name without value in v2 api, you should get [`400-Invalid cmd`](https://github.com/emsesp/EMS-ESP/blob/c56dc7b3286645e81df4be3730c3bf75480294b8/src/WebAPIService.cpp#L56) if not writable, but [`200-Invalid`](https://github.com/emsesp/EMS-ESP/blob/c56dc7b3286645e81df4be3730c3bf75480294b8/src/WebAPIService.cpp#L118) for writable values.

tp1de, 2021-07-09:
> If you send the name without value in v2 api, you should get [`400-Invalid cmd`](https://github.com/emsesp/EMS-ESP/blob/c56dc7b3286645e81df4be3730c3bf75480294b8/src/WebAPIService.cpp#L56) if not [...]

Yes for single commands - that's right. 
Maybe I was too fast testing all fields on initialization. I will try with a delay after each http get. (In V3 I use 100 msec)

MichaelDvP, 2021-07-09:
BTW: Mode setting only works with names, not enum-numbers (v2 and v3). It's because of `thermostat_ha_cmd`, but we can move the alpha-check there and allow numbers for other commands.

proddy, 2021-07-09:
@MichaelDvP I'm out on vacation so ok with any changes you want to make. I'll review when I'm back.

MichaelDvP, 2021-07-11:
In new dev all enums starting from zero. I've also changed the Junkers mode to display the values that are set.
Numbers for setting thermostat-modes do not work in v2 and have not worked in v3, now they should work in v3.

FredericMa, 2021-07-16:
Uptime is now 24h without a reboot. I would consider this as solved since I had at least 5 reboots within 24h before. Thanks!

FredericMa, 2021-07-16:
@MichaelDvP I've checked the data I receive but now I see `nofrost` for both `mode` and `modetype` whereas `mode` was manual and auto before:
```
{"id":106,"datetime":"11:53:03 16/07/2021",
"hc2":{"seltemp":15,"currtemp":23.3,"mode":"nofrost","modetype":"nofrost","heattemp":22.5,"ecotemp":19,"nofrosttemp":15},
"hc1":{"seltemp":15,"currtemp":23.3,"mode":"nofrost","modetype":"nofrost","heattemp":28,"ecotemp":19,"nofrosttemp":15}}
```

MichaelDvP, 2021-07-16:
That's right, Junkers have the mode settings `nofrost/eco/heat/auto` but reports in monitor telegram `manual/auto` with `nofrost/eco/heat` modetype. To synchronize settings with reported value i have to use the complete modelist. Otherwise the api will report for `emsesp.local/api/thermostat/mode` a wrong list and also the selectbox in web shows wrongs settings.
I mainly want to know if the reported modes fits the real settings. Only in auto-mode the modetype gives additional information `eco/heat`. (that's the same for other thermostats like my RC35, having mode `night/day/auto` and modetype `night/day`)

FredericMa, 2021-07-16:
I can confirm that the mode settings and the reported settings match.

tp1de, 2021-07-18:
@MichaelDvP I compared V2 and V3 API responses and I have the following remarks:

- in V3 thermostat.minexttemp is now missing (was available before in V3 and is still available in V2)
- enum is returned in V2 sometimes as text / sometimes als numbers - this is independent of using id=0
e.g. boiler.wwcircpumpmode, thermostat.mode, thermostat.modetype
- fieldname of wwcircmode and wwcircpumpmode differ in V3/V2

MichaelDvP, 2021-07-20:
> in V3 thermostat.minexttemp is now missing 

fixed

> enum is returned in V2 sometimes as text / sometimes als numbers - this is independent of using id=0
e.g. text: boiler.wwcircpumpmode, thermostat.mode, thermostat.modetype

Right, i already mentioned that modes does not work with numbers in v2 and v3. In v3 i changed this in [this commit](https://github.com/emsesp/EMS-ESP32/commit/5f44eb14ada94ecc6775ff4f8dcabe33fa620fb2).

> fieldname of wwcircmode and wwcircpumpmode differ in V3/V2

And a lot more fieldnames. This is from #35

tp1de, 2021-07-20:
@proddy @MichaelDvP 
I would highly appreciate to understand what is your target and change path for API V2 / V3 enum datafields.
The last change (text instead of numbers in V3) was a surprise for me and I got a lot of warnings on my ioBroker-Adapter.

The API interface shall support integration into other home automation systems. 
There is a need to minimize breaking changes and discuss them in advance.

What is your intention now?
Do you intend to return values within device info or text for enum datafields? 
Will the id=0 option to get values in V2 still exist in the future if it will be text?

What about posts in V2 / V3? 
Shall I prepare to always post text (V2 / V3) to be on the safe side or will be values acceptable for the future as well?

MichaelDvP, 2021-07-20:
> The last change (text instead of numbers in V3) 

Can you exactly explane what change do you mean? I did not see such a change.

tp1de, 2021-07-20:
> Can you exactly explane what change do you mean? I did not see such a change.

API V3 get request:   ems-esp/api/thermostat:

b3:  returns numeric values
```
{
  "id": 158,
  "datetime": "10:47:12 20/07/2021",
  "floordry": 0,
  "dampedoutdoortemp": 16.4,
  "floordrytemp": 0,
  "building": 2,
  "wwmode": 4,
  "wwcircmode": 0,
  "wwextra1": 0,
  "hc1": { .............
```

b4: returns text 
```
{
  "id": 158,
  "datetime": "10:39:07 20/07/2021",
  "floordry": "off",
  "dampedoutdoortemp": 16.3,
  "floordrytemp": 0,
  "building": "heavy",
  "minexttemp": -12,
  "wwmode": "own_prog",
  "wwcircmode": "off",
  "wwextra1": 0,
  "hc1": { .......
```

tp1de, 2021-07-20:
Within API V3 I can read enum info of the respective state and transform the text into values and vice versa.
This requires some more read access within the ioBroker Adapter. I have it implemented now on my test system. (read only until now). What about posts and what will the future target?

I just ask myself what kind of coding shall I implement not to have it changing all the time .....

MichaelDvP, 2021-07-20:
Can not reproduce. Setting bool and enum formats to number gives me for mqtt and api:
```
{
  "id": 86,
  "lastcode": "d1(820) 08.07.2021 16:00",
  "datetime": "11:05:33 20/07/2021",
  "intoffset": 0,
  "minexttemp": -11,
  "inttemp1": 29.3,
  "inttemp2": 29.9,
  "damping": 1,
  "dampedoutdoortemp": 19,
  "building": 1,
  "wwmode": 2,
  "wwcircmode": 0,
  "hc2": {
    "seltemp": 0,
    "mode": 2,
    "modetype": 1,
    "daytemp": 22,
    "nighttemp": 17,
    "designtemp": 66,
    "offsettemp": -1,
    "holidaytemp": 15,
    "targetflowtemp": 5,
    "summertemp": 14,
    "summermode": 1,
    "holidaymode": 0,
    "nofrosttemp": 5,
    "roominfluence": 0,
    "minflowtemp": 15,
    "maxflowtemp": 85,
    "flowtempoffset": 0,
    "heatingtype": 1,
    "reducemode": 1,
    "controlmode": 0,
    "control": 2,
    "program": 0,
    "pause": 0,
    "party": 0,
    "tempautotemp": 0,
    "noreducetemp": -20,
    "switchtime": ""
  }
}
```

proddy, 2021-07-20:
I couldn't reproduce either. maybe its a caching issue?

@MichaelDvP please hold off from any pushes - I'm going to commit some changes in a bit.

tp1de, 2021-07-20:
> Can not reproduce. Setting bool and enum formats to number gives me for mqtt and api:

Yes Michael I checked again:  When **re-entering the format options** it works. I just have seen that while uploading b4 yesterday coming from b3 the settings where changed (I haven't recognized that). Not just `Formatting Options` but also `Bypass Access Token authorization on API calls`. I was already wondering why I can't change values anymore ..... and downgrading from b4 to b3 deletes most of the settings !
So I better check all settings after upload of a new firmware version in the future. I will update my ioBroker documentation for this.
BTW: Can I set the Formatting Options by API V3 to secure the right settings?

So values **and** text will be available for get / post in the future in V3 and I do not need to change anything?

Thanks for your support Michael.

proddy, 2021-07-20:
ah, glad it's working. you're using the latest dev build so really living off the edge, which is exciting but painful sometimes! When this release is stable we'll document the (breaking) changes and any auto-migration steps. thanks for your continued patience and testing

tp1de, 2021-07-20:
> thanks for your continued patience and testing

Since I am retired I need a hobby .... No problem for me but sometimes it is very difficult to understand all the changes ...

The ioBroker adapter does not work good with the last stable ESP32 build. So I wrote in the documentation to use the last dev build. Some people are already using the ioBroker adapter. Hopefully they don't update too often.
Do you have any idea, when to release a next stable version for ESP32?

proddy, 2021-07-20:
> Do you have any idea, when to release a next stable version for ESP32?

Soon I hope. We have to finish the IPv6 stuff you requested, plus make sure the Dallas changes with renaming and offsets works for everyone. The roadmap is https://github.com/emsesp/EMS-ESP32/milestones/v3.2. Since it's not a patch release I'll bump it to the next minor 3.2. And we have to update the documentation, especially the API changes.

MichaelDvP, 2021-07-26:
I think we can close this here 
Enums are now all starting from zero and have no blank fields, minexttemp is added, modes can set and read with numbers.

V2 (OT here) is (and i think it stays) unchanged, the values  boiler.wwcircpumpmode, thermostat.mode, thermostat.modetype are not enums and to change is a lot of work. We can open in v2 issues.

closed 2021-07-26 by MichaelDvP