emsesp/EMS-ESP32 #79: Occasionally on a clean install, MQTT subscriptions are not sent out

proddy, 2021-06-30:
As originally reported by Dan on the Discord chat, the MQTT subscriptions are not sent until a setting is changed in the MQTT Settings page. I suspect its the init() call is in the wrong place - needs looking into

---

> Dan Ohnesorg — Yesterday at 23:11
this time I have seen the system+boiler after reboot, definitively didn't work in the last tries, If I change the "Subscribe format" , I even don't need the reboots, set MQTT IP > list is empty > change "Subscribe format" to ".. all heating circuits", list is full, no reboot needed

```
ems-esp:$ show mqtt
MQTT is connected
MQTT publish fails count: 0

MQTT topic subscriptions:

MQTT queue is empty

000+00:03:05.229 I 0: [mqtt] MQTT disconnected: TCP
000+00:03:05.324 I 1: [mqtt] MQTT connected
ems-esp:$ show mqtt
MQTT is connected
MQTT publish fails count: 0

MQTT topic subscriptions:
 ems-esp/system/send
 ems-esp/system/publish
 ems-esp/system/fetch
 ems-esp/boiler/wwtapactivated
 ems-esp/boiler/reset
 ems-esp/boiler/selflowtemp
 ems-esp/boiler/selburnpow
 ems-esp/boiler/heatingactivated
 ems-esp/boiler/heatingtemp
 ems-esp/boiler/pumpmodmax
in time 00:03:05.229 is changed the "Subscribe format" setting
```

MichaelDvP, 2021-06-30:
I think this is by design. If mqtt is disabled there is no subscription on registering commands and no registration of subscription (mqtt.cpp line 60, register in line 88).

After enable mqtt there has to be a reboot to make the subscriptions.

proddy, 2021-06-30:
that would explain it. Let's see what Dan says. Maybe needs a call to `resubscribe()` when the "MQTT Enabled" is enabled in `Mqtt::start()`

MichaelDvP, 2021-06-30:
`resubscribe` is called on every change, but that does not help, it only resuscribes to registered topics, but the topics are not registered. You have to call `mqtt_subfunctions_.emplace_back(device_type, std::move(topic), std::move(cb));` also if mqtt is disabled and only skip the `queue_subscribe`.
I think this is from 8266 to save all possible memory if mqtt is disabled.

MichaelDvP, 2021-06-30:
I've made a small change that should fix it. Or is there something else?

The issue as i understand and tested: Boot emsesp with mqtt disabled, than enable mqtt -> no subscriptions (only after reboot)
Now subscriptions are stored also if mqtt is disabled and on connect the `resubscribe` should send out.
Every change in mqtt parameters results in `mqtt_reset` and `resubscribe`.

proddy, 2021-07-01:
fixed by Michael. thanks

closed 2021-07-01 by proddy