emsesp/EMS-ESP32 #80: Version": "3.1.2b0": ERROR [telegram] Rx: 0B 88 19 00 20 D0 FF (CRC FF != 69)...

zibous, 2021-06-30:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*
ERROR [telegram]....

**Steps to reproduce**
*Steps to reproduce the behavior.*
Show system log

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*
```json
{
  "System": {
    "version": "3.1.2b0",
    "uptime": "000+00:00:11.459",
    "freemem": 189
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "#telegrams received": 22,
    "#read requests sent": 8,
    "#write requests sent": 0,
    "#incomplete telegrams": 4,
    "#tx fails": 0,
    "rx line quality": 82,
    "tx line quality": 100,
    "#MQTT publishes": 63,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas reads": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)",
      "handlers": ""
    }
  ]
}
```
**Additional context**
*Add any other context about the problem here.*
```log
000+00:02:00.442   ERROR [telegram] Rx: 0B 88 19 00 20 D0 FF (CRC FF != 69)
000+00:02:01.054   ERROR [telegram] Rx: 0B 88 19 00 20 D0 FF (CRC FF != 69)
000+00:02:02.339   ERROR [telegram] Rx: 0B 88 16 00 20 EC FF (CRC FF != 2D)
000+00:02:04.269   ERROR [telegram] Rx: 0B 90 FF 00 20 01 A5 19 FF (CRC FF != 2B)
000+00:02:07.153   ERROR [telegram] Rx: 0B 90 FF 00 20 01 AF 13 FF (CRC FF != 35)
000+00:02:09.103   ERROR [telegram] Rx: 0B 90 FF 00 20 01 9B 27 FF (CRC FF != 69)
```

MichaelDvP, 2021-06-30:
Funny, it's the same as in #78, only the echo gives crc-errors, but sending is without error. In this case there is a additional FF after the CRC received. I think it's tx-mode 2 (ems+) timing that sending the break is influencing the receiver.

@proddy: a while ago i've added `emsesp.cpp` [line 1072](https://github.com/emsesp/EMS-ESP32/blob/c634c3987412b13b9e3c0610550f321262a15547/src/emsesp.cpp#L1072), just to see the echo in the log. It has no other function. 
If it produces irritation and false errors on ems+ systems maybe it's better to remove the line.

proddy, 2021-06-30:
CRC errors are normal, so I think we should also declassify them to WARNINGS (or DEBUG) and not Errors. It can be scary to view the System Log for the first time and see a whole page of errors. I get around 0.2% errors (110 from 50K received EMS telegrams) probably related to the quality of the wires.

I'll look at the uart code too....

tp1de, 2021-07-01:
I tried with tx mode 1. The number of crc errors are the same as with ems+. in my case 0.8%.

MichaelDvP, 2021-07-01:
I've made a change to dev: Tx-Echos with wrong crc will only give a trace-logmessage, Tx-fails and all other Rx-crc-errors are counted as before, but crc are logged as warning. It doesn't skip any information, but the false negatives are not shown as error.

tp1de, 2021-07-01:
To share my findings in respect to crc errors (mqtt: rxfails / api v3: #incomplete telegrams):

- Heating system: KB192i with IP-inside (KM200), MM100, RC310 and 2 Heating Circuits (Radiators / Floorheating)
- Approx 200 messages / telegrams per minute received - **are all counted even unknown telegram types?**

I checked different wiring and connection types of the ems-gateway to ems-bus without last change from @MichaelDvP :

1. Existing installation by connecting to ems-bus-out of mm100 module with 2m thin (0,1 mm2) cable (slightly twisted) 
2. Connecting with 10cm 2x0.75 mm2 cable to mm100 module
3. Connecting with BBQKees service jack cable to the boiler service jack

crc errors compared to rxrecieved  where:

1. 0.8% 
2. cables where twisted: 1.4% ( I was astonished)
   cable not twisted: 0.8 %
3. 0.8%

So my conclusion: Cable should not be twisted but connection point is not so critical and #crc errors might depend more on #telegrams / minute (collisions on ems-bus).

The polling of KM200 (120 fields every 90 secs) with http gets does not increase the #telegrams received.
I thought that every get request to KM200 should result in 2 ems-bus telegrams. (KM200 request to device + device sending data). But these telegrams are maybe not counted by ems-esp but certainly increase bus-traffic.
  
I tried with/without KM200 polling. Percentage of crc-errors stays constant as well. 

BTW: I tested different bus-id's - There is no influence on #incomplete telegrams.

tp1de, 2021-07-01:
> @MichaelDvP 
> 
> I've made a change to dev: Tx-Echos with wrong crc will only give a trace-logmessage, Tx-fails and all other Rx-crc-errors are counted as before, but crc are logged as warning. It doesn't skip any information, but the false negatives are not shown as error.

I just installed this version. No i do not see any mqtt rxfails / api #incomplete telegrams  anymore.
And in web-ui I do not see any crc messages (debug mode). **Is this right?**
What are trace-logmessages and where should I see them?

MichaelDvP, 2021-07-02:
> Maybe it would be good to have an indicator about crc percentage on rxfails/rxreceived to understand ems-bus quality.

This indicator is untouched, you have rx line quality and # incomplete telegrams  in `show ems` , also tx-quality and tx-fails, counting if we miss something. Also in log levels warning and above you'll get the message about crc if there is one in a telegram we want to receive.

The echo-messages are not processed in any way, a wrong received crc in such a message does not do anything and is no indicator for line quality. All incomming messages we want to process are checked and wrong crc is counted in # inclomplete telegrams. When sending we check for reply and `ack` from the adressed ems-client and will count errors in tx-fails. If a sended message really has a wrong crc than we will get a tx-error.

tp1de, 2021-07-02:
@MichaelDvP thanks for clarifying. 

At the end your explanation shows that the counting of the echo message crc errors gave a wrong interpretation about ems-bus line quality. My status is now within updated firmware of ems-esp:
[image]

BTW: What is an echo message?

proddy, 2021-07-02:
Also note that the Rx quality has a threshold of 5% (EMS_BUS_QUALITY_RX_THRESHOLD in telegram.h) so you'll only see it not be 100% when more than 5% of the incoming Rx telegrams are incomplete. This is under the assumption that there will always be a few bad telegrams here and there and if people buy a BBQKees board are start seeing 97% within the first few minutes they'll freak out and think the board is broken.

Maybe we should lower it to 3%?

MichaelDvP, 2021-07-02:
> BTW: What is an echo message?

I wanted to link [here](https://emswiki.thefischer.net/doku.php?id=wiki:ems:ems-bus), but it seems to be broken.

EMS-Bus works with 2 lines full duplex by using different modulations for sending receiving. 
The master sends messages in volatage modulation and the clients receive them. The clients sends in current modulation and only the master receives them. If a client sends something, the master receives the current modulation and echos the message byte for byte in voltage modulation, so all clients can receive. 

e.g. if we are requsting version from thermostat: 
```
ems-esp2:# read 10 2
000+00:21:22.114 N 3: [telegram] Rx: 0B 90 02 00 20 7C
000+00:21:22.243 N 4: [telegram] Rx: 10 0B 02 00 56 01 12 6C
```
- we are sending `0B 90 02 00 20 7C` in current modulation to the master
- the master echos in voltage modulation `0B 90 02 00 20 7C`
- the thermostat answers in current modulation `10 0B 02 00 56 01 12 6C`
- the master echos in voltage-modulation `10 0B 02 00 56 01 12 6C`
- we are receiving `10 0B 02 00 56 01 12 6C`

Because the master starts echoing after the first byte we are sending, localy the receiving may be disturbed by drawing current for sending from the bus and making a volage drop on the receiver.  
But if we are receiving the answer from the thermostat we know that communication was successful, so we don't have to check the echo to our sending. Also if we are sending a value to the thermostat the thermostat will send an `ack` back and we know that communication was successful.

Before my commit on 10.May the echo was ignored, the N3-message was not logged. But i want to see the timings for sending and receiving the answer in the log.

closed 2021-07-02 by proddy