emsesp/EMS-ESP32 #82: show real NPT time (if NTP service active) in Web's System Log

proddy, 2021-07-04:
Show real-time in the log, if possible, otherwise fallback to uptime

MichaelDvP, 2021-07-05:
That's a bit tricky, because we want to log the message time in ms, not the time we publish it. I tested with a simple offset to uptime, set once if NTP is enabled:
```
void WebLogService::operator<<(std::shared_ptr<uuid::log::Message> message) {
    if (log_messages_.size() >= maximum_log_messages_) {
        log_messages_.pop_front();
    }
    log_messages_.emplace_back(log_message_id_++, std::move(message));
    EMSESP::esp8266React.getNTPSettingsService()->read([&](NTPSettings & settings) {
        if (!settings.enabled || (time(nullptr) < 1500000000UL)) {
            time_offset_ = 0;
        } else if (!time_offset_) {
            time_offset_ = time(nullptr) - uuid::get_uptime_sec();
        }
    });
}

char * WebLogService::messagetime(char * out, const uint64_t t) {
    if (!time_offset_) {
        strcpy(out, uuid::log::format_timestamp_ms(t, 3).c_str());
    } else {
        time_t t1 = time_offset_ + t / 1000ULL;
        strftime(out, 25, "%F %T", localtime(&t1));
        snprintf_P(out, 25, "%s.%03d", out, (uint16_t)(t % 1000));
    }
    return out;
}
```
and call:
```
    char                time_string[25];
    logEvent["t"] = messagetime(time_string, message.content_->uptime_ms);
```
looks ok to me. If you don't have other ideas, i can merge to dev.

proddy, 2021-07-05:
looks great, check it in!

tp1de, 2021-07-06:
I just implemented. I just wonder why time is 3-4 minutes ahead of NTP-time?

[image]

tp1de, 2021-07-06:
I did another restart and now time is actual time (no difference anymore).

It's good to have real time within this log to compare with other logs like on my Fritzbox.

proddy, 2021-07-19:
think we can close this one.

closed 2021-07-19 by proddy