emsesp/EMS-ESP32 #83: Enable Network for IPV6

tp1de, 2021-07-05:
Since I watch WiFi stability I recognized that IPV6 is not enabled (like on my km200 too).

Most of my WLAN/LAN Devices support IPV6 and I recommend to enable on EMS-ESP as well.

proddy, 2021-07-05:
I'm not sure it will help with WiFi stability, since EMS-ESP is only a WiFi client. I could make the Web Server and Telnet Server respond to IPV6 but this is just peer-to-peer.  Can you give some examples of your other devices that support IPv6 so I can take a look?

MichaelDvP, 2021-07-05:
I've tested to enable IPv6 for dynamic IP, adding to `WebStatusService.cpp` line 39:
```
    case SYSTEM_EVENT_STA_CONNECTED:
        WiFi.enableIpV6();
        break;
    case SYSTEM_EVENT_ETH_CONNECTED:
        ETH.enableIpV6();
        break;
    case SYSTEM_EVENT_GOT_IP6:
#ifndef EMSESP_STANDALONE
        EMSESP::logger().info(F("WiFi Connected with IP=%s, hostname=%s"), WiFi.localIPv6().toString().c_str(), WiFi.getHostname());
#endif
        break;
```
Using the above addition i got the ipv6, can ping it, but can not connect to Webpage and telnet via IPv6.
AsyncTCP does not handle it for now. There is a [PR](https://github.com/me-no-dev/AsyncTCP/pull/105), but not merged.

proddy, 2021-07-05:
We use local copies of the libraries so I updated AsyncTCP with https://github.com/husarnet/AsyncTCP/tree/master/src and added your changes. Seems to work with WiFi. The Eth causes a crash which needs looking into

proddy, 2021-07-05:
found the reason for the crash and made a small change

tp1de, 2021-07-05:
Seems to work. Good job !

[image]

tp1de, 2021-07-06:
@proddy @MichaelDvP 

I tested a bit more:

1.  DHCP vor IPv6 is working
2.  Web-ui is available with IPV6 address
3.  API V3 is working with IPV6
4.  my ioBroker Adaper works with IPV6 address for EMS-ESP
5.  **Not working:** Enter IPV6 addresses for Syslog-Server or MQTT-Broker (not a valid ip address)

I think 5 is a check you implemented and can be easily adjusted. The rest is working.

proddy, 2021-07-07:
> Not working: Enter IPV6 addresses for Syslog-Server or MQTT-Broker (not a valid ip address)

The fix here is to add an additional form field validator like `interface/src/validators/isIP.ts` to allow 8 groups of hextets (e.g. 2001:0db8:85a3:0000:0000:8a2e:0370:7334) or a range.

MichaelDvP, 2021-07-07:
The regexp is one fix (it should allow also shorteners like FE00::1), but mqtt and syslog does not handle ipv6 for now.
Mqtt uses AsyncTCP, it's only a small addition, i'm currently testing a modification.
Syslog seems to have some defines (ARP_CHECK, NDP_CHECK, LWIP_xx) to handle ipv6, but i do not understand for now. Also `destination(host, port)` is uses only IPv4. 

BTW: @proddy What do you think about adding the WiFi-tweaks (IPv6, Bandwidth, Tx-power to Network settings? Maybe some other people want to use it without recompiling.
[image]

proddy, 2021-07-07:
> BTW: @proddy What do you think about adding the WiFi-tweaks (IPv6, Bandwidth, Tx-power) to Network settings? Maybe some other people want to use it without recompiling.

@MichaelDvP good idea. I would call it "WiFi low power mode"

tp1de, 2021-07-07:
Excellent idea to have selectable parameters since I was not able yet to compile for use with 20Mhz bandwidth.

proddy, 2021-07-07:
it does more than just the HT20, it lowers the tx transmit power, disables sleep mode and turns off Bluetooth. basically, it lowers the power consumption of the whole gateway board (BBQKees and I have been doing some benchmarking)

tp1de, 2021-07-07:
That's good to hear. The 40MHz bandwidth ( max 150 Mbit/s) is not needed at all. So why not making it standard?

MichaelDvP, 2021-07-07:
I would like to split the tweaks in single options, because each option can be usefull, but not always the combination.
- WiFi Tx-power as shown with dBm setting, i use 0-20 dBm, not full resolution, no negatives.
- WiFi Low Bandwidth
- Disable WiFi Sleepmode, afair this only for some Linksys-routers and meshes. I think better as an extra checkbox.

Bluetooth disable is currently not in the [tweaks](https://github.com/emsesp/EMS-ESP32/blob/3ea53a8012ecaa010c522068c7ce8cab77fbdf7e/src/system.cpp#L228), it is [here](https://github.com/emsesp/EMS-ESP32/blob/3ea53a8012ecaa010c522068c7ce8cab77fbdf7e/src/system.cpp#L303), but not used.

proddy, 2021-07-07:
Then maybe have an advanced section since there are few options here. we can add Bluetooth and the 160/240Mhz clock speed to be complete.

MichaelDvP, 2021-07-07:
Clock-setting better fits to ADC, LED and board-profile in the emsesp-settings, the others are network options.
Changing clock resets all connections like reboot, it should be a setting only get active on next reboot.

Bluetooth is a bit strange, calling `esp_bt_controller_disable();` will alloc ~30kB of heap, seems the controller first loads a complete bt-stack to heap and then disables the controller. Also with `esp_bt_mem_release(ESP_BT_MODE_BTDM);` the heap stays low. I dont ike to add such a option if it has side-effects not fully understand.

proddy, 2021-07-08:
We don't need to add clock settings, I'm quite happy with the stock 240Mhz. It's only since you mentioned it. So up to you if you want to add.
I didn't realize the BT added some much heap, let's remove that.
So add the Tx Power and Bandwidth as options in the Network Settings page?

tp1de, 2021-07-08:
Has anybody tested if reducing ESP32 power consumption to minimum by switching off bluetooth, reducing tx power and bandwith and maybe reducing to 160 MHz there is a chance to get the gateway ems-bus powered again? (without extrnal usb or jack power)

This might be an argument

proddy, 2021-07-08:
> Has anybody tested if reducing ESP32 power consumption to minimum by switching off bluetooth, reducing tx power and bandwith and maybe reducing to 160 MHz there is a chance to get the gateway ems-bus powered again? (without external usb or jack power)

that was what me and Kees were working on a few months ago, and then stopped. I'll need to dig up my old notes.

MichaelDvP, 2021-07-08:
I've pushed some changes, 160MHz works only after reboot, i've added it to `settings/options`, the other options to `network`, for now without a extra section. Mqtt works for me with ipv4 and ipv6 now. (Needs some tests with inconsistant settings like IPv6 enabled, but only v4 provided).

The tweak-function in not called now, but i leave it in the code if you want to go back. Disable bt-power is not important, i think it is very low power if not initialized and not sending. There should be some registers to depower the module, but i havn't found them yet.

For AP i have set fixed to low bandwidth, it is more stable and compatible, i think an option is not needed.

tp1de, 2021-07-08:
@MichaelDvP I just uploaded the bin file but I can't find the options and IPV6 is not enabled anymore by startup. 

Do I need to compile?

bbqkees, 2021-07-08:
> 
> 
> Has anybody tested if reducing ESP32 power consumption to minimum by switching off bluetooth, reducing tx power and bandwith and maybe reducing to 160 MHz there is a chance to get the gateway ems-bus powered again? (without external usb or jack power)
> 
> This might be an argument

I did a lot of testing but with those settings on the NodeMCU 32S board in the Gateway S32 it still uses way too much power for reliable bus powering.
I worked on another design for a while with another chip and that seemed promising.
However, the latest range of EMS thermostats like the TC200 and the RC310 uses a lot of power already, so adding another bus-powered device will start giving bus errors for many users. So I decided to just stop working on bus powered modes and add a power supply. That will always work for everyone and gives the most stable bus connection.

tp1de, 2021-07-08:
> I did a lot of testing but with those settings on the NodeMCU 32S board in the Gateway S32 it still uses way too much power for reliable bus powering.
> I worked on another design for a while with [...]

Thanks for the reply Kees. The settings are working for me and I will watch if with 20Mhz bandwidth I will get the rebooting issues after some hours as well.

tp1de, 2021-07-08:
@bbqkees

Just a question. For testing the ioBroker Adapeter with API V2 as well I connected 2 gatewayboards.
One jack powered ESP32 while watching usb-console with laptop and the ESP8266 USB-powerded connected to MM100 Mixer EMS-Bus out.

Both are working fine and have different bus id's, but I recognized that with 2 gateways I get 1-3 "incomplete telegrams" (crc errors) per minute on the bus. Is this "normal" or can it be avoided.

MichaelDvP, 2021-07-08:
> I didn't realize the BT added some much heap, let's remove that.

I found another way to disable BT that does not affect memory. Don't know if really saves power.

proddy, 2021-07-18:
coming back to this one, all we need to do is 
- fix the web to allow ipv6 addresses. use a custom validator with a new regex allowing all combinations
- get syslog service working with ipv6. it should be possible, just need to dive into the code again
- put in a PR to AsyncMQTTClient to handle ipv6

shall I start on these Michael?

MichaelDvP, 2021-07-19:
> shall I start on these Michael?

Yes, the regex now should allow all combinations, but some more that are invalid (more than two `:` in row, more than one `::`).

proddy, 2021-07-19:
added a new regex for IPv4 and IPv6 and tested using https://regex101.com/r/yyslXu/3

proddy, 2021-07-21:
MQTT with IPv6 works too. Tested with Mosquitto 2.0.11 (had to set IPv6 in the config first). The catch-ya is that the IPv6 host needs to be written as IPv6 expanded (8 hex XXXX:).

MichaelDvP, 2021-07-21:
> MQTT with IPv6 works too.

Sure, since[ this commit.](https://github.com/emsesp/EMS-ESP32/commit/2d7449aebac926282311438de2ff85ebc0085b20)
As mentioned mqtt uses AsyncTCP and it was only a little addition. But syslog does not work with IPv6. In the code there are options for v6, but i did not understand how to trigger it.

proddy, 2021-07-21:
Correct

proddy, 2021-07-21:
I played around a little with the syslog library but it's going be hard to get it to work with the ESP32 because of the UDP library. I saw just leave it - I doubt anyone will use it anyway.

MichaelDvP, 2021-07-26:
Yes, UDP and syslog library for esp32 does not work with IPv6 (confirmed by syslog author). I changed the the settings-input to allow only IPv4 or a hostname. For the hostname i need to change a bit in lib and starting procedure. The name can only be resolved after connection, but we want to start syslog before and queue the messages.

proddy, 2021-07-29:
@MichaelDvP any more additions to this for v3.2 before I release it?

closed 2021-08-01 by proddy