emsesp/EMS-ESP32 #84: Feature Request: Description for Dallas sensors

alexandergeiler, 2021-07-17:
I have 9 Dallas sensors connected to my EMS-ESP.
It is hard to distinguish them because you only see the ID and it is hard to remember IDs.

I would like to have a description field for each sensor (see screenshot)
[image]
This will make it much easier to identify which sensor is taking which readings

I would be great to have the description also available at MQTT (see screenshot)
[image]

MichaelDvP, 2021-07-18:
I've tested a bit, a suggestion for setting the names from the web is this:
[image]
[image]

proddy, 2021-07-18:
I was thinking about this during my holiday.  Instead of showing the Sensor #, ID and Temperature like we have now with : 

[image]

We rename "ID" to "Name". Then on the left side of the row add the pencil icon which allows you to rename the "Name". The name is defaulted to that long ID.

MichaelDvP, 2021-07-18:
Ok, i've used a button around the number, but pencil is better to remove for non-admin. 
As you see i also added the analog (if activated).
I can push, or push a feature-build after testing the last changes.

MichaelDvP, 2021-07-18:
[image]

MichaelDvP, 2021-07-18:
I've only done some short tests, all seems to work. I've pushed to dev, some more testers with many dallas are welcome.
Web forces to names, the setting ìd/no/name` is for api/mqtt only. Mqtt does not like spaces in names, i use the space to seperate name and offset. Console command `sensorname` can  be called with id (any) or sensornumber (sensor has to be connected). 

I like to automatic reload values after setting a value or name, but did not get managed it for now.

proddy, 2021-07-18:
really excellent work, love it. We could enhance the dialog box to not allow spaces in the names, and have the offset as a separate field. If you want any help with coding let me know. I'm self-isolating for the next 7 days so may have odd bits of spare time

proddy, 2021-07-20:
> @MichaelDvP really excellent work, love it. We could enhance the dialog box to not allow spaces in the names, and have the offset as a separate field. If you want any help with coding let me know. I'm self-isolating for the next 7 days so may have odd bits of spare time

I'm making these changes and doing a bit of refactoring along the way

proddy, 2021-07-20:
> I'm making these changes and doing a bit of refactoring along the way

finished with the changes. The edit screen has a small update. most of the changes were internal

MichaelDvP, 2021-07-20:
> finished with the changes.

Can you manage it to reload the values after setting.  We see the snackbar saying that the name is changed, but need to hit reload to see the new name in the list, that's annoying (same for the device-values.).

MichaelDvP, 2021-07-20:
BTW: We can change the `<OutlinedInput` to 
```
            <TextValidator
              validators={['matchRegexp:^([a-zA-Z0-9_.-]{0,19})$']}
              errorMessages={['Not a valid sensorname']}
```
to make sure that there are not spaces and the name is not to long.

proddy, 2021-07-21:
ok adding it now

MichaelDvP, 2021-07-22:
I think it's now fully working. @alexandergeiler please check if you have any suggestions for improvment. 
If not we can close this issue.

alexandergeiler, 2021-07-23:
@MichaelDvP , looks great.
I would have added an alias instead of storing that in the ID field.
But I am fine with that too.
Thank´s a lot!

Regards,
Alexander

closed 2021-07-27 by proddy