emsesp/EMS-ESP32 #86: Does changing hysteresis of boiler works for anybody on EMS+ / RC310?

tp1de, 2021-07-19:
Changing of boiler hysteresis as no effect on my system with EMS+, RC310 and KB192i.

The values can be changed (WEB-UI or API) and stay as changed but have no effect on boiler switch-on and -off temperatures.
Does this work for anybody?

proddy, 2021-07-23:
I don't have an RC3x0 so can't test. Perhaps ask @sq4bja, @bartplessers or @MichaelDvP ? It could be the value is being overwritten by another device

MichaelDvP, 2021-07-23:
I have RC35 with ems1.0 commands. I've checked the RC300 command and it writes to correct telegram/position. But on ems+ the value is maybe write protected and needs a change via the F7/F9 types. If someone with RC300 log the change of the values we can see how it works.

tp1de, 2021-07-23:
Read and write of hysteresis work technically. This means I can update the values and they will stay on these values. Just that these values have no impact at all on working boiler hysteresis.

On the RC310 itself hysteresis can not be set. So I can't debug the commands.

MichaelDvP, 2021-07-24:
Ah, i was confused by RCxxx, it is the boiler telegram 0xE6, the parameters came from [here](https://github.com/Th3M3/buderus_ems-wiki/blob/master/Einstellungen%20des%20Regelger%C3%A4ts%20MC110.md#typ-0xe6-einstellungen-heizung). 

My boiler uses 0x16 telegram and that worked for me. Last winter i've played a bit with boiler hyteresis and thermostat flow temp offset to see the changes and if i can reduce burner starts. Hysteresis is now set to -6/+10. 

I think most people have there heating now in summermode and can not check hyteresis.

tp1de, 2021-07-24:
> if i can reduce burner starts. Hysteresis is now set to -6/+10

That's what I wanted to do as well, but hysteresis stayed at -6/+6 whatever I entered

> I think most people have there heating now in summermode and can not check hyteresis.

Same for me, so I will wait until October ....

MichaelDvP, 2021-07-24:
Maybe this setting only works on unmodulated burners (like mine). Afaik KB192 is modulated and should reduce pwer on the given temperature instead of switching inside a hysteresis.

proddy, 2021-11-10:
@tp1de did it work?

tp1de, 2021-11-10:
I had a discussion with a Buderus technician today about extending my KB192i System into a hybrid System with heating pump (KBH 192i). I asked during this talk about (hidden) possibilities to extend boiler hysteresis. He confirmed that hysteresis is fixed to +- 6°C and can't be changed. Therefore hysteresis parameters are not working for KB192i / GB192i or GB212 and similar boilers.

proddy, 2021-11-10:
shall we document and close this issue? I'm doing some tidying up in prep for the next release. Then I can concentrate on v3.4 which will have an improved webUI and German translation

tp1de, 2021-11-10:
Best would be to have hysteresis data fields only available for unmodulated boilers (if possible).
If this is too difficult is should be mentioned in the description of the hysteresis parameters.