emsesp/EMS-ESP32 #87: EMS-ESP restart via http

SpaceTeddy, 2021-07-20:
sometimes EMS-ESP does not reconnect MQTT, for ever reason. Rest API is still available and I would like to restart and/or reconnect ems-esp via http. Is it possible to do that and if yes which commands I have to send. Unfortunately I did not find a restart command for the RestApi.

thanks a lot 

Chris

proddy, 2021-07-20:
Which version are you on? Could it be related to https://github.com/emsesp/EMS-ESP32/issues/42 ? If you're still having re-connect issues then we should look into fixing the root cause. Let me know how often it occurs. I do have a newer MQTT library we can try out.

For restarting we could add a restart command (endpoint). For now you can also use

```
http://ems-esp/rest/restart\?access_token\={ACCESS_TOKEN}
```
or
```sh
$ curl -i -X POST http://ems-esp/rest/restart\?access_token\={ACCESS_TOKEN}
or
$ curl -i -H "Authorization: Bearer {ACCESS_TOKEN}" -X POST http://ems-esp/
rest/restart
```

SpaceTeddy, 2021-07-20:
I have a dev version from 4-5 days ago. I think that the ems-esp lost mqtt every 5-10 days. A simple restart of the device helps. Therefore I check the latest mqtt heartbeat timestamp in "fhem" and if it is longer that i.E. 120sec. ago, then I want to restart or reconnect the ems-esp.

> http://ems-esp/rest/restart\?access_token\={ACCESS_TOKEN}

great, I going to try it .) 

thanks a lot for your quick feedback... as always !

SpaceTeddy, 2021-07-20:
mhh, I tried the http command with generated Admin Token from the ems-esp security site. (key symbol) unfortunately it does not restart :( 
any idea?

proddy, 2021-07-20:
I remember now, for security reasons you can't invoke it via a browser. you need to POST it from the command line or a script. I'll see if I can implement an official restart function for you today

SpaceTeddy, 2021-07-20:
ok, no need to hurry. 

> curl -i -X POST http://ems-esp/rest/restart\?access_token\={ACCESS_TOKEN}

command works fine!

proddy, 2021-07-20:
added a `restart` command, via GET so works on the browser's URL too. Remember to provide the access bearer token.

closed 2021-07-20 by proddy

SpaceTeddy, 2021-07-21:
Hi proddy,
sry to ask again, but I updated the latest b6 and checked the REST command in the browser again.
>http://192.168.0.4/rest/restart\?access_token\="xxxxxxxxxx...."

but unfortunately no restart. Did I made anything wrong?

proddy, 2021-07-21:
yes I tested again it doesn't work from the browser - this is for security reasons. You'll need to script it or add it as an MQTT command. I'm going to try out a new MQTT lib so hopefully you'll never have to restart EMS-ESP manually!

SpaceTeddy, 2021-07-21:
no problem, I'm going to use the curl command.