emsesp/EMS-ESP32 #88: Extending ADC support

proddy, 2021-07-22:
Currently, ADC is either enabled/disabled and hardcoded to GPIO36 (ADC1 CH0) in the code. Should we make this a little nicer, and have the pin become an option for a channel (0 to 7) like with with the LED, Dallas, Button etc..?

Also how about adding some more refinement like setting the attenuation 0, 2.5, 6 or 11 dB.

I don't use this function so can't determine if it's useful for others.

I was looking at esphome and how they did this? https://github.com/esphome/esphome/pull/2024/files

@MichaelDvP  ?

bbqkees, 2021-07-23:
I thought it was something specifically built in by/for @MichaelDvP. Not sure if anyone else uses it at all.

MichaelDvP, 2021-07-23:
I've tested a bit with raw reads and attentuation. It gave horrible results, large differences in measured value for the different settings. Seems that adc, reference and attentuator have series spread. Chips from 2018 on have a calibration field, using this it's much better.   [(see here)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html).

I think only changing `analogRead()` to `analogReadMilliVolts()` will do, The arduino function [checks if there is a calibration](https://github.com/espressif/arduino-esp32/blob/6e47e18a1b6f18d5f0b9b21eba2a202ee30601ef/cores/esp32/esp32-hal-adc.c#L179). With external input of 1166 mV i get readings from 1164 to 1169 mV for the different attentuators, i think there no real need to set this and stay with default 11dB (3.9V). If someone wants it we can change the checkbox `enable adc` to a selectionbox `off/1.1V/2.2V/3.9V`, i think dB-values are to techie.

Making a port selection also is not really needed, the led/dallas settings were to adapt to different boards with builtin led and avoid conflicts with eth, since board-profiles the individual settings are rarly needed. The analog ports are unused and the default pins are free.

To have more options maybe we can add other inputs added to sensor values, like reading a digital input, hallsensor, [counter](https://github.com/espressif/esp-idf/tree/166c30e/examples/peripherals/pcnt/pulse_count_event) or [rotary encoder](https://github.com/espressif/esp-idf/tree/166c30e/examples/peripherals/pcnt/rotary_encoder).
Also some outputs to set: settings the DAC(io25), setting a pwm.

(if adding hall we need to move adc to io37, since hall uses adc1_0(io36)  and adc1_3(io39) internally.)

proddy, 2021-07-23:
Ok, this is good info to keep in this issue for future reference. I suggest keep the latest change and we can close this issue, and pick it up later if needed. I was just curious after reading about the ESP32 latest libs experiencing issues when reading the voltages (https://github.com/espressif/arduino-esp32/issues/5041)

MichaelDvP, 2021-07-24:
I did not know the arduino issue, but luckily tested with `analogSetAttenuation`, which is not affected.
I think you can close this issue.