emsesp/EMS-ESP32 #89: Bosch Bosch GC7000iW,  telegram errors and wwactivated not responding

elRadix, 2021-07-24:
**Bug description**
My Bosch GC7000iW 35 C 23, is not setting warm active to on
directly it responds with an error 

**Steps to reproduce**
logon to telnet and calling the command 
call boiler wwactivated on

or through the web interface the state of wwactivated stays off

solution is to physically to boiler and manually set the state back on ON

**Expected behavior**
set wwactivated to ON

**Screenshots**
[image]

additional telegram errors 
`(the last 30 messages are buffered and new log events are shown in real time)

000+13:22:02.862   ERROR [telegram] Rx: 8B 88 16 19 20 16 88 0B 16 19 00 1E 00 CD (CRC CD != F1)
000+13:22:55.373    INFO [telnet] New connection from [192.168.1.3]:60382 accepted
000+13:22:55.373    INFO  [shell] Allocated console pty1 for connection from [192.168.1.3]:60382
000+13:22:55.466    INFO [telnet] Connection from [192.168.1.3]:60092 closed
000+13:22:55.466    INFO  [shell] Shutdown console pty0 for connection from [192.168.1.3]:60092
000+13:23:02.519   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:23:03.927   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:23:54.994  NOTICE  [shell] su session opened on console
000+13:24:02.389   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:24:57.640    INFO [telnet] Connection from [192.168.1.3]:60382 closed
000+13:24:57.640    INFO  [shell] Shutdown console pty1 for connection from [192.168.1.3]:60382
000+13:25:02.578   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:26:00.894   ERROR [telegram] Rx: 8B 88 14 00 20 2C 88 0B 14 00 00 73 FD 02 (CRC 02 != 7A)
000+13:27:03.275   ERROR [telegram] Rx: 8B 88 16 19 20 16 88 0B 16 19 00 1E 00 CD (CRC CD != F1)
000+13:27:03.835   ERROR [telegram] Rx: 8B 88 16 19 20 16 88 0B 16 19 00 1E 00 CD (CRC CD != F1)
000+13:27:04.333   ERROR [telegram] Rx: 8B 88 16 19 20 16 88 0B 16 19 00 1E 00 CD (CRC CD != F1)
000+13:27:05.369   ERROR [telegram] Rx: 8B 88 26 00 20 E4 88 0B 26 00 01 05 00 0D 00 1E 52 64 C3 (CRC C3 != B4)
000+13:28:00.873   ERROR [telegram] Rx: 8B 88 14 00 20 2C 88 0B 14 00 00 73 FF 00 (CRC 00 != 78)
000+13:28:01.868   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:28:03.467   ERROR [telegram] Rx: 8B 88 26 00 20 E4 88 0B 26 00 01 05 00 0D 00 1E 52 64 C3 (CRC C3 != B4)
000+13:30:01.642   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:31:01.769   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:31:03.046   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:31:04.031   ERROR [telegram] Rx: 8B 88 26 00 20 E4 88 0B 26 00 01 05 00 0D 00 1E 52 64 C3 (CRC C3 != B4)
000+13:32:02.315   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:32:02.734   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:33:01.610   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:33:03.025   ERROR [telegram] Rx: 8B 88 33 00 20 B0 88 0B 33 00 08 00 3C FB 00 28 00 02 46 D8 FF FF 00 C4 (CRC C4 != 79)
000+13:34:00.903   ERROR [telegram] Rx: 8B 88 14 00 20 2C 88 0B 14 00 00 74 05 F4 (CRC F4 != 8C)
000+13:34:01.904   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)
000+13:35:02.188   ERROR [telegram] Rx: 8B 88 16 19 20 16 88 0B 16 19 00 1E 00 CD (CRC CD != F1)
000+13:35:03.195   ERROR [telegram] Rx: 8B 88 26 00 20 E4 88 0B 26 00 01 05 00 0D 00 1E 52 64 C3 (CRC C3 != B4)
000+13:36:01.384   ERROR [telegram] Rx: 8B 88 19 19 20 2A 88 0B 19 19 80 00 95 (CRC 95 != 1B)`

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from http://ems-esp.local/api?device=system&cmd=info*
`{
  "System": {
    "version": "3.1.1",
    "uptime": "000+13:30:31.581",
    "freemem": 177
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "HT3",
    "#telegrams received": 29814,
    "#read requests sent": 5543,
    "#write requests sent": 54,
    "#incomplete telegrams": 1273,
    "#tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "#MQTT publishes": 10554,
    "#MQTT publish fails": 0,
    "#dallas sensors": 0,
    "#dallas reads": 0,
    "#dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)",
      "handlers": ""
    }
  ]
}`
**Additional context**
*Add any other context about the problem here.*

proddy, 2021-07-24:
try this
- grab the latest dev build from 
- in console turn on debug (`log debug`)
- do your `call boiler wwactivated on`
- watch on the log which telegrams are sent and what the responses are. If the Tx is failing it will tell you.
- do `show ems` and take a look at "#tx fails"

elRadix, 2021-07-24:
running the following firmware:
EMS-ESP Version
v3.1.2b6
 
command seems to run successfully this time but the warm water still remains deactivated

`
bosch:#
bosch:# call boiler wwactivated on
000+00:09:09.190 I 23: [command] Calling boiler command 'wwactivated', value on, id is default
000+00:09:09.190 I 24: [boiler] Setting boiler warm water active on
000+00:09:09.271 D 25: [telegram] Sending write Tx [#150], telegram: 8B 08 33 01 08 FE
000+00:09:09.307 D 26: [emsesp] Last Tx write successful
000+00:09:09.307 D 27: [telegram] Sending post validate read, type ID 0x34 to dest 0x08
000+00:09:11.430 D 28: [telegram] Sending read Tx [#151], telegram: 8B 88 34 00 20 AC
000+00:09:11.506 D 29: [emsesp] Last Tx read successful
bosch:# show ems
EMS Bus is connected.

EMS Bus info:
  Tx mode: 3
  Bus protocol: HT3
  #telegrams received: 447
  #read requests sent: 40
  #write requests sent: 3
  #incomplete telegrams: 0
  #tx fails (after 3 retries): 0
  Rx line quality: 100%
  Tx line quality: 100%

Rx Queue is empty

Tx Queue is empty

bosch:#
`
[image]
[image]
[image]
[image]

proddy, 2021-07-24:
can you post the output of `show ems`

MichaelDvP, 2021-07-24:
One thing i see is that the post-validation is to 0x34 (like wwtemp, etc), but for wwactivated it should be 0x33. 
Second there is a special case for HT3, sending `8` to activate, ems uses `0xFF`. We should check if this value is correct for this boiler.
@elRadix please use loglevel all and set the wwactivated from the boilerpanel to on and off, the log should show 0x33 telegrams. Or in terminal use `read 8 33` before and after changing the state. Is the command `call boiler wwactivated off` working if manual set to on?

proddy, 2021-07-24:
I noticed he had a bus protocol of HT3 (from `ems show`) while the master boiler is a Bosch GC7000 which I found odd. That would explain why it's sending 0x08 and not 0xff to activate.

MichaelDvP, 2021-07-24:
Yes, but sending is successfull from source 8B, so HT3 should be right. Also with 3.1.2b6 there are no incomplete telegrams.

elRadix, 2021-07-24:
hope this helps
[image]

elRadix, 2021-07-24:
here is the latest ems output

[image]

elRadix, 2021-07-24:
@MichaelDvP

with log level all

2021-07-24 20:18:44.435   TRACE 9:      [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 CC 00 00 80 00 00 80 00 80 00 80 00 00
2021-07-24 20:18:46.954    INFO 10:     [command]    Calling boiler command 'wwactivated', value true, id is default
2021-07-24 20:18:46.954    INFO 11:     [boiler]     Setting boiler warm water active on
2021-07-24 20:18:46.983   DEBUG 12:     [telegram]   Sending write Tx [#22], telegram: 8B 08 33 01 08 FE
2021-07-24 20:18:46.999   TRACE 13:     [emsesp]     Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 08 (offset 1)
2021-07-24 20:18:47.018   DEBUG 14:     [emsesp]     Last Tx write successful
2021-07-24 20:18:47.018   DEBUG 15:     [telegram]   Sending post validate read, type ID 0x34 to dest 0x08
2021-07-24 20:18:47.110   TRACE 16:     [emsesp]     Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 00 3C FB 00 28 00 02 46 D8 FF FF 00
2021-07-24 20:18:49.244   DEBUG 17:     [telegram]   Sending read Tx [#23], telegram: 8B 88 34 00 20 AC
2021-07-24 20:18:49.256   TRACE 18:     [emsesp]     Me(0x0B) <- Boiler(0x08), UBAMonitorWW(0x34), data: 20
2021-07-24 20:18:49.313   DEBUG 19:     [emsesp]     Last Tx read successful
2021-07-24 20:18:49.313   TRACE 20:     [emsesp]     Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 00 00 ED 80 00 A0 00 00 01 00 00 01 93 00 03 D9 00 80 00
2021-07-24 20:18:53.408    INFO 21:     [command]    Calling boiler command 'wwactivated', value false, id is default
2021-07-24 20:18:53.408    INFO 22:     [boiler]     Setting boiler warm water active off
2021-07-24 20:18:53.627   DEBUG 23:     [telegram]   Sending write Tx [#24], telegram: 8B 08 33 01 00 F6
2021-07-24 20:18:53.642   TRACE 24:     [emsesp]     Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 00 (offset 1)
2021-07-24 20:18:53.662   DEBUG 25:     [emsesp]     Last Tx write successful
2021-07-24 20:18:53.666   DEBUG 26:     [telegram]   Sending post validate read, type ID 0x34 to dest 0x08
2021-07-24 20:18:53.746   TRACE 27:     [emsesp]     Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 00 3C FB 00 28 00 02 46 D8 FF FF 00
2021-07-24 20:18:53.952   TRACE 28:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 00 ED 80 00 A0 00 00 01 00 00 01 93 00 03 D9 00 80 00
2021-07-24 20:18:54.183   TRACE 29:     [emsesp]     Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 00 00 00 00 00 00 00 00 00 00 00 00
2021-07-24 20:18:54.436   TRACE 30:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 15 00 00 00 03 40 C0 00 ED 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
2021-07-24 20:18:54.658   TRACE 31:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 00 00 00 00 00 03 E4 00 01 93 00 00 00 00 00 00 00 00 0B 80 00
2021-07-24 20:18:55.030   TRACE 32:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMaintenanceStatus(0x1C), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2021-07-24 20:18:55.329   TRACE 33:     [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 CC 00 00 80 00 00 80 00 80 00 80 00 00
2021-07-24 20:18:55.886   DEBUG 34:     [telegram]   Sending read Tx [#25], telegram: 8B 88 34 00 20 AC
2021-07-24 20:18:55.900   TRACE 35:     [emsesp]     Me(0x0B) <- Boiler(0x08), UBAMonitorWW(0x34), data: 20
2021-07-24 20:18:55.962   DEBUG 36:     [emsesp]     Last Tx read successful
2021-07-24 20:18:55.964   TRACE 37:     [emsesp]     Boiler(0x08) -> Me(0x0B), UBAMonitorWW(0x34), data: 00 00 ED 80 00 A0 00 00 01 00 00 01 93 00 03 D9 00 80 00
2021-07-24 20:19:03.995   TRACE 38:     [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 CC 00 00 80 00 00 80 00 80 00 80 00 00
2021-07-24 20:19:04.219   TRACE 39:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 00 ED 80 00 A0 00 00 01 00 00 01 93 00 03 D9 00 80 00
2021-07-24 20:19:04.530   TRACE 40:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 15 00 00 00 03 40 C0 00 ED 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
2021-07-24 20:19:13.887   TRACE 41:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 15 00 00 00 03 40 C0 00 ED 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
2021-07-24 20:19:14.130   TRACE 42:     [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 00 CC 00 00 80 00 00 80 00 80 00 80 00 00
2021-07-24 20:19:14.491   TRACE 43:     [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 00 ED 80 00 A0 00 00 01 00 00 01 93 00 03 D9 00 80 00

elRadix, 2021-07-24:
the state of wwactivated still remains off, its not changing its state at all

MichaelDvP, 2021-07-24:
I see that in telegram 33 the ww  in second position is off (00): 
>  Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 00 3C FB 00 28 00 02 46 D8 FF FF 00

but need to know what is right value for on. Switch on manual and type in terminal `read 8 33`
(The [original post](https://github.com/emsesp/EMS-ESP/issues/268) is abit confusing, because it's 08 in first place, but this is also 08 if wwactivated is switched off.)

> Turning wwactivated OFf works, but not back ON from EMS

That's good, so the 08 not working. Try in terminal: 
`call system send "0B 08 33 01 FF`
to turn on.

elRadix, 2021-07-25:
> Edit: I have changed it in [my testbuild](https://github.com/MichaelDvP/EMS-ESP32/releases/download/latest/EMS-ESP-3_1_2a6-ESP32.bin) if you want to try with a a bin.

just uploaded your testbuild and it works now ON and OFF  for wwactivated

I have yet to test the other commands, but warm water is now solved!
```

2021-07-25 11:16:18.712  INFO 324:    [boiler]   Setting boiler warm water activated off
2021-07-25 11:16:18.758 DEBUG 325:    [telegram] Sending write Tx [#82], telegram: 8B 08 33 01 00 F6
2021-07-25 11:16:18.773 TRACE 326:    [emsesp]   Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: 00 (offset 1)
2021-07-25 11:16:18.793 DEBUG 327:    [emsesp]   Last Tx write successful
2021-07-25 11:16:18.793 DEBUG 328:    [telegram] Sending post validate read, type ID 0x33 to dest 0x08
2021-07-25 11:16:18.845 TRACE 329:    [emsesp]   Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 00 37 FB 00 28 00 02 46 D8 FF FF 00
2021-07-25 11:16:19.190 DEBUG 330:    [telegram] Sending read Tx [#83], telegram: 8B 88 33 00 20 B0
2021-07-25 11:16:19.190 TRACE 331:    [emsesp]   Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 20
2021-07-25 11:16:19.261 DEBUG 332:    [emsesp]   Last Tx read successful
2021-07-25 11:16:19.261 TRACE 333:    [emsesp]   Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 00 37 FB 00 28 00 02 46 D8 FF FF 00
2021-07-25 11:16:23.701 TRACE 334:    [emsesp]   Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 00 01 5C 80 00 A0 00 00 01 00 00 01 68 00 03 DF 00 80 00
2021-07-25 11:16:24.056 TRACE 335:    [emsesp]   Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 01 A9 00 00 00 03 40 C0 01 5C 80 00 80 00 00 00 FF 00 00 00 CB 00 00 00
2021-07-25 11:16:24.281 TRACE 336:    [emsesp]   Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 02 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 00
2021-07-25 11:16:25.174 DEBUG 337:    [command]  Calling boiler command 'wwactivated', value true, id is default
2021-07-25 11:16:25.174  INFO 338:    [boiler]   Setting boiler warm water activated on
2021-07-25 11:16:25.237 DEBUG 339:    [telegram] Sending write Tx [#84], telegram: 8B 08 33 01 FF 09
2021-07-25 11:16:25.317 TRACE 340:    [telegram] Incomplete Rx: 8B 08 33 01 FF 09 FF
2021-07-25 11:16:25.767 DEBUG 341:    [telegram] Sending write Tx [#85], telegram: 8B 08 33 01 FF 09
2021-07-25 11:16:25.767 TRACE 342:    [emsesp]   Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: FF (offset 1)
2021-07-25 11:16:25.801 DEBUG 343:    [emsesp]   Last Tx write successful
2021-07-25 11:16:25.801 DEBUG 344:    [telegram] Sending post validate read, type ID 0x33 to dest 0x08
2021-07-25 11:16:25.886 TRACE 345:    [emsesp]   Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 37 FB 00 28 00 02 46 D8 FF FF 00
2021-07-25 11:16:26.165 DEBUG 346:    [telegram] Sending read Tx [#86], telegram: 8B 88 33 00 20 B0
2021-07-25 11:16:26.180 TRACE 347:    [emsesp]   Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 20
2021-07-25 11:16:26.226 DEBUG 348:    [emsesp]   Last Tx read successful
2021-07-25 11:16:26.226 TRACE 349:    [emsesp]   Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 37 FB 00 28 00 02 46 D8 FF FF 00

```

MichaelDvP, 2021-07-25:
Ok, I've also fixed it in the official dev build. Please do further tests with with official version, mine is sometimes a bit apart.

proddy, 2021-07-31:
please re-open if this is still happening.

commit f97cdcb4 (2021-07-25): fix #89

Also Norbert have 0xFF on his list.
closed 2021-07-31 by proddy