emsesp/EMS-ESP32 #91: U+FEFF in syslog please only when needed

wrohdewald, 2021-07-31:
There was issue 496 discussing this, but it seems to be deleted.

IIRC, in 496 you argued that you want to use UTF-8 everywhere which I think is very good. So in syslog you start all lines with U+FEFF (BOM) indicating UTF-8. 

However this is contrary to all other software I have ever seen writing to syslog. This prefix is not needed if the text can correctly be encoded as US-ASCII which is true for all your log entries I have seen so far.

So - if you can code your log lines to US-ASCII and back to UTF-8 and still have the same content, please do not use BOM.

That is still in accordance with RFC 5424 which even shows valid examples without BOM

proddy, 2021-07-31:
The issue is still there, just in the old ESP8266 project which I'm no longer supporting (https://github.com/emsesp/EMS-ESP/issues/496#issuecomment-728812345) 

What is the actual issue you are facing? Can you provide examples and why this is causing you fustration?

wrohdewald, 2021-07-31:
Same issue as in #496. Normally you would use "less" for looking at kern.log. It shows U+FEFF in reverse. No other entry I ever saw in syslogs needed that BOM because they all encode correctly in US-ASCII.

Please let me reiterate that this change does **not** violate RFC 5424.

From **less kern.log**:

Jul 27 16:34:10 speicher kernel: [1014681.811319] loop9: detected capacity change from 0 to 8
Jul 30 18:32:25 ems-esp - <U+FEFF>000+00:00:05.013 T 140: [emsesp] 20(0x20) -> Thermostat(0x10), ?(0xF7), data: FF 01 CD 29

proddy, 2021-07-31:
depends on the syslog server settings. On Ubuntu:

```
❯ tail -f /var/log/syslog
Jul 31 15:42:57 ems-esp emsesp ﻿001+22:51:59.891 T 29: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 25 73 00 02 10 65 40 01 BE 80 00 02 09 00 00 12 30 4C 01 1C 5A 02 00
Jul 31 15:42:57 ems-esp mqtt ﻿001+22:52:00.024 D 30: Publishing topic ems-esp/boiler_data (#5148, retain=0, retry=1, size=698, pid=1)
Jul 31 15:42:57 ems-esp mqtt ﻿001+22:52:00.125 D 31: Publishing topic ems-esp/boiler_data_ww (#5149, retain=0, retry=1, size=473, pid=1)
Jul 31 15:42:57 ems-esp mqtt ﻿001+22:52:00.227 D 32: Publishing topic ems-esp/thermostat_data (#5150, retain=0, retry=1, size=141, pid=1)
Jul 31 15:42:57 ems-esp mqtt ﻿001+22:52:00.329 D 33: Publishing topic ems-esp/dallassensor_data (#5151, retain=0, retry=1, size=13, pid=1)
Jul 31 15:42:58 ems-esp emsesp ﻿001+22:52:00.899 T 34: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 26 73 00 02 10 6D 40 01 B5 80 00 01 D0 00 07 12 30 4C 01 1C 5C 02 00
Jul 31 15:42:59 ems-esp emsesp ﻿001+22:52:01.882 T 35: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 24 73 00 0A 10 6D 40 01 B8 80 00 01 B0 01 3B 13 30 4C 01 1C 5C 02 00
Jul 31 15:42:59 ems-esp mqtt ﻿001+22:52:01.957 D 36: Publishing topic ems-esp/tapwater_active (#5152, retain=0, retry=1, size=2, pid=1)
Jul 31 15:43:00 ems-esp emsesp ﻿001+22:52:02.893 T 37: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 24 73 00 0A 10 65 40 01 C2 80 00 01 B2 01 7D 13 3D 48 00 C9 5B 02 00
Jul 31 15:43:01 ems-esp emsesp ﻿001+22:52:03.863 T 38: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 26 73 59 0A 10 65 40 01 CD 80 00 01 BB 01 97 13 3D 48 00 C9 5D 02 00
Jul 31 15:43:03 ems-esp emsesp ﻿001+22:52:05.924 T 39: Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 02 20 73 6F 0A 10 65 40 01 DC 80 00 01 C5 01 97 13 3D 48 00 C9 5C 02 00
```

if you want to fix to ASCII I'll welcome any PR.

MichaelDvP, 2021-08-20:
No complains on the new version, i thonk we can close.

closed 2021-08-20 by MichaelDvP