emsesp/EMS-ESP32 #94: i miss climate values

alexvr6, 2021-08-02:
Hello,
after switching from V2 to V3 I am missing some values. With V2 I had about 175 values ​​and with V3 I only have 125 values.
In V3, among other things, the climate values ​​are missing. There is only binary and sensor available. Do I have some wrong settings?

proddy, 2021-08-02:
Hi, a few questions to help narrow down what is happening
 - Which version of V3 did you try?
 - Does EMS-ESP detect all the devices?
 - Is Tx Mode set correctly?
 - Are you seeing any Tx or Rx errors (check quality in Web or `show ems` in Console)
 - Are you seeing the missing values in the Console, Web and MQTT?
 - Or are you talking about Home Assistant?
 - Do you have a list/dump of all the values from V2 to compare against?

alexvr6, 2021-08-02:
I am using version 3.2.0b2.
TX mode is EMS (without +).
TX / RX quality is 100%.
I could submit a list / dump of all values ​​from V2 later, for that I have to convert first.
I miss the values ​​in HomeAssitant, I only have 130 entities there, with v2 it was 175.

 The following errors can be found in the log:
2021-08-02 12:06:46.897 WARNING 6:      [telegram]   Incomplete Rx: C4 08 00 18 00 07 00 ED 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:07:16.789 WARNING 7:      [telegram]   Incomplete Rx: 88 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:07:46.089 WARNING 8:      [telegram]   Incomplete Rx: C4 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:08:06.822 WARNING 9:      [telegram]   Incomplete Rx: C4 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:08:28.258 WARNING 10:     [telegram]   Incomplete Rx: C8 08 00 07 00 0B 03 02 00 00 00 00 00 00 00 00 00 00
2021-08-02 12:10:44.724 WARNING 11:     [telegram]   Incomplete Rx: C4 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:10:47.724 WARNING 12:     [telegram]   Incomplete Rx: C8 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:11:16.676 WARNING 13:     [telegram]   Incomplete Rx: C8 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:11:56.741 WARNING 14:     [telegram]   Incomplete Rx: C4 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:13:06.724 WARNING 15:     [telegram]   Incomplete Rx: 98 08 00 18 00 07 00 ED 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:13:56.893 WARNING 16:     [telegram]   Incomplete Rx: 98 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:14:46.926 WARNING 17:     [telegram]   Incomplete Rx: C4 08 00 19 00 00 C6 00 EB 80 00 00 00 00 00 00 AC 87 14 A8 90 00 00 00 14 A8 90 00 AC 87
2021-08-02 12:15:06.822 WARNING 18:     [telegram]   Incomplete Rx: C4 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:15:46.786 WARNING 19:     [telegram]   Incomplete Rx: 88 08 00 19 00 00 C7 00 EB 80 00 00 00 00 00 00 AC 87 14 A8 90 00 00 00 14 A8 90 00 AC 87
2021-08-02 12:16:06.863 WARNING 20:     [telegram]   Incomplete Rx: 88 08 00 18 00 07 00 ED 00 00 00 00 00 60 80 00 80 00 00 DD 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:16:08.767 WARNING 21:     [telegram]   Incomplete Rx: C4 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:16:10.993 WARNING 22:     [telegram]   Incomplete Rx: 31 08 00 07 00 0B 01 02 00 00 00 00 00 00 00 00 00 00
2021-08-02 12:16:36.826 WARNING 23:     [telegram]   Incomplete Rx: 98 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:17:06.847 WARNING 24:     [telegram]   Incomplete Rx: 88 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:17:16.966 WARNING 25:     [telegram]   Incomplete Rx: C8 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:18:26.998 WARNING 26:     [telegram]   Incomplete Rx: 31 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:18:46.924 WARNING 27:     [telegram]   Incomplete Rx: C8 08 00 19 00 00 C7 00 EB 80 00 00 00 00 00 00 AC 87 14 A8 90 00 00 00 14 A8 90 00 AC 87
2021-08-02 12:19:03.151 WARNING 28:     [telegram]   Incomplete Rx: C4 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:19:44.365 WARNING 29:     [telegram]   Incomplete Rx: C4 08 00 07 00 0B 01 02 02 00 00 00 00 00 00 00 00 00
2021-08-02 12:19:46.778 WARNING 30:     [telegram]   Incomplete Rx: 18 08 00 18 00 07 00 ED 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:19:56.905 WARNING 31:     [telegram]   Incomplete Rx: 88 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:20:16.902 WARNING 32:     [telegram]   Incomplete Rx: C8 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:20:26.943 WARNING 33:     [telegram]   Incomplete Rx: 31 08 00 18 00 07 00 EF 00 00 00 00 00 60 80 00 80 00 00 DF 00 00 0A 30 48 00 CB 00 00 00
2021-08-02 12:20:27.929 WARNING 34:     [telegram]   Incomplete Rx: C8 08 00 07 00 0B 03 02 00 00 00 00 00 00 00 00 00 00

proddy, 2021-08-02:
ok, home assistant. If you're coming from V2 then delete everything in HA (remove the devices from the Integrations->MQTT) and also delete any retained messages using something like MQTTExplorer. 

Then compare what you see in the WebUI to what is being shown in HA. When you see any differences we can zoom in and fix those. It looks like the thermostat data is missing since you mentioned Climate?

alexvr6, 2021-08-02:
I deleted everything and reintegrated it, no change.
The following is not displayed in the HA (it is displayed in the Webui):
Gas off, flame current, heating activated,
maintenance scheduled, maintenance set time, maintenance set date. The following is displayed in the HA but not in the WebUI:
climate.thermostat_hc1, climate.thermostat_hc2,
sensor.thermostat_hc1_cobtrol_device, sensor.thermostat_hc2_cobtrol_device and several values ​​with sensor.system.xxx

In the HA you can see the following: http://angelroda-wetter.de/tmp/heiz2.JPG

proddy, 2021-08-02:
For the missing boiler commands (like gas, flame current)
- Go to HA, Configuration->Integrations->MQTT->EMS-ESP Boiler and see if 'Boiler gas' is there. If not click on the 'MQTT INFO' to see where the payload is, you can ctrl-F to search. I think this could be a configuration issue.
- You could also use MQTTExplorer and search for 'gas' and it'll show you the MQTT topics (all 3 of them)

For the missing thermostat, the same thing
- Go to HA, Configuration->Integrations->MQTT->EMS-ESP Thermostat, and you should see 'Thermostat hc1' with some values.

if you could post some of your settings I'll double-check your config. Web UI->Dashboard->EMS-ESP HELP->Download all settings. The MQTT settings are most interesting.

alexvr6, 2021-08-02:
I was able to find some values:
[http://angelroda-wetter.de/tmp/ha1.JPG](url)
[http://angelroda-wetter.de/tmp/ha2.JPG](url)
Some values ​​are unknown.

Here are the settings:
{
  "System": {
    "version": "3.2.0b2"
  },
  "Network": {
    "hostname": "ems-esp32",
    "static_ip_config": false,
    "enableIPv6": false
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "192.168.178.51",
    "port": 1883,
    "username": "alex",
    "client_id": "ems-esp32",
    "keep_alive": 60,
    "clean_session": false,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "ha_climate_format": 1,
    "ha_enabled": true,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "subscribe_format": 0
  },
  "NTP": {
    "enabled": true,
    "server": "time.google.com",
    "tz_label": "Europe/Berlin",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": true,
    "port": 8266
  },
  "Settings": {
    "tx_mode": 1,
    "tx_delay": 5,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "master_thermostat": 0,
    "shower_timer": false,
    "shower_alert": false,
    "rx_gpio": 23,
    "tx_gpio": 5,
    "dallas_gpio": 18,
    "dallas_parasite": false,
    "led_gpio": 2,
    "hide_led": true,
    "notoken_api": true,
    "dallas_format": 1,
    "bool_format": 1,
    "enum_format": 1,
    "analog_enabled": false,
    "pbutton_gpio": 0,
    "board_profile": "NODEMCU"
  }
}

proddy, 2021-08-03:
ok so looks like the thermostat is not loading the values from MQTT. Get MQTTExplorer and let's see what is happening. Search for `hc1` and you should see a few entries like:

[image]

check the `ems-esp/thermostat_data` and `homeassistant/climate/ems-esp/thermostat_hc1` topics.

alexvr6, 2021-08-03:
OK, I have these entries:

[image]

proddy, 2021-08-03:
I can't see the sensors (which is where the data is) but from what I can tell everything looks fine. The data is there, the configs are correct and I assume they look fine in the MQTT integrations section in HA (see previous comment)

alexvr6, 2021-08-06:
Hello,
with the V3 I don't get it to work. The data gets stuck somewhere:
[image]

After the dismantling to V2 everything works again.
[image]

If something new arises, I will report

proddy, 2021-08-06:
Ok, I'm not sure what else I can do. You will need to troubleshoot on your side. As I've mentioned before, go to HA->Configuration->Integrations->Mosquitto MQTT->devices->EMS-ESP Thermostat and then check the values and click 'MQTT INFO' to see where the data should be coming from. There must be a mismatch.

Also switching between V2 and V3 will mess up your HA and MQTT configs so make sure you clean everything.

closed 2021-09-28 by proddy