emsesp/EMS-ESP32 #96: wwsettemp foron boiler not writable anymore

tp1de, 2021-08-10:


proddy, 2021-08-10:
I changed this in 3.2.1 (see https://github.com/emsesp/EMS-ESP32/releases/tag/v3.2.1). 

It was incorrectly assigned. Use `wwSetTemp`

tp1de, 2021-08-10:
I use ems-esp/api/boiler http-request to get all fields. I get wwsettemp and other fields back.
With ems-esp/api/boiler/wwsettemp I get:
[image]

Not writable anymore as in version 3.2.0b0.

proddy, 2021-08-10:
sorry, i mixed it up, use http://ems-esp/api/boiler/wwseltemp

tp1de, 2021-08-11:
@proddy @MichaelDvP 

v 3.2.2: I compared km200 and ems-esp datafields in respect to ww temperature settings for RC310 and recognized the following:

- wwsettemp = dhwCircuits.dhw1.currentSetpoint -> read only - OK and actual setpoint
- wwseltemp = writable but unclear and inconsistent behavior:
... until v3.2.0: is changing dhwCircuits.dhw1.singleChargeSetpoint - but only used for single charges.
... now v3.2.2: is changing dhwCircuits.dhw1.temperatureLevels.high -**set value correctly but wwseltemp set back to 60°C afterward**s

RC310 has input fields for: 

1. single charge temperature
2. temperature levels high
3. temperature levels low
4. temperature levels off 

The temperature levels 2 and 3 chosen depends on the ww switch program. During day high level, during night low level.
The holiday mode set by holiday switch program then sets the off level during holidays.

Within ems-esp switch programs and holiday modes are not available but the respective temperature levels should be made accessible. 

For me is the actual version not consistent for RC310.

proddy, 2021-08-11:
what should the correct behavior be when changing `wwseltemp` ? I only tested on my RC20 which doesn't have the other modes.

tp1de, 2021-08-11:
> what should the correct behavior be when changing `wwseltemp` ?

When wwseltemp is writing to the temperature level high field like in actual version v3.2.2, 
then I wonder while writing seems to work and the set value is persistant, 
why is wwseltemp changing afterwards back to 60°C?

Are writing and reading of wwseltemp referencing different telegram types/value-positions?

To integrate the other field values we should wait for #97.

MichaelDvP, 2021-08-12:
Afaik:
- wwsettemp is the monitor value published by boiler. It is influenced by day/night and solar power. in ems telegram 0x34, ems+ telegram 0xE9. wwsettemp can not be written. This value is used to trigger the ww-heating of the boiler (with wwhysteresis).
- wwseltemp is the setting-value, send to the boiler. Ems in telegram 0x33, ems+ telegram 0xEA. 

But seltemp can also be written by other devices, in my case the controller MC10 have a rotary knob for ww-temp, this has to be set to `auto` otherwise the MC10 overwrites any seltemp. 
The ems+ system has settings in the thermostat RC300 for wwtemp and wwtemplow, the RC300 will overwrite any setting send to the boiler directly by emsesp, you have to use the thermostat wwtemp to change the boiler setting. You can check y watching telegram 0xEA and change seltemp from emsesp (0x0B->0x08), after a few seconds the thermostat (0x10->0x08) will change the  
value back.

tp1de, 2021-08-12:
> The ems+ system has settings in the thermostat RC300 for wwtemp and wwtemplow, the RC300 will overwrite any setting send to the boiler directly by emsesp, you have to use the thermostat wwtemp to change the boiler setting.

Thanks Michael understood that wwseltemp can be overwritten by other devices. 
In former versions until v3.2.0 wwseltemp was changing one time charging temperature.

Now in v3.2.2 wwseltemp is changing the parameter temperature level high of RC310 with telegram to 08 (boiler), type EA offset 6. This is identical to changing this value by km200 ... so far so good. This setting stays persistant and wwsettemp is changed accordingly (read only is clear now). Btw. no message to thermostat RC310 - the thermostat setting is syncronized from boiler.

I would just expect that this value in ems-esp stays as beeing set but it is overwritten by 60°C all the time. 
So write to boiler works correctly but reading is changing the value afterwards.

> you have to use the thermostat wwtemp to change the boiler setting.

I do not have wwtemp in thermostat !

proddy, 2021-08-12:
>I do not have wwtemp in thermostat !

on the thermostat it's called `wwsettemp`.  There is also a `wwsettemplow`. The `commands` API endpoint will list all the available commands.

tp1de, 2021-08-12:
@proddy  **now I am getting confused.**

1. YES these commands are shown http://ems-esp/api/thermostat/commands
2. NO these fields are not shown in http://ems-esp/api/thermostat  (wwsettemp and wwsettemplow and others are missing)
3. YES details are shown in http://ems-esp/api/thermostat/wwsettemp

Since I use (2) to identify all available fields per device I don't recognize that these fields exist at all.
There are other fields shown by command missing as well.

?????? Is this a bug ??????

proddy, 2021-08-12:
`wwsettemp` and `wwsettemplow` are commands, they will not show up in (2) as they don't have values (by the way in the API v3 I wanted to separate out commands from device values but we could never agree).  (3) is a bug.

tp1de, 2021-08-12:
I have difficulties to understand you. Up to yet I understood that device values can be read only or writable.
If they are writable than values can be changed by "commands" with use of API - so far so good.

But what are commands without a device value (field)? For me these can't exist.

proddy, 2021-08-12:
:-) I don't understand myself either, most of the times!

So yes it is possible to have commands and endpoints that don't have values. 

But in your case (I just checked the code) both the wwsetemp and wwsettemplow are real device values from the Thermostat and can be written too. So I was mistaken. They come from the telegram 0x31B. Try querying it using a `read` command. You should see the values in the console, web and API. If not it means either 0x31B is not polled or broadcasted or the values are empty. If I had an RC310 I could test myself so I'm just going on what Michael did (and he did some great work on RC3x0 support)

tp1de, 2021-08-12:
ems-esp:# read 10 31B
001+07:17:46.469 N 43: [emsesp] Me(0x0B) <- Thermostat(0x10), RC300WWtemp(0x31B), data: 20
001+07:17:46.509 N 44: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300WWtemp(0x31B), data: <empty>
001+07:17:46.920 N 45: [emsesp] Gateway(0x48) <- Thermostat(0x10), RC300WWtemp(0x31B), data: 02
001+07:17:46.940 N 46: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300WWtemp(0x31B), data: <empty>

What does this mean?

I recognized that I can't find 31B telegrams in syslog but in console watch?

ems-esp:# watch 31B
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a telegram type of 0x31B
001+07:22:31.599 N 75: [emsesp] Gateway(0x48) <- Thermostat(0x10), RC300WWtemp(0x31B), data: 01 (offset 1)
001+07:22:31.626 N 76: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300WWtemp(0x31B), data: <empty> (offset 1)
001+07:22:31.648 N 77: [emsesp] Gateway(0x48) <- Thermostat(0x10), RC300WWtemp(0x31B), data: 01
001+07:22:31.674 N 78: [emsesp] Thermostat(0x10) -> Gateway(0x48), RC300WWtemp(0x31B), data: <empty>

proddy, 2021-08-12:
doesn't look like it has any data in the telegram. Maybe try 31C ? Also strange to see empty data blocks. Could you watch in raw format so we can see the complete telegram?

tp1de, 2021-08-12:
31C does not show any telegrams.

ems-esp:# watch raw 31B
Watching incoming telegrams, displayed as raw bytes
Filtering only telegrams that match a telegram type of 0x31B
001+07:47:54.064 N 231: [telegram] Rx: 48 90 FF 01 01 02 1B 64
001+07:47:54.089 N 232: [telegram] Rx: 10 48 FF 01 02 1B 7A
001+07:47:54.609 N 233: [telegram] Rx: 48 90 FF 00 01 02 1B 6C
001+07:47:54.635 N 234: [telegram] Rx: 10 48 FF 00 02 1B 7E

MichaelDvP, 2021-08-13:
I've looked a the latest commits and the only change i see is [this](https://github.com/emsesp/EMS-ESP32/commit/15541c52ceefaf13d1ad5db3579744fc91eb54e0). In all versions there was wwsettemp read from 0xE9, wwseltemp read from 0xEA. Writing the new temp was always to 0xEA, but was called wwsettemp in versions less 3.2, now it's correctly called wwseltemp, that's the only difference. If something changes the temperature back to 60°C it is not from emsesp and not version dependent.

@tp1de  Your RC300 seems not to  support the 31B telegram (i think it was from RC310), the telegram is empty but also the KM200 tries to read it.. So something other sets the boiler-ww-temp back to 60 degree. Please watch the EA telegram (or `log all`), change the value via emsesp and wait until the vlaue changes back to 60.

MichaelDvP, 2021-08-13:
@tp1de You have mentioned that you have input fields:
- single charge temperature
- temperature levels high
-  temperature levels low
-  temperature levels off

Please change each of these values while `log all` or `watch ea`. If we can identify the fields, we can add these values. The EA-telegram have some unknown temperatures in pos 16, 17, 18.

tp1de, 2021-08-13:
setting with km200: telegram 48->08 type EA and offsets:

-     single charge temperature: 10
-     temperature levels high: 06
-     temperature levels low: 12
-     temperature levels off: read only value 0

Reading telegrams 08 -> 0B type EA - positions starting counting from 0:

-     single charge temperature: pos 16
-     temperature levels high: pos 6
-     temperature levels low:  pos 18

Is this normal that writing and reading positions differ?

MichaelDvP, 2021-08-13:
> Is this normal that writing and reading positions differ?

No, that's why i ask for log. I suppose you are confused by decimal/hex offsets, but without log this only looking into a crystal ball.

tp1de, 2021-08-13:
> I suppose you are confused by decimal/hex offsets

Yes you are right the offsets a hex out of telegrams. So positions match.

tp1de, 2021-08-13:
There is another km200 field: dhwCircuits.dhw1.chargeDuration
Changing it results in telegram 48 --> 10 (thermostat) with EMS+ message type 1F5 and offset 0A (hex) - 10 (dec)

Values set are in 15 minutes - eg. 2 = 30 min and 8 = 120 minutes with telegram 48 10 FF 0A 01 F5 08 51

proddy, 2021-09-22:
@MichaelDvP @tp1de what do we do with this issue?

MichaelDvP, 2021-09-22:
@proddy  I think you can close, the temperature levels and chargeDuration is implemented.

tp1de, 2021-09-22:
wwsettemp is read only and shows the actual set temperature for ww. So this is ok.

- wwseltemplow is working like km200 dhwCircuits.dhw1.temperatureLevels.low. So this is ok.

- wwseltemp should correspond to hwCircuits.dhw1.temperatureLevels.high
Value is always 60°C and when I change to 54°C the right telegram is send: 0B 08 EA 06 36 49  
Telegram is ok and when I change by km200 the telegram is: 48 08 EA 06 36 1D
But value is not changed and on next read I get again 60°C .

Could you check why?

MichaelDvP, 2021-09-22:
1. Try to fetch what the km200 sends. 
2. Some boilers uses the 0x35 for setting ww. Try `call system send "0B 08 35 03 36"`

tp1de, 2021-09-22:
>     1. Try to fetch what the km200 sends.

Ems-esp and km200 are sending the same telegram - see above.
When I change the value the right telegram is send by ems-esp and the value is changed in km200 / RC310. So changing works.

**Nevertheless the value read is always set back to 60°C and does not stay on the value selected.**

tp1de, 2021-09-29:
@MichaelDvP 
Is there anything I can do to get this issue solved?

MichaelDvP, 2021-09-29:
Watch the EA telegram and find out what set's it back. 
If there is nothing make a full log from setting the new value via EA to next read from emsesp with 60C. Maybe find another telegram that causes the reset.

tp1de, 2021-09-29:
Setting wwseltemp to 54°C by ems-esp dashboard gives the following EA telegrams:

<html>
<body>

2021-9-29 10:55:15 | 08 0B EA 00 00 00 00 00 00 01 36 F8 00 0A 01 02 4B D8 00 B4 3C 3C 2A 0A 3C 64 64 00 00 01 0B
-- | --
2021-9-29 10:55:08 | 08 48 EA 1B AF
2021-9-29 10:55:01 | 08 00 EA 00 00 00 00 00 00 01 36 F8 00 0A 01 02 4B D8 00 B4 3C 3C 2A 0A 3C 64 64 00 00 01 74
2021-9-29 10:55:01 | 08 0B EA 00 00 00 00 00 00 01 36 F8 00 0A 01 02 4B D8 00 B4 3C 3C 2A 0A 3C 64 64 00 00 01 0B
2021-9-29 10:55:00 | 0B 08 EA 06 36 49

</body>
</html>

wwseltemp as shown in the dashboard is set back to 60°C but wwsettemp is correctly set and shown as 54°C.
Do you read the right telegram offset for wwseltemp? Offset 6 in telgram 08 00 EA is 0x36 representing 54°C .

MichaelDvP, 2021-09-29:
Yes, the wwSelTemp is read from 0xE9, [introduced here](https://github.com/emsesp/EMS-ESP/issues/382). This seems to be wrong, i suspect also for the disinfection temp, which is set in EA and now read from E9. I'll change both to read back the same paramter which is written..
Please also check E9 before and after changing wwSelTemp and wwDisinfectTemp.

tp1de, 2021-09-29:
Setting wwseltemp back to 52°C and afterwards to 54°C gives the following E9 telegrams (before / after):

<html>
<body>

2021-9-29 12:07:06 | 08 00 E9 00 36 01 DE 01 DE 00 00 00 00 4B 3C 00 00 00 00 F7 39 00 06 B6 00 00 00 36 00 36 B2
-- | --
2021-9-29 12:06:56 | 08 00 E9 00 34 01 DE 01 DE 00 00 00 00 4B 3C 00 00 00 00 F7 39 00 06 B6 00 00 00 34 00 34 FD

</body>
</html>

tp1de, 2021-09-29:
Setting wwdisinfecttemp from 70°C to 75°C gives the following E9 telegrams (before / after):

<html>
<body>

2021-9-29 12:13:37 | 08 00 E9 00 36 01 DB 01 DB 00 00 00 00 4B 3C 00 00 00 00 F7 39 00 06 B6 00 00 00 36 00 36 FF
-- | --
2021-9-29 12:13:27 | 08 00 E9 00 36 01 DB 01 DB 00 00 00 00 46 3C 00 00 00 00 F7 39 00 06 B6 00 00 00 36 00 36 2C

</body>
</html>

tp1de, 2021-09-29:
The same setting changes from 70 to 75°C on wwdisinfecttemp in EA telegrams:

<html>
<body>

2021-9-29 12:20:43 | 08 00 EA 00 00 00 00 00 00 01 36 F8 00 0A 01 02 4B D8 00 B4 3C 3C 2A 0A 3C 64 64 00 00 01 74
-- | --
2021-9-29 12:20:43 | 0B 08 EA 0C 4B 20
2021-9-29 12:20:16 | 08 0B EA 00 00 00 00 00 00 01 36 F8 00 0A 01 02 46 D8 00 B4 3C 3C 2A 0A 3C 64 64 00 00 01 74

</body>
</html>

tp1de, 2021-09-29:
> Can you confirm that wwSetTemp is right?

wwsettemp is 0x36 representing 54°C in offset 0 in E9 telegram. That's right as mentioned above.
wwsettemp is always the right value just wwseltemp is reset to 60°C.

tp1de, 2021-09-29:
@MichaelDvP 
I have seen that you have loaded new version b10 with corrections. 
wwseltemp is now working correctly ... super.

I think we can close this issue.

closed 2021-09-29 by MichaelDvP