emsesp/EMS-ESP32 #98: Extend API to change the SysLog Log Level

proddy, 2021-08-17:
I propose a new system command called `syslog_level` called via `/ems-esp/api/system/syslog_level` and take a JSON body as

```json
{ "log": "<level str>" }
``` 

`"log"` can also be `"value"` to be compatible with the other calls.

where `<level str>` is one of  OFF, EMERG, ALERT, CRIT, ERR, WARNING, NOTICE, INFO, DEBUG, TRACE, ALL.

proddy, 2021-08-17:
Perhaps also a special <level str> called `RAW` which does the same as `watch raw` in the console. i.e. only the telegrams as raw bytes are sent to the SysLog.

MichaelDvP, 2021-08-17:
The raw option affects all loggers (console/web/syslog) and also the watch command (which affects also all loggers). I think it's confusing to have this as extra level in syslog command. Maybe better to have the raw-switch as extra api command (`/api/system/log_raw: {"value":true}`) or (imo not so good) as extra parameter: `/api/system/syslog: {"level":"all","raw":true}`
We can also add a parameter to syslog to transmit only the selected level, not all levels below: `{"level":"trace", "raw":true", "filter":true}` (as a hack we should allow for filtered-trace also the  notice logs if watch is activated)

proddy, 2021-08-17:
Ok. so for now
1. add `/api/system/log_raw: {"value":true}` 
2. add `/api/system/syslog_level: {"log": "<level str>"}`

that will satisfy what Thomas needs if he uses NOTICE and turns on raw. The filter is a nice idea but I don't think it's worth the extra work for his edge case. Unless others will use it.

MichaelDvP, 2021-08-17:
> if he uses NOTICE and turns on raw

Ups, we have a misunderstanding, i thought about the "raw" option in web-settings, that only switches "watch" from pretty to hex, but do not activate watch. (btw: I've used raw a long time, but pretty_telegram is stable and trustworthy since a long time now, raw is mostly obsolete). 
For this it's better to change 1. to `api/system/watch: {"value": "off|on|raw"}` to stay in sync with console. Than `log notice` and `watch on` will work like in console and affects console, web-log and syslog.

proddy, 2021-08-17:
ok. agree. keep it in sync is smart

tp1de, 2021-08-21:
@proddy @MichaelDvP Are any changes to syslog done in v3.2.2a0 ?

I was testing for km200 delay in #90 and recognized that syslog output changed.

Syslog level trace and output in hex works fine as I proposed - just the rx messages - FINE !

The other loglevels send wrong format to syslog server so I gett error messages from syslog. 
In my case node-red syslog node with log-level set to debug:

`"Invalid message received: {"timestamp":"2021-08-21T19:28:06.175Z","address":"192.168.178.71","family":"IPv4","port":59707,"size":136,"msg":"<7>1 2021-08-21T21:27:00.026586+2:00 ems-esp emsesp - - - 000+00:35:32.043 D 8353 No telegram type handler found for ID 0x8E4 (src 0x08)"}"`

MichaelDvP, 2021-08-22:
My alpha-version sometimes differ from official version. This #90 test was intended only to test the km200 handling, please switch back to offiicial beta for syslog discussions.

tp1de, 2021-08-23:
@proddy @MichaelDvP 

Syslog errors (wrong format) is also related to last dev version v3.2.2b0.

proddy, 2021-09-26:
@tp1de can this be closed?

tp1de, 2021-09-26:
Yes

closed 2021-09-27 by proddy