emsesp/EMS-ESP32 #99: MQTT unstable

Boui3D, 2021-08-18:
After changing to v3.2.1 MQTT is unstable. 
Restart my Fritzbox oder my Homeserver (PI4, includes Iobroker) MQTT is disable and remains deactivated.

proddy, 2021-08-18:
what did you upgrade from? what do you mean by unstable? Any logs you can share?

Boui3D, 2021-08-18:
I think it was the v3.1.0 or v3.1.1. It´s been three weeks, since I did the upgrade. I went on vacation I´m not watching it any further. I noticed it yesterday when I updated my server and it´s repatable. The MQTT Status changes to Disconnect an never comes up again.
I did a Restart of my Fritzbox now to create this result.

000+00:00:00.000 I 0: [emsesp] EMS Device library loaded with 79 records
000+00:00:00.000 I 1: [emsesp] Starting Access Point with captive portal on 192.168.4.1
000+00:00:04.878 I 2: [emsesp] WiFi Connected with IP=192.168.178.75, hostname=ems-esp
000+00:00:04.906 I 3: [mqtt] MQTT connected
000+00:00:10.000 I 4: [emsesp] Stopping Access Point
000+01:10:56.696 I 5: [mqtt] MQTT disconnected: TCP
000+01:11:00.298 I 6: [emsesp] Starting Access Point with captive portal on 192.168.4.1
000+01:12:02.879 I 7: [emsesp] WiFi connection dropped, stopping NTP
000+01:12:32.885 I 8: [emsesp] WiFi connection dropped, stopping NTP
000+01:13:34.169 I 9: [emsesp] WiFi Connected with IP=192.168.178.75, hostname=ems-esp
000+01:13:40.305 I 10: [emsesp] Stopping Access Point

I´ve never seen this problem before.

MichaelDvP, 2021-08-18:
I can reproduce. If wifi connection is dropped and reconnected (disable the accesspoint for some secondes), mqtt stays disconnected.
It can also be triggered by `Network Settings` - `SAVE`, after wifi connection is reenabled mqtt stays disconnected. Changing mqtt settings does not help, only reboot. 
I have not tested with older versions, not sure if this is a new bug.

MichaelDvP, 2021-08-18:
I think i've found the cause. @Boui3D: Please check the [new dev](https://github.com/emsesp/EMS-ESP32/releases) and give feedback (or close this issue if fixed).

MichaelDvP, 2021-08-18:
Oh, now reconnect after wifi drop works, but mqtt-IPv6 does not connect anymore. Needs some more fixing/testing...

Boui3D, 2021-08-18:
I stay tuned. Thanks for your commitment.

krombel, 2021-08-19:
Hi there.
I experienced the same issue.
As I am in an v4 environment: The latest beta build should work fine for me, right?
I installed it either way and will report once I notice an issue. For now it is connected and pushing metrics.

MichaelDvP, 2021-08-19:
Ok, now it works for me with IPv4 and IPv6. Tested with mosquitto and iobroker-mqtt for wifi reconnect and mqtt-broker-restart.
I don't really understand what was wrong, only that is was a timing issue on reconnecting wifi.
(Issue started with IPv6 integration in v3.1.2b3).

proddy, 2021-08-19:
Thanks Michael. I was worried it was due to the updated AsyncMQTTClient lib to 0.9.0 which went into our 3.2.0 release (August 6). https://github.com/marvinroger/async-mqtt-client/pull/254

Boui3D, 2021-08-19:
Thanks Michael. It works well for me.
I love this projekt.

MichaelDvP, 2021-08-19:
Some more tests (except disconnecting a really long time), got always a reconnect, i think it is fixed. Please reopen if i missed a essential test.

commit 165dd3b4 (2021-08-18): fix #99, mqtt reconnect after wifi drop
closed 2021-08-19 by MichaelDvP