emsesp/EMS-ESP32 #100: no attribute 'hamode' when rendering '{{value_json.hc1.hamode}}'

WouterKr, 2021-08-19:
Hi,

Most of this kind of messages has been solved already but this one occurs in my setup. I am using the ems-esp with homeassisant integration set and the mqtt setting default:

**Bug description**
After a restart or during operation the following messages appears in the core log of homeassistant (every 10 sec based on default value mqtt setting):

```
2021-08-19 19:24:11 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'hamode' when rendering '{{value_json.hc1.hamode}}'
2021-08-19 19:24:11 ERROR (MainThread) [homeassistant.components.mqtt.climate] Invalid modes mode: 
```

**Steps to reproduce**
A restart of the device triggers this messages but occurs during operation also.

**Expected behavior**
No error message in the core log of homeassistant and a correct rendering of the variable

**Device information**
{
  "System": {
    "version": "3.2.1",
    "uptime": "000+01:02:42.414",
    "uptime_sec": 3762,
    "freemem": 183
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 3335,
    "read requests sent": 524,
    "write requests sent": 0,
    "incomplete telegrams": 0,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT publishes": 1084,
    "MQTT publish fails": 0,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.07)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Logamatic TC100/Moduline Easy (DeviceID:0x18, ProductID:202, Version:02.22)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x0A"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.04)",
      "handlers": ""
    }
  ]
}

proddy, 2021-08-19:
If you have MQTTExplorer installed could you fire it up and type in `hamode` in the search. Select the topic `thermostat_data`. You should see something like the below:

[image]

we're looking for hamode in the nested json hc1

WouterKr, 2021-08-19:
"hamode" is only found ounce and not in the nested json hc1 as far as I can see:

[image]

[image]

Btw : I noticed the "retain" flag. This is not set in the config settings of the ems-esp.

proddy, 2021-08-19:
the homeassistant MQTT discovery topics are all retained by EMS-ESP, that's normal. Otherwise everytime you restart HA it'll loose all EMS-ESP integrations.

So hamode is missing. What do you see when you run `http://ems-esp/api/thermostat` ? Mine is

```json
{
  "id": 77,
  "errorcode": "(0)",
  "datetime": "22:12:21 19/08/2021",
  "hc1": {
    "seltemp": 20.5,
    "currtemp": 22.4,
    "hatemp": 22.4,
    "hamode": "auto",
    "mode": "auto"
  }
}
```

WouterKr, 2021-08-20:
http://ems-esp/api/thermostat gives me:

{
  "id": 202,
  "hc1": {
    "seltemp": 21.4,
    "currtemp": 21.4,
    "hatemp": 21.4
  }
}

MichaelDvP, 2021-08-20:
> Logamatic TC100/Moduline Easy (DeviceID:0x18, ProductID:202, Version:02.22)

From this thermostat we know only a single monitor telegram and only two values out of 26.
If there are other parameters to set/read you can help by starting a log-all from the webpage and change on the thermostat some values. Than download the log and post it here with a report what you have changed (e.g. set setpoint to 22, set mode to xy, set mode back to yz, ...) .

WouterKr, 2021-08-20:
I hope i did the right thing by starting with a setpoint of 20 to 21,22,23 and back again and added the logfile.
Please let me know I you need anything else.
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7020171/log.txt)

WouterKr, 2021-08-20:
Thank you for the quick fix! Just uploaded the firmware and now the hamode is set and no errors in HA core log anymore.

proddy, 2021-08-20:
great, nice fix Michael.

closed 2021-08-20 by proddy

MichaelDvP, 2021-08-20:
> log.txt

That's fine. But there are indeed no other data. Also there is no clock in that thermostat (because it does not have modes).
`2021-08-20 10:20:55.431 TRACE 75: [emsesp] Thermostat(0x18) -> Me(0x0B), RCTime(0x06), data: <empty>`
The monitor telegram is long, but all values are zero, only setpoint and current temperature as already registered.
`2021-08-20 10:20:56.653 TRACE 96: [emsesp] Thermostat(0x18) -> Me(0x0B), EasyMonitor(0x0A), data: 00 00 00 00 00 00 00 00 08 86 07 D0 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

@WouterKr  Sorry, there are no more information from that thermostat.

@proddy The weblog sometimes creates double messages, I think i know what is the cause. I'll test and fix.

proddy, 2021-08-20:
> The weblog sometimes creates double messages, I think i know what is the cause. I'll test and fix.

@MichaelDvP thanks, and also I like what you did with the Level filter in the web UI so feel free to check in some of your other improvements too.

@WouterKr there are a few extensions to HA that work with the Easy-type thermostats, like https://github.com/bosch-thermostat/home-assistant-bosch-custom-component

WouterKr, 2021-08-20:
Well if there is no more information to get then all is working then, thank you  ;-)
And thanks for the tip on the Bosch custom component in HA. Will look into it.

MichaelDvP, 2021-08-20:
@proddy 
> @MichaelDvP thanks, and also I like what you did with the Level filter in the web UI so feel free to check in some of your other improvements too.

I'm not sure what you mean. For my personal  preference i have completly removed all your filtering, show all messages that come from logger (up to set level). If changing the level in web-ui, all messages stay and only new messages come with new level. But i thought you prefer to show only messages with selected level. ([this change in my build](https://github.com/MichaelDvP/EMS-ESP32/commit/03556bb75904962dc8ae6bad6deca863eed772fe))

BTW: also offtopic here: do you think [this](https://github.com/MichaelDvP/EMS-ESP32/commit/f3bf949535a19e1dbc77be2a24e817746f6790f5) can be a solution for km200 users? In normal operation polls come less than 10 sec, but in boiler-cold-start the master scans in order of id, 0x48 comes late and km sees no registered service-key before.

proddy, 2021-08-20:
yes, I meant what you did in your own fork (I was being nosey). I like what you do by removing the filtering. 

km200 - to be honest, I'm not sure what is causing the crash and what actually crashes. I don't like the Tx Delay so it's worth trying something different. Perhaps a few of us should put some cash together and buy a 2nd hand km200 unit? We'd learn a lot.

MichaelDvP, 2021-08-20:
Ok, i'll merge. 
I wasn't sure this is the right way. Filtering messages and set log level seems double. My solution needs saving the loglevel to have the option to log from boot. 
Alternative solution is to keep filtering and always leave internal loglevel "all" for the buffer. Than it's possible to decide later which level should be displayed.

commit 8d4ae697 (2021-08-20): fix #100 add missing hamode to EasyMonitor