emsesp/EMS-ESP32 #101: Version  "3.2.2b0" -  ERROR 3414:   [telegram]   Last Tx Read operation failed after 3 retries...

zibous, 2021-08-20:
**Bug description**
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

**Steps to reproduce**
*Steps to reproduce the behavior.*
--

**Expected behavior**
*A clear and concise description of what you expected to happen.*
The last **Version 3.2.2b0** shows Error on telegramm.
**Version v3.1.2b1** works w/o any error messages.

```log
2021-08-20 10:24:05.751   ERROR 3414:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 F5
2021-08-20 10:53:04.732   ERROR 3415:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 B9
2021-08-20 10:55:03.791   ERROR 3416:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 A5
2021-08-20 10:57:07.299   ERROR 3417:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 02 1B
```

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

[image]

**Device information**
```

{
  "System": {
    "version": "3.2.2b0",
    "uptime": "000+02:03:47.802",
    "uptime_sec": 7427,
    "freemem": 179
  },
  "Status": {
    "bus": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 7909,
    "read requests sent": 1977,
    "write requests sent": 0,
    "incomplete telegrams": 0,
    "tx fails": 11,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT publishes": 2563,
    "MQTT publish fails": 0,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:07.00)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)",
      "handlers": ""
    }
  ]
}
```
**Additional context**
*Add any other context about the problem here.*
Is there a workaround for this ?

proddy, 2021-08-20:
It's normal to see Rx and Tx errors which can be due to multiple issues like noise on the EMS line, timing issues on the ESP32 or just glitches here and there. They can be largely ignored. Your Rx and Tx quality shows 100%.

But I see you haven't been able to send out any Tx yet, so perhaps something else is wrong. You'll need to isolate this. Try changing a setting use the Web UI with the Log set to debug and trace if it works. 

Did it work in previous versions? Are you using a BBQKees S32? Do you really need a 60 second delay on Tx??

MichaelDvP, 2021-08-20:
I think for the ems+ components (RC300) the tx-mode 2 (ems+)  should work better, but 11 errors in 1977 reads ist nothing to worry about.

zibous, 2021-08-20:
@proddy 
Thanks for your replay.

> Are you using a BBQKees S32? Do you really need a 60 second delay on Tx??

Yes BBQKees S32, but i do not change the delay on Tx, it is the default value.

> But I see you haven't been able to send out any Tx yet, so perhaps something else is wrong. You'll need to isolate this. Try changing a setting use the Web UI with the Log set to debug and trace if it works.

Result for debug and trace:
```
2021-08-20 12:35:27.966   TRACE 6231:   [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
2021-08-20 12:35:28.186   TRACE 6232:   [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 39 01 F9 01 FA 01 00 00 03 00 00 07 9B 00 00 BB 00 80 00
2021-08-20 12:35:28.496   TRACE 6233:   [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 00 F4 00 00 00 00 00 40 80 00 01 F9 80 00 00 00 FF 30 48 00 CB 00 00 00
2021-08-20 12:35:28.759   TRACE 6234:   [emsesp]     Thermostat(0x10) -> All(0x00), RC300Monitor(0x2A5), data: 02 72 01 8A (offset 13)
2021-08-20 12:35:30.804   TRACE 6235:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:35:35.077   DEBUG 6236:   [mqtt]       Publishing topic ems-heizung/boiler_data (#5038, retain=0, retry=1, size=643, pid=1)
2021-08-20 12:19:23.644   TRACE 4022:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:23:26.101   TRACE 4581:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:27:28.697   TRACE 5126:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:31:30.170   TRACE 5676:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:35:30.804   TRACE 6235:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:36:03.135   ERROR 6325:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 B9
2021-08-20 12:39:01.715   ERROR 6739:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 A5
2021-08-20 12:39:33.300   TRACE 6799:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:43:33.906   TRACE 7353:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:46:06.934   ERROR 7727:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 AF
2021-08-20 12:47:35.579   TRACE 7918:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:51:38.241   TRACE 8484:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:54:03.665   ERROR 8827:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 A5
2021-08-20 12:55:40.676   TRACE 9030:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 12:59:41.403   TRACE 9579:   [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:03:43.782   TRACE 10137:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:07:46.236   TRACE 10697:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:10:07.422   ERROR 11056:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 01 20 01 9B
2021-08-20 13:11:48.831   TRACE 11256:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:15:49.538   TRACE 11793:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:19:51.083   TRACE 12344:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:23:53.496   TRACE 12893:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:24:06.073   ERROR 12980:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 01 20 02 1B
2021-08-20 13:27:04.216   ERROR 13360:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 01 20 01 A5
2021-08-20 13:27:04.216   ERROR 13360:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 01 20 01 A5
2021-08-20 13:27:55.195   TRACE 13437:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:31:56.775   TRACE 14019:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:31:56.775   TRACE 14019:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:35:58.431   TRACE 14597:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:39:59.269   TRACE 15148:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:42:04.177   ERROR 15431:  [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 01 20 01 A5
2021-08-20 13:44:02.396   TRACE 15714:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
2021-08-20 13:48:02.936   TRACE 16259:  [emsesp]     Thermostat(0x10) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00
[... cut ...]
2021-08-20 17:15:23.255   ERROR 11:     [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 AF
2021-08-20 17:22:19.516 WARNING 12:     [telegram]   Incomplete Rx: FE 08 00 34 00 39 02 58 02 58 21 00 00 03 00 00 07 AF 00 00 BD 00 80 00
2021-08-20 17:22:27.350   ERROR 13:     [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 02 1B
2021-08-20 17:43:21.292   ERROR 14:     [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 90 FF 00 20 01 B9
```

No Errors with Version Version v3.1.2b1

MichaelDvP, 2021-08-20:
> No Errors with Version Version v3.1.2b1

That's clear, v3.1.2b1 had issue #78 and rebooted on any tx-error, not logging it. But the errors are there before, tx hasn't changed. Check power supply, cable connections, tx-mode.

MichaelDvP, 2021-08-21:
TX works in current loop, thin or long ems-cables can cause tx-errors. How is the ems bus connected? If using service-jack, use the cable from BBQKees, not cheap thin audio lines. If connected via ems clamps, use min. 0.5mm^2 lines.

proddy, 2021-08-21:
also @zibous if you're able to get all the readings from the EMS devices and changing settings (like thermostat temp or boiler temps) works then you can safely ignore the errors. It's like looking at Windows 10's event viewer logs - you'll be surprised how many errors

zibous, 2021-08-22:
@proddy 
Thanks, o.k closing eyes and ignore the errors.

@MichaelDvP 
I connected the BBQKees Gateway to the ems bus because I couldn't find a service port. It is interesting that the polarity is important, because if I connect the cables to the EMS BUS incorrectly, the RC310 will be deactivated.

With all cable variants (10cm), regardless of whether 0.5 mm2 cable or data cable (shielded), I always get the error messages. But can change values with the dashboard and these are also transferred and adopted.

[image]

MichaelDvP, 2021-08-23:
> It is interesting that the polarity is important, because if I connect the cables to the EMS BUS incorrectly, the RC310 will be deactivated.

This is strange. I think the tx-errors are not caused by emsesp, the thermostat connection is bad and emsesp miss the thermostat answer. Please check your trace-log: Are there before the tx-errors messages like `Me(0x0B) <- thermostat(0x10),.. `?
If yes, the thermostat does not answer, but emsesp works ok. Double check the cable connection to thermostat.

For understanding: the ems-bus uses voltage modulation for master to device communication and current modulation for device to master communication, direct communication between devices is not possible. A request to thermstat works like this:
1. master sends a poll: 0x0B is allowed to talk
2. we send a request for a telegram from thermostat to the master
3. master repeats the request (so that thermostat can receive it)
4. thermostat sends the reply to master
5. master repeats the reply
6. we send an ack to master
7. master repeats the ack. 

If we receive the message `Me(0x0B) <- thermostat(0x10),.. `, that is the master echo from 3. and indicated that sending works. But if we dont get the thermostat rply, it is counted as tx-error.

zibous, 2021-08-23:
> This is strange. I think the tx-errors are not caused by emsesp, the thermostat connection is bad and emsesp miss the thermostat answer. Please check your trace-log: Are there before the tx-errors messages like `Me(0x0B) <- thermostat(0x10),.. `?

Negativ, can see no errors on `Me(0x0B) <- thermostat(0x10),.. `:

```log2021-08-23 08:26:22.155   ERROR 4068:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 14 0D 20
2021-08-23 08:26:30.470   ERROR 4069:   [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 16 0D 20
2021-08-23 08:28:08.573   TRACE 4070:   [emsesp]     Thermostat(0x10) -> Boiler(0x08), ?(0x23), data: 00 00 00
2021-08-23 08:28:09.533   TRACE 4071:   [emsesp]     Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 00 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
2021-08-23 08:28:09.875   TRACE 4072:   [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 39 02 08 02 07 21 00 00 03 00 00 07 F4 00 00 C1 00 80 00
2021-08-23 08:28:10.229   TRACE 4073:   [emsesp]     Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 05 00 FB 00 00 00 00 00 40 80 00 02 07 80 00 00 00 FF 30 48 00 CB 00 00 00
2021-08-23 08:28:11.041   DEBUG 4074:   [mqtt]       Publishing topic ems-heizung/boiler_data (#7266, retain=0, retry=1, size=645, pid=1)
2021-08-23 08:28:11.143   DEBUG 4075:   [mqtt]       Publishing topic ems-heizung/boiler_data_ww (#7267, retain=0, retry=1, size=490, pid=1)
2021-08-23 08:28:11.244   DEBUG 4076:   [mqtt]       Publishing topic ems-heizung/thermostat_data (#7268, retain=0, retry=1, size=602, pid=1)
2021-08-23 08:28:16.903   DEBUG 4077:   [emsesp]     Fetching values for device ID 0x08
2021-08-23 08:28:16.903   DEBUG 4078:   [telegram]   Tx read request to device 0x08 for type ID 0x14
2021-08-23 08:28:16.903   DEBUG 4079:   [telegram]   Tx read request to device 0x08 for type ID 0x19
2021-08-23 08:28:16.903   DEBUG 4080:   [telegram]   Tx read request to device 0x08 for type ID 0x16
2021-08-23 08:28:16.903   DEBUG 4080:   [telegram]   Tx read request to device 0x08 for type ID 0x16
2021-08-23 08:28:16.903   DEBUG 4081:   [telegram]   Tx read request to device 0x08 for type ID 0x33
2021-08-23 08:28:16.903   DEBUG 4081:   [telegram]   Tx read request to device 0x08 for type ID 0x33
2021-08-23 08:28:16.903   DEBUG 4082:   [telegram]   Tx read request to device 0x08 for type ID 0x26
2021-08-23 08:28:16.903   DEBUG 4082:   [telegram]   Tx read request to device 0x08 for type ID 0x26
2021-08-23 08:28:16.903   DEBUG 4083:   [emsesp]     Fetching values for device ID 0x09
2021-08-23 08:28:16.903   DEBUG 4084:   [emsesp]     Fetching values for device ID 0x10
2021-08-23 08:28:16.903   DEBUG 4085:   [telegram]   Tx read request to device 0x10 for type ID 0x2A5
2021-08-23 08:28:16.903   DEBUG 4085:   [telegram]   Tx read request to device 0x10 for type ID 0x2A5
2021-08-23 08:28:16.903   DEBUG 4086:   [telegram]   Tx read request to device 0x10 for type ID 0x2B9
2021-08-23 08:28:16.903   DEBUG 4087:   [telegram]   Tx read request to device 0x10 for type ID 0x2AF
2021-08-23 08:28:16.903   DEBUG 4088:   [telegram]   Tx read request to device 0x10 for type ID 0x29B
2021-08-23 08:28:16.903   DEBUG 4089:   [telegram]   Tx read request to device 0x10 for type ID 0x2F5
2021-08-23 08:28:16.903   DEBUG 4089:   [telegram]   Tx read request to device 0x10 for type ID 0x2F5
2021-08-23 08:28:16.903   DEBUG 4090:   [telegram]   Tx read request to device 0x10 for type ID 0x31B
2021-08-23 08:28:16.903   DEBUG 4090:   [telegram]   Tx read request to device 0x10 for type ID 0x31B
2021-08-23 08:28:16.903   DEBUG 4091:   [telegram]   Tx read request to device 0x10 for type ID 0x23A
2021-08-23 08:28:16.903   DEBUG 4092:   [telegram]   Tx read request to device 0x10 for type ID 0x240
2021-08-23 08:28:17.199   DEBUG 4093:   [telegram]   Sending read Tx [#110], telegram: 0B 88 14 00 20
2021-08-23 08:28:17.278   TRACE 4094:   [telegram]   Incomplete Rx: 0B 88 14 00 20 E4
2021-08-23 08:28:17.695   DEBUG 4095:   [telegram]   Sending read Tx [#123], telegram: 0B 88 14 00 20
2021-08-23 08:28:17.695   TRACE 4096:   [emsesp]     Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 20
2021-08-23 08:28:17.742   DEBUG 4097:   [emsesp]     Last Tx read successful
2021-08-23 08:28:17.742   TRACE 4098:   [emsesp]     Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 01 9A C1
2021-08-23 08:28:17.742   TRACE 4098:   [emsesp]     Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 01 9A C1
2021-08-23 08:28:18.126   DEBUG 4099:   [telegram]   Sending read Tx [#111], telegram: 0B 88 19 00 20
2021-08-23 08:28:18.126   TRACE 4100:   [emsesp]     Me(0x0B) <- Boiler(0x08), UBAMonitorSlow(0x19), data: 20
2021-08-23 08:28:18.209   DEBUG 4101:   [emsesp]     Last Tx read successful
2021-08-23 08:28:18.209   TRACE 4102:   [emsesp]     Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 00 A5 80 00 80 00 00 00 00 00 00 00 CC 00 09 98 00 00 00 00 01 A4 00 00 0B 80 00
2021-08-23 08:28:18.490   DEBUG 4103:   [telegram]   Sending read Tx [#124], telegram: 0B 88 19 19 20
2021-08-23 08:28:18.507   TRACE 4104:   [emsesp]     Me(0x0B) <- Boiler(0x08), UBAMonitorSlow(0x19), data: 20 (offset 25)
2021-08-23 08:28:18.527   DEBUG 4105:   [emsesp]     Last Tx read successful
2021-08-23 08:28:18.527   TRACE 4106:   [emsesp]     Boiler(0x08) -> Me(0x0B), UBAMonitorSlow(0x19), data: 80 00 (offset 25)
[... cut ...]
2021-08-23 08:26:07.507    INFO 4:      [mqtt]       MQTT connected
2021-08-23 08:26:13.001    INFO 5:      [emsesp]     Stopping Access Point
2021-08-23 08:31:09.551   ERROR 6:      [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 16 00 20
2021-08-23 08:37:08.890   ERROR 7:      [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 26 00 20
```

It also confuses me, because tap water is always 0 l / min and I cannot tell whether the shower is being used.

[image]

bbqkees, 2021-08-23:
@zibous which boiler do you have?

MichaelDvP, 2021-08-23:
In this log there is only the first message an tx-error, but we need to see the messages before to see if it is emsesp or other device.

This is a normal request sequence (send - receive master echo - receive answer and success):
```
2021-08-23 08:28:20.319   DEBUG 4118:   [telegram]   Sending read Tx [#126], telegram: 0B 88 16 19 20
2021-08-23 08:28:20.336   TRACE 4119:   [emsesp]     Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20 (offset 25)
2021-08-23 08:28:20.354   DEBUG 4120:   [emsesp]     Last Tx read successful
2021-08-23 08:28:20.354   TRACE 4121:   [emsesp]     Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: 00 3C 00 (offset 25)
```
Here we have a failed send to the master (echo incomplete) and repeated send with success:
```
2021-08-23 08:28:26.282   DEBUG 4154:   [telegram]   Sending read Tx [#118], telegram: 0B 90 FF 00 20 01 9B
2021-08-23 08:28:26.363   TRACE 4155:   [telegram]   Incomplete Rx: 0B 90 FF 00 20 01 9B 27
2021-08-23 08:28:27.940   DEBUG 4156:   [telegram]   Sending read Tx [#130], telegram: 0B 90 FF 00 20 01 9B
2021-08-23 08:28:27.940   TRACE 4157:   [emsesp]     Me(0x0B) <- Thermostat(0x10), RC300Curves(0x29B), data: 20
2021-08-23 08:28:28.003   DEBUG 4158:   [emsesp]     Last Tx read successful
2021-08-23 08:28:28.003   TRACE 4159:   [emsesp]     Thermostat(0x10) -> Me(0x0B), RC300Curves(0x29B), data: 01 01 00 00 00 02 05 30 3C
```
But since the errors are low it is not really important.

> It also confuses me, because tap water is always 0 l / min and I cannot tell whether the shower is being used.

What kind of warm water system do you have? Systems with buffer does not have a flow-sensor, the burner is switched if the buffer cools down, only flow-systems starts the burner if water flows.

zibous, 2021-08-23:
@MichaelDvP 
> But since the errors are low it is not really important.

Thanks for checking the errors.

> What kind of warm water system do you have? Systems with buffer does not have a flow-sensor, the burner is switched if the buffer cools down, only flow-systems starts the burner if water flows.

LOGAMAX PLUS GB172-14 WITH RC310
BOILER GEMINOX BS 150

[image]

bbqkees, 2021-08-24:
@zibous Your boiler has the service jack underneath a plastic flap on the left of the display (its also mentioned in the wiki)
[image]

zibous, 2021-08-24:
@bbqkees 
Thank you, it is available, but if I plug in the original cable (without connecting it to the gateway), the controller and RC310 are switched off. All displays are then out of order. It looks like the jack socket has a switch for service mode.

Therefore, I only have the option of using the connection via the BUS terminals with a short cable.

I do not get any help from Bosch or the supplier whether there is an error, because I should not connect any third-party device to the BUS or service port, even if the system has only been installed for a month.

I am unsure whether the controller or the RC310 is working properly.

bbqkees, 2021-08-24:
You need to insert the cable fully, there might be a piece of rubber or plastic in the way.
Otherwise the jack cable may shortcut the connector. 
You could also try out another jack cable.

zibous, 2021-08-24:
@bbqkees 
Negative, same problem as with the BUS connection if I do not use the correct polarity.
1. Cable with the straight jack plug connected to the service port.
    - ALL is O.K (Controller and RC310 is working)
2. The angle jack plug is plugged into the gateway
     - LOGAMAX PLUS GB172-14 reboots 
     -  RC310 is then no longer activated (display is dark).

bbqkees, 2021-08-30:
@zibous because the cable does work in one orientation, I think it's the angled jack plug side that does not fit properly into the connector.

proddy, 2021-09-26:
closing. re-open if it still an issue.

closed 2021-09-26 by proddy