emsesp/EMS-ESP32 #103: Same product_id on both Boiler and Thermostat

Sunbuzz, 2021-09-01:
I have run into a situation where I have two devices that use the same product_id.
as EMSESP::add_device() only compares product__id, it randomly updates the devices after each Version(0x02).

I have temporary solved this by adding +1 to the product_id, if the device_id = 0x10, but this is not a nice solution.
`    if (device_id != 0x08 && product_id == 172) product_id += 1;
` 

Here is the output from Version(0x02):

ems-esp:# read 08 02
000+07:56:43.474 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), Version(0x02), data: AC 02 03 AD 02 02 00 00 00 0D

ems-esp:# read 10 02
000+07:56:49.029 N 5: [emsesp] Thermostat(0x10) -> Me(0x0B), Version(0x02), data: AC 02 03

MichaelDvP, 2021-09-01:
Can you please describe your system. Brand 0x0D is unknown, also thermostat type_id 0xAC. I think you have added to your own database. 

I see two possible solutions: 
1. when updating the devices, check also the device_type: change [this line](https://github.com/emsesp/EMS-ESP32/blob/94a71998a79db0fae83e45a5b640cb3341d4fdd1/src/emsesp.cpp#L933) to `if (device.product_id == product_id && device.device_type == emsdevice->device_type()) {` I think this will work, but shows the same id for boiler and thermostat, making possible problems with homeassistant.
2. Your boiler have a second product_id (173) and version (2.02). What version information is shown on the display of your thermostat/controller? If we prefere/register the highest product-id, maybe we have no conficts, but need to update the database with the yet unknown second and third ids. This can be done by by changing [these lines](https://github.com/emsesp/EMS-ESP32/blob/94a71998a79db0fae83e45a5b640cb3341d4fdd1/src/emsesp.cpp#L747-L754) to 
```
    if (telegram->message_length > 8 && telegram->message_data[6] != 0x00) {
        offset = 6;
    } else if (telegram->message_length > 5 && telegram->message_data[3] != 0x00) {
        offset = 3;
    } else if (telegram->message_data[0] == 0x00) {
        return; // ignore whole telegram
    }
 ```

Sunbuzz, 2021-09-01:
I have a IVT Geo 5xx, IVT have been making heatpumps in Sweden for a very long time, but Bosch acquired them a couple of year ago, and on their latest models they have been moving over from CAN-bus to EMS-bus.
The first model that featured an EMS-bus was called Geo 3xx, (which is identical to Bosch Compress 7001i) had an interface they called "Rego 2000" and it looked like this :

[image]

The next generation is called Geo5xx, and now they call the interface "Rego 3000" and it looks like this:

[image]

Which is the model I have. I also have a pool mixer (MP100) which is why I dove into this project, I need to connect an external circulation pump, that I turn on and off when it meets certain conditions.

This is how my system looks with the stock firmware:

ems-esp:$ show devices
These EMS devices are currently active:

(1) Boiler: Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco (DeviceID:0x08, ProductID:172, Version:02.03)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A 

(2) Gateway: KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)

(99) Unknown: unknown (DeviceID:0x50, ProductID:236, Version:03.03)

(99) Unknown: unknown (DeviceID:0x09, ProductID:240, Version:38.03)

(99) Unknown: Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco (DeviceID:0x10, ProductID:172, Version:02.03)

(32) Unknown: unknown (DeviceID:0x11, ProductID:204, Version:26.02)

I have found the product_id (0x10) to be compatible with a RC300 thermostat, and when I make changes on the display it is sent from the src_id (0x09) So I just added a controller (0x09, product_id:240) and that works, but I don't know what to call it.

The MP100 is the 0x11 with product_id:204, and I have made a mixer for it, and it works fine.

So, with these firmware changes it looks like this:

ems-esp:$ show devices
These EMS devices are currently active:

(1) Boiler: IVT Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/IVT Geo 5xx (DeviceID:0x08 ProductID:172, Version:02.03)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A 

(3) Thermostat: IVT Rego 2000/3000 (DeviceID:0x10, ProductID:173, Version:02.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240 

(4) Mixer: MP100 (DeviceID:0x11, ProductID:204, Version:26.02)
 This Mixer will respond to telegram type IDs: 0x5BA 

(5) Gateway: KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)

(2) Controller: Rego 3000 (DeviceID:0x09, ProductID:240, Version:38.03)

(0) Unknown: unknown (DeviceID:0x50, ProductID:236, Version:03.03)

BTW, I did just like that (`if (device.product_id == product_id && device.device_type == emsdevice->device_type())`) before adding +1 to the product_id, but thought I got random names on the boiler/thermostat, maybe I am wrong, I'll try that again and confirm.

Sunbuzz, 2021-09-01:
Your second suggestion seems to work well!
(I have not had the chance to read the version on the display yet)

If I take the secondary? product_id from Version(0x02) I get this after adding the heatpump to the device_list:

ems-esp:$ show devices
These EMS devices are currently active:

(1) Boiler: IVT IVT Geo 5xx (DeviceID:0x08 ProductID:173, Version:02.02)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A 

(3) Thermostat: IVT Rego 2000/3000 (DeviceID:0x10, ProductID:172, Version:02.03) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240 

(4) Gateway: KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)

(2) Controller: Rego 3000 (DeviceID:0x09, ProductID:240, Version:38.03)

(99) Unknown: unknown (DeviceID:0x50, ProductID:236, Version:03.03)

MichaelDvP, 2021-09-01:
The first solution should also work and allow same productID for different devices. The second is difficult to add to standard-software, it breaks the device-recognition for a lot of people and we have to re-add them with the 2nd or 3rd productID, which we don't now yet. Iff there are issues with HA, i think it's easier to make the mqtt-id unique again by adding more information to it.

Do you need any help with the MP100? Or have you all values you need.
Do you know what the deviceID 0x50 is? Can you check what's connetcted to the ems-bus and make a trace/all-log to see the telegrams coming from this device.

Sunbuzz, 2021-09-01:
Aha, I didn't know secondary product_id's existed.
I'll try the first method
I have the pool mixer covered, I think I have been able to map everything. 

In the mean while, here is the readings from the display:

[image]

[image]

Translated:
Display/Panel: NF38.03
Installation board: V02.03 (0x08)
IO board: V02.02 (0x08, secondary product_id)
Pool module: V26.02 (0x11)
Internet Gateway V04.07 (0x48)
Wireless module: V 03.03 (0x50)

I don't know what the "Wireless module" is, I don't think it has wifi, but there is a Wireless thermometer connected (both a receiver & transmitter) but it's awfully quiet on 0x50.

MichaelDvP, 2021-09-01:
> I don't know what the "Wireless module" is.

Is it this? 
[T1RF_6720842575.pdf](https://github.com/emsesp/EMS-ESP32/files/7093026/T1RF_6720842575.pdf)
Then the  outdoortemperature should already be mapped to boiler/thermostat values. But we can add the deviceID to the library to have no unknown devices.

Sunbuzz, 2021-09-01:
No, I don't think thats the one, It is my indoor thermometer, I don't have a model nr for it.
I'll see if I can get it from the service company.

Sunbuzz, 2021-09-02:
I also have a product_d (0x40) who answers to Version(0x02) with no data, is this common?

000+00:02:21.917 N 2: [emsesp] 40(0x40) -> All(0x00), ?(0x435), data: 00 E7
000+00:02:22.355 N 3: [emsesp] Me(0x0B) <- 40(0x40), Version(0x02), data: 20
000+00:02:22.386 N 4: [emsesp] 40(0x40) -> Me(0x0B), Version(0x02), data: <empty>
000+00:02:33.468 N 5: [emsesp] Me(0x0B) <- 40(0x40), Version(0x02), data: 20
000+00:02:33.501 N 6: [emsesp] 40(0x40) -> Me(0x0B), Version(0x02), data: <empty>
000+00:03:22.043 N 7: [emsesp] 40(0x40) -> All(0x00), ?(0x435), data: 00 E7
000+00:03:22.445 N 8: [emsesp] Me(0x0B) <- 40(0x40), Version(0x02), data: 20
000+00:03:22.475 N 9: [emsesp] 40(0x40) -> Me(0x0B), Version(0x02), data: <empty>
000+00:03:31.995 N 10: [emsesp] Me(0x0B) <- 40(0x40), Version(0x02), data: 20
000+00:03:32.024 N 11: [emsesp] 40(0x40) -> Me(0x0B), Version(0x02), data: <empty>

MichaelDvP, 2021-09-03:
It is not very common, but there are also some devices that does not answer to version request, [like modem 0x0D](https://github.com/emsesp/EMS-ESP32/blob/94a71998a79db0fae83e45a5b640cb3341d4fdd1/src/emsesp.cpp#L724)

With the productID 0x40 it seems you have found your wireless sensor, sending every minute a single temperature value (23.1°C). in telegram 0x435. I think 0x50 is the RF-base and 0x40 the sensor. Do you get the same value as roomtemperature from your thermostat, or should we add the sensor to devices.

Sunbuzz, 2021-09-03:
Yes!! You are right, it sends out the current indoor temperature.
So now we know what 0x40 and 0x50 is at least, 0x40 is showing up as a "unknown" but I don't know in which category it would fit.

MichaelDvP, 2021-09-03:
My suggestion is to use connect (or generic) for the base 0x50 i.e.:
```
{236, DeviceType::CONNECT, F("wireless sensor base"), DeviceFlags::EMS_DEVICE_FLAG_NONE}
```
and 0x40 is a bit complicated, i think with empty version messages it is not show because of [this](https://github.com/emsesp/EMS-ESP32/blob/94a71998a79db0fae83e45a5b640cb3341d4fdd1/src/emsesp.cpp#L741). Without product-id we also will not get a database match. Do you need this telegram, or is the value republished from thermostat as root-temperature?
Another problem is that norberts listes 0x40 as powermodule hc9, but a conflict wil be very unlikely.

Sunbuzz, 2021-09-04:
I think this will be fine! I don't need the 0x40 as it is as you say - republished as room temperature.

proddy, 2021-09-28:
@MichaelDvP @Sunbuzz shall we add productID 236 as a CONNECT device as suggested and close this?

Sunbuzz, 2021-09-29:
Yes, of course. I did this immediate in my own build, and forgot to make a pull request.

Thank you.

proddy, 2021-09-29:
no worries, if its just the one line I'll add it now. ok?