emsesp/EMS-ESP32 #105: Unable to change any values on Junkers FW200

simasjoris, 2021-09-04:
ESP-32 Running latest firmware (v3.2.1)

Bug description
My Junkers FW200 controller displays all information correctly, but changing any values has no effect. 
Connection Status shows `Tx Errors` with ~20% TX success rate.
Console also shows TX errors (see attached screenshot).

Tried changing all available TX protocols in settings without any success. Also tried changing  flow, heating and other values 
for boiler device - same TX errors.

Steps to reproduce
Open web console dashboard. Select FW200 thermostat, change temperature or date. 

Expected behavior
Temperature or date changes

Screenshots
[image]

Device information
```
{
  "System": {
    "version": "3.2.1",
    "uptime": "000+00:44:57.787",
    "uptime_sec": 2697,
    "freemem": 167
  },
  "Status": {
    "bus": "connected, instable tx",
    "bus protocol": "HT3",
    "telegrams received": 3433,
    "read requests sent": 100,
    "write requests sent": 4,
    "incomplete telegrams": 0,
    "tx fails": 82,
    "rx line quality": 100,
    "tx line quality": 18,
    "MQTT publishes": 1196,
    "MQTT publish fails": 0,
    "dallas sensors": 0,
    "dallas reads": 0,
    "dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:18.03)",
      "handlers": "0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.07)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x16F 0x170 0x171 0x172 0x165 0x166 0x167 0x168"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x21 ProductID:102, Version:16.64)",
      "handlers": "0x10C 0x1E"
    },
    {
      "type": "Mixer",
      "name": "Junkers IPM (DeviceID:0x20 ProductID:102, Version:16.64)",
      "handlers": "0x10C 0x1E"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:18.03)",
      "handlers": ""
    }
  ]
}
```

proddy, 2021-09-04:
are you able to change boiler settings?
did it work in previous firmware versions or is this the first time you tried it?
also which hardware are you using and how is it connected to the EMS lines.

simasjoris, 2021-09-04:
I'm using EMS Gateway S32-LR connected to the BUS interface in the distribution box:
 
[image]

The Distribution box is about a meter away from the gas heater with integrated FW200 thermostat.

At first I tried using the firmware that came originally with the device (think it was v3.0.x), but that version did not even have an option to edit any values in the dashboard. I've updated it to the latest earlier today.

Unfortunately no luck with boiler settings either, I can read them but changes are not applied and errors are thrown in the console.

proddy, 2021-09-04:
Ok, thanks for sharing. We'll come back to you with some next steps to debug

bbqkees, 2021-09-04:
Can you try and move the Gateway to the orange BUS terminal?
Your TX is very low. Maybe that will help.

simasjoris, 2021-09-05:
Thanks for a suggestion, but after switching to orange BUS terminal the issue persists - I can read the values, but not change them.

Correct me if I'm wrong, but from the following logs it seems that some TX commands are successful (I'm assuming they are value read requests), while manual TX commands always fail. 
```
000+00:17:08.397   DEBUG 732:    [telegram]   Sending read Tx [#132], telegram: 8B 88 14 00 20
000+00:17:08.436   DEBUG 733:    [emsesp]     Last Tx read successful
000+00:17:10.018   DEBUG 734:    [mqtt]       Publishing topic ems-esp/boiler_data (#510, retain=0, retry=1, size=595, pid=1)
000+00:17:10.119   DEBUG 735:    [mqtt]       Publishing topic ems-esp/boiler_data_ww (#511, retain=0, retry=1, size=454, pid=1)
000+00:17:10.221   DEBUG 736:    [mqtt]       Publishing topic ems-esp/thermostat_data (#512, retain=0, retry=1, size=181, pid=1)
000+00:17:10.323   DEBUG 737:    [mqtt]       Publishing topic ems-esp/mixer_data (#513, retain=0, retry=1, size=140, pid=1)
000+00:17:11.390   DEBUG 738:    [telegram]   Sending read Tx [#133], telegram: 8B 88 19 00 20
000+00:17:11.456   DEBUG 739:    [emsesp]     Last Tx read successful
000+00:17:13.229   DEBUG 740:    [telegram]   Sending read Tx [#187], telegram: 8B 88 19 19 20
000+00:17:13.268   DEBUG 741:    [emsesp]     Last Tx read successful
000+00:17:15.380   DEBUG 742:    [telegram]   Sending read Tx [#134], telegram: 8B 88 16 00 20
000+00:17:15.431   DEBUG 743:    [emsesp]     Last Tx read successful
000+00:17:17.819   DEBUG 744:    [telegram]   Sending read Tx [#135], telegram: 8B 88 33 00 20
000+00:17:17.863   DEBUG 745:    [emsesp]     Last Tx read successful
000+00:17:19.745   DEBUG 746:    [telegram]   Sending read Tx [#136], telegram: 8B 90 FF 00 20 00 6F
000+00:17:20.095   DEBUG 747:    [mqtt]       Publishing topic ems-esp/boiler_data (#514, retain=0, retry=1, size=593, pid=1)
000+00:17:20.196   DEBUG 748:    [mqtt]       Publishing topic ems-esp/boiler_data_ww (#515, retain=0, retry=1, size=454, pid=1)
000+00:17:20.298   DEBUG 749:    [mqtt]       Publishing topic ems-esp/thermostat_data (#516, retain=0, retry=1, size=181, pid=1)
000+00:17:20.399   DEBUG 750:    [mqtt]       Publishing topic ems-esp/mixer_data (#517, retain=0, retry=1, size=140, pid=1)
000+00:17:22.396   DEBUG 751:    [telegram]   Sending read Tx [#188], telegram: 8B 90 FF 00 20 00 6F
000+00:17:26.107   DEBUG 752:    [telegram]   Sending read Tx [#189], telegram: 8B 90 FF 00 20 00 6F
000+00:17:27.010   DEBUG 753:    [telegram]   Sending read Tx [#190], telegram: 8B 90 FF 00 20 00 6F
000+00:17:27.082    INFO 754:    [command]    Calling thermostat command 'datetime', value 11:13:06 05/09/2021, id is default
000+00:17:27.082    INFO 755:    [thermostat] Date and time: 05.09.2021-11:13:06
000+00:17:27.084    INFO 756:    [thermostat] Setting date and time
000+00:17:27.316   ERROR 757:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 00 6F
000+00:17:30.088   DEBUG 758:    [mqtt]       Publishing topic ems-esp/boiler_data (#518, retain=0, retry=1, size=593, pid=1)
000+00:17:30.189   DEBUG 759:    [mqtt]       Publishing topic ems-esp/boiler_data_ww (#519, retain=0, retry=1, size=454, pid=1)
000+00:17:30.290   DEBUG 760:    [mqtt]       Publishing topic ems-esp/thermostat_data (#520, retain=0, retry=1, size=181, pid=1)
000+00:17:30.392   DEBUG 761:    [mqtt]       Publishing topic ems-esp/mixer_data (#521, retain=0, retry=1, size=140, pid=1)
000+00:17:30.790   DEBUG 762:    [telegram]   Sending write Tx [#191], telegram: 8B 10 06 00 15 09 0B 05 0D 06 48 7F
000+00:17:30.828   DEBUG 763:    [emsesp]     Last Tx write successful
000+00:17:30.828   DEBUG 764:    [telegram]   Sending post validate read, type ID 0x06 to dest 0x10
000+00:17:31.750   DEBUG 765:    [telegram]   Sending read Tx [#192], telegram: 8B 90 06 00 20
000+00:17:32.104    INFO 766:    [command]    Calling thermostat command 'hc1/seltemp', value 25, id is default
000+00:17:32.106    INFO 767:    [thermostat] Setting thermostat temperature to 25.0 for heating circuit 1, mode auto
000+00:17:35.279   DEBUG 768:    [telegram]   Sending write Tx [#194], telegram: 8B 10 FF 10 00 65 32
000+00:17:36.426   DEBUG 769:    [telegram]   Sending write Tx [#195], telegram: 8B 10 FF 10 00 65 32
000+00:17:39.176    INFO 770:    [command]    Calling thermostat command 'hc2/seltemp', value 25, id is default
000+00:17:39.178    INFO 771:    [thermostat] Setting thermostat temperature to 25.0 for heating circuit 2, mode auto
000+00:17:39.576   DEBUG 772:    [telegram]   Sending write Tx [#197], telegram: 8B 10 FF 10 00 66 32
000+00:17:39.608   DEBUG 773:    [emsesp]     Last Tx write successful
000+00:17:39.608   DEBUG 774:    [telegram]   Sending post validate read, type ID 0x170 to dest 0x10
000+00:17:40.009   DEBUG 775:    [mqtt]       Publishing topic ems-esp/boiler_data (#522, retain=0, retry=1, size=593, pid=1)
000+00:17:40.110   DEBUG 776:    [mqtt]       Publishing topic ems-esp/boiler_data_ww (#523, retain=0, retry=1, size=454, pid=1)
000+00:17:40.212   DEBUG 777:    [mqtt]      
000+00:17:45.008   DEBUG 780:    [telegram]   Sending read Tx [#199], telegram: 8B 90 FF 00 20 00 70
000+00:17:48.556   DEBUG 781:    [telegram]   Sending read Tx [#200], telegram: 8B 90 FF 00 20 00 70
000+00:17:50.008   DEBUG 782:    [mqtt]       Publishing topic ems-esp/boiler_data (#526, retain=0, retry=1, size=595, pid=1)
[... cut ...]
000+00:17:50.314   DEBUG 785:    [mqtt]       Publishing topic ems-esp/mixer_data (#529, retain=0, retry=1, size=140, pid=1)
000+00:17:51.176   DEBUG 786:    [telegram]   Sending read Tx [#201], telegram: 8B 90 FF 00 20 00 70
000+00:17:51.455   ERROR 787:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 00 70
```

Could it be the protocol issue? I'm mostly testing with HT3, but also tried changing to EMS or EMS+, without any observable differences.

proddy, 2021-09-05:
Yes it looks like some of the Tx is getting through. What you could try is going into the console, then `su` and  `log all`  and typing `call thermostat temp 18`  a few times to see the success rate.

simasjoris, 2021-09-05:
I've tried sending the `call thermostat temp 18` command 5 times. Here are the logs from the console:
```

┌───────────────────────────────────────
│ EMS-ESP version 3.2.1                │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└───────────────────────────────────────

ems-esp:$ su
Password:
000+00:06:29.728 N 0: [shell] su session opened on console
000+00:06:30.791 E 1: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 00 6F
000+00:06:40.415 E 2: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 00 70
ems-esp:# log all
Log level: all
000+00:06:43.813 T 3: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 27 46 00 00 F6 03 01 03 64 0A 04
000+00:06:44.031 T 4: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 36 46 00 00 F6 03 01 03 64 0A 04
000+00:06:44.375 T 5: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 2C 46 00 00 F6 03 01 03 64 0A 04
000+00:06:44.593 T 6: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 33 46 00 00 F6 03 01 03 64 0A 04
000+00:06:44.811 T 7: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 2E 46 00 00 F6 03 01 03 64 0A 04
000+00:06:45.062 T 8: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 30 46 00 00 F6 03 01 03 64 0A 04
000+00:06:45.279 T 9: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 33 46 00 00 F6 03 01 03 64 0A 04
000+00:06:45.541 D 10: [telegram] Sending read Tx [#177], telegram: 8B 90 FF 00 20 00 65
000+00:06:45.560 T 11: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x165), data: 20
000+00:06:46.058 T 12: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 28 46 00 00 F6 03 01 03 64 0A 04
000+00:06:46.277 T 13: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 39 46 00 00 F6 03 01 03 64 0A 04
000+00:06:46.590 T 14: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 26 46 00 00 F6 03 01 03 64 0A 04
000+00:06:46.807 T 15: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 3A 46 00 00 F6 03 01 03 64 0A 04
000+00:06:47.025 T 16: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 24 46 00 00 F6 03 01 03 64 0A 04
000+00:06:47.337 T 17: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 3A 46 00 00 F6 03 01 03 64 0A 04
000+00:06:47.555 T 18: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 25 46 00 00 F6 03 01 03 64 0A 04
000+00:06:47.773 T 19: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 39 46 00 00 F6 03 01 03 64 0A 04
000+00:06:48.055 T 20: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 25 46 00 00 F6 03 01 03 64 0A 04
000+00:06:48.273 T 21: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 39 46 00 00 F6 03 01 03 64 0A 04
000+00:06:48.512 T 22: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 22 01 A4 46 2B 09 13 A5 C0 80 00 80 00 01 48 FF FF 0C 00 00 00 00 00 02 18
000+00:06:48.727 D 23: [telegram] Sending read Tx [#178], telegram: 8B 90 FF 00 20 00 65
000+00:06:48.746 T 24: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x165), data: 20
ems-esp:# call thermostat temp 18
000+00:06:49.076 I 25: [command] Calling thermostat command 'temp', value 18, id is default
000+00:06:49.076 I 26: [thermostat] Setting thermostat temperature to 18.0 for heating circuit 1, mode auto
000+00:06:49.076 T 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 F6 01 F6 81 00 05 03 00 02 BB D4 00 20 F9 00
000+00:06:49.301 T 28: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 35 46 00 00 F6 03 01 03 64 0A 04
000+00:06:49.521 T 29: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 2C 46 00 00 F6 03 01 03 64 0A 04
000+00:06:49.833 T 30: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 32 46 00 00 F6 03 01 03 64 0A 04
000+00:06:50.019 D 31: [mqtt] Publishing topic ems-esp/boiler_data (#262, retain=0, retry=1, size=594, pid=1)
000+00:06:50.050 T 32: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 2F 46 00 00 F6 03 01 03 64 0A 04
000+00:06:50.120 D 33: [mqtt] Publishing topic ems-esp/boiler_data_ww (#263, retain=0, retry=1, size=454, pid=1)
000+00:06:50.221 D 34: [mqtt] Publishing topic ems-esp/thermostat_data (#264, retain=0, retry=1, size=181, pid=1)
000+00:06:50.313 D 35: [telegram] Sending write Tx [#180], telegram: 8B 10 FF 10 00 65 24
000+00:06:50.331 T 36: [emsesp] Me(0x0B) -> Thermostat(0x10), JunkersSet(0x165), data: 24 (offset 16)
000+00:06:50.331 D 37: [mqtt] Publishing topic ems-esp/mixer_data (#265, retain=0, retry=1, size=138, pid=1)
000+00:06:50.342 D 38: [emsesp] Last Tx write successful
000+00:06:50.342 D 39: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+00:06:50.550 T 40: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 33 46 00 00 F6 03 01 03 64 0A 04
000+00:06:50.767 T 41: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 2A 46 00 00 F6 03 01 03 64 0A 04
000+00:06:51.080 T 42: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 37 46 00 00 F6 03 01 03 64 0A 04
000+00:06:51.297 T 43: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 28 46 00 00 F6 03 01 03 64 0A 04
[... cut ...]
000+00:07:27.938 T 247: [emsesp] Mixer(0x21) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 (offset 2)
000+00:07:28.001 T 248: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 37 46 00 00 F6 03 01 03 64 0A 04
000+00:07:28.219 T 249: [emsesp] Boiler(0x08) -> All(0x00), UBAParameters(0x16), data: FF 26 46 00 00 F6 03 01 03 64 0A 04
ems-esp:#
```
I can't see any other error messages except for the usual ` Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 00 6F` ones. Does that mean that the temperature set commands are actually sent to the device?

proddy, 2021-09-05:
yes, looks like it works most of the times. Is the thermostat actually set to 18? You can check with the `show` command

simasjoris, 2021-09-05:
No, unfortunately it is still stuck at 22:
```
ems-esp:$ show
EMS-ESP version 3.2.1

Boiler: HT3 (DeviceID:0x08, ProductID:95, Version:18.03)
  heating active: off
  warm water active: off
  selected flow temperature: 32 °C
  burner selected max power: 70 %
  heating pump modulation: 0 %
  outside temperature: 13.3 °C
  current flow temperature: 33.0 °C
  return temperature: 35.0 °C
  mixing switch temperature: 29.7 °C
  system pressure: 1.2 bar
  gas: off
  heating pump: off
  fan: off
  ignition: off
  heating activated: on
  heating temperature: 50 °C
  burner pump max power: 100 %
  burner pump min power: 10 %
  pump delay: 3
  burner min period: 3
  burner min power: 0 %
  burner max power: 70 %
  hysteresis on temperature: -10 °C
  hysteresis off temperature: 0 °C
  burner current power: 0 %
  # burner starts: 194560
  total burner operating time: 1343 days 12 hours 52 minutes
  total heat operating time: 1219 days 2 hours 41 minutes
  total UBA operating time: 1875 days 17 hours 58 minutes
  service code number: 0
  maintenance scheduled: off
  maintenance set time: 2300 hours
  ww set temperature: 54
  ww type: buffer
  ww circulation active: off
  ww current intern temperature: 64.6
  ww current extern temperature: 64.6
  ww current tap water flow: 0.0
  ww one time charging: off
  ww disinfecting: off
  ww charging: off
  ww recharging: off
  ww temperature ok: on
  ww active: off
  ww heating: off
  ww # starts: 8442
  ww active time: 124 days 10 hours 11 minutes

Thermostat: Junkers FW200 (DeviceID:0x10 ProductID:106, Version:12.07)
  date/time: 14:42:27 05/09/2021
  hc1 selected room temperature: 22.0
  hc1 current room temperature: 34.0
  hc1 mode type: heat
  hc2 selected room temperature: 22.0
  hc2 current room temperature: 34.0
  hc2 mode type: heat

Controller: HT3 (DeviceID:0x09, ProductID:95, Version:18.03)
```

proddy, 2021-09-05:
It's strange that the value is not being set. I see
```
ems-esp:# call thermostat temp 18
000+00:07:00.477 I 110: [thermostat] Setting thermostat temperature to 18.0 for heating circuit 1, mode auto
000+00:07:01.009 D 113: [telegram] Sending write Tx [#196], telegram: 8B 10 FF 10 00 65 24
000+00:07:01.026 D 114: [emsesp] Last Tx write successful
000+00:07:01.026 D 115: [telegram] Sending post validate read, type ID 0x16F to dest 0x10
000+00:07:01.026 T 116: [emsesp] Me(0x0B) -> Thermostat(0x10), JunkersSet(0x165), data: 24 (offset 16)
```
the value 0x24 (which is 18 * 2) is sent and validated. 

@MichaelDvP any idea what could be happening?

simasjoris, 2021-09-05:
I also have[ junkers FB 10](https://www.manualscat.com/en/junkers-fb-10-manual?p=1) remote thermostat that was used originally to control the temperature remotely. I wanted to try setting values for it as an alternative, but it doesn't seem to be detected in the device list:
```
Unknown: Junkers unknown (DeviceID:0x18 ProductID:109, Version:16.01)
```
It looks just like the FR 10 thermostat, which is in the supported devices list. Perhaps it would be possible to reuse the same commands for FB 10 as a workaround?

simasjoris, 2021-09-05:
I actually managed to change a setting! 
Boiler has properties `maintenance scheduled` and `maintenance set time` which do persist after changing. They are the only changeable settings though - all other setting change back to original values.

MichaelDvP, 2021-09-05:
> the value 0x24 (which is 18 * 2) is sent and validated.

Sent: yes, validated: no, the validation is sent out, but the reply is missing.
Looks like the thermostat is not responding to the monitor telegrams, 0x16F, 0x170, etc. 

@simasjoris Please do on terminal the commands `read 10 16F`, `read 10 170`, also check settings with `read 10 165` and also the monitor from the remote thermostat `read 18 16F`.

Maybe junkers remote thermostats act in a different way and values are not monitored by master. Or the thermostat uses different monitor telegrams, then we need a longer (some minutes) log, mybe with `watch 10` in terminal. Some telegrams from the remote thermostat (`watch 18`)  could also help.

@proddy The remote thermostat should be added to device-lib after the actual pr is merged. First we also need to test which commands it responses to. 
@simasjoris Please post also he reply to `read 18 179` (older telegram-type) and `read 18 165` (newer telegram-type).

simasjoris, 2021-09-05:
@proddy  and @MichaelDvP, I appreciate your effort to help me out! It's truly interesting to delve into the secret communications of the heating system in my house :)

 I've tried the read commands, but it looks like there was no output. I've also tried them after enabling `watch 10` and `watch 18` - here's the full log:
```

┌───────────────────────────────────────
│ EMS-ESP version 3.2.1                │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└───────────────────────────────────────

ems-esp:$ su
Password:
000+05:24:51.683 N 0: [shell] su session opened on console
000+05:24:52.989 E 1: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 01 20 00 65
ems-esp:# read 10 16F
ems-esp:# read 10 170
ems-esp:# read 10 165
ems-esp:# read 18 16F
000+05:25:17.188 E 2: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 FF 01 20 00 6F
ems-esp:# watch 10
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x10
000+05:25:23.890 N 3: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x165), data: 20
000+05:25:24.837 N 4: [emsesp] Thermostat(0x10) <- Mixer(0x20), IPMStatusMessage(0x10C), data: 01
000+05:25:27.195 N 5: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x165), data: 20
000+05:25:28.136 N 6: [emsesp] Thermostat(0x10) <- Mixer(0x20), ?(0x10D), data: 01
000+05:25:28.161 N 7: [emsesp] Mixer(0x20) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+05:25:31.067 N 8: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersSet(0x165), data: 20
000+05:25:31.344 E 9: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 01 20 00 65
000+05:25:31.576 N 10: [emsesp] Thermostat(0x10) <- Mixer(0x21), IPMStatusMessage(0x10C), data: 01
000+05:25:31.606 N 11: [emsesp] Mixer(0x21) -> Thermostat(0x10), IPMStatusMessage(0x10C), data: 02
000+05:25:33.868 N 12: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
ems-esp:# read 10 16F
000+05:25:34.820 N 13: [emsesp] Thermostat(0x10) <- Mixer(0x21), ?(0x10D), data: 01
000+05:25:34.847 N 14: [emsesp] Mixer(0x21) -> Thermostat(0x10), ?(0x10D), data: <empty>
000+05:25:37.111 N 15: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:38.058 N 16: [emsesp] Thermostat(0x10) <- Boiler(0x08), Version(0x02), data: 0A
000+05:25:38.077 N 17: [emsesp] Boiler(0x08) -> Thermostat(0x10), Version(0x02), data: 5F 12 03 00 00 00 00 00 00 00
000+05:25:40.104 N 18: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:41.071 N 19: [emsesp] Thermostat(0x10) <- Mixer(0x20), Version(0x02), data: 0A
000+05:25:41.120 N 20: [emsesp] Mixer(0x20) -> Thermostat(0x10), Version(0x02), data: 66 10 40 12 34 56 12 34 56 02
000+05:25:41.207 N 21: [emsesp] Unknown(0x18) -> Thermostat(0x10), ?(0x123), data: 00 D0
000+05:25:44.096 N 22: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:44.378 E 23: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 01 20 00 6F
000+05:25:45.043 N 24: [emsesp] Thermostat(0x10) <- Mixer(0x21), Version(0x02), data: 0A
000+05:25:45.087 N 25: [emsesp] Mixer(0x21) -> Thermostat(0x10), Version(0x02), data: 66 10 40 12 34 56 12 34 56 02
000+05:25:47.588 N 26: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:48.533 N 27: [emsesp] Thermostat(0x10) <- Unknown(0x18), Version(0x02), data: 0A
000+05:25:48.579 N 28: [emsesp] Unknown(0x18) -> Thermostat(0x10), Version(0x02), data: 6D 10 01 12 34 56 12 34 56 02
000+05:25:51.330 N 29: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:52.290 N 30: [emsesp] Thermostat(0x10) -> All(0x00), RCTime(0x06), data: 15 09 14 05 07 10 00 01
000+05:25:54.573 N 31: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:55.517 N 32: [emsesp] Thermostat(0x10) <- Boiler(0x08), UBAParameters(0x16), data: 0C
000+05:25:55.540 N 33: [emsesp] Boiler(0x08) -> Thermostat(0x10), UBAParameters(0x16), data: FF 35 46 00 00 F6 03 01 03 64 0A 04
000+05:25:57.567 N 34: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:25:57.849 E 35: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 01 20 00 6F
000+05:25:58.526 N 36: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x16F), data: 03 02 00 DC 01 48 00 D0 0A
000+05:26:02.052 N 37: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x170), data: 03 02 00 DC 01 48 F3 34 0A
000+05:26:05.084 N 38: [emsesp] Unknown(0x18) <- Thermostat(0x10), Version(0x02), data: 0A
000+05:26:05.119 N 39: [emsesp] Thermostat(0x10) -> Unknown(0x18), Version(0x02), data: 6A 0C 07 00 00 00 00 00 00 02
000+05:26:07.543 N 40: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:26:11.193 N 41: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:26:15.028 N 42: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:26:19.269 N 43: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x16F), data: 20
000+05:26:19.551 E 44: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 01 20 00 6F
000+05:26:23.260 N 45: [emsesp] Me(0x0B) <- Thermostat(0x10), JunkersMonitor(0x170), data: 20
[... cut ...]
000+05:28:21.032 N 118: [emsesp] Me(0x0B) <- Unknown(0x18), JunkersSet(0x165), data: 20
000+05:28:21.315 E 119: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 98 FF 01 20 00 65
000+05:28:30.309 N 120: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 23 01 75 46 00 01 03 20 C0 80 00 80 00 01 5B FF FF 0C 00 00 00 00 00 00 00
000+05:28:34.019 N 121: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 23 01 71 46 00 01 03 20 C0 80 00 80 00 01 5D FF FF 0C 00 00 00 00 00 00 00
ems-esp:#
```

MichaelDvP, 2021-09-06:
That's strange, the telegrams are right, the thermostat publishes the monitors for hc1 and 2:
```
000+05:25:58.526 N 36: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x16F), data: 03 02 00 DC 01 48 00 D0 0A
000+05:26:02.052 N 37: [emsesp] Thermostat(0x10) -> All(0x00), JunkersMonitor(0x170), data: 03 02 00 DC 01 48 F3 34 0A
```
The remote thermostat only sends the actual roomtemperature in telegram 123 (and have no telegrams 0x16F, ff.):
```
000+05:25:41.207 N 21: [emsesp] Unknown(0x18) -> Thermostat(0x10), ?(0x123), data: 00 D0
```
The roomtemp is republished in pos 6 of thermostat 0x16F in an extra field. We need some logic to take this as room-temperature, but the value 0xF334 for hc without remote-thermostat seems strange (-327.6°C). 

But the thermostat does not reply to any request from emsesp. It's not a physical issue, telegrams are sent out and repeated by the master, but the thermostat ignore them. I have no clue why. A shot in the dark: Try to change the emsesp-address to 0x0D (modem), maybe your thermostat doesn't like the service-key.

simasjoris, 2021-09-06:
Changing to 0x0D (modem) did not have any observable effect, but after double-checking the boiler properties I've managed to find two more values that I can change - `selected flow temperature` and `ww circulation pump frequency`. I must have missed  them somehow as it takes a couple of refreshes for the values to appear.

It looks to me that thermostat properties `hc1/hc2 current room temperature` are most likely mislabeled. I think they actually refer to heated water temperature for perimeters 1 and 2, as their values `30.4` are similar to boiler's `current flow temperature` 

Ideally I would like to be able to control the hot water temperature in the system so that I can map it with a remote temperature sensor via home assistant. I'm hoping I can achieve that with the previously mentioned `selected flow temperature` in the boiler, but perhaps FB 10 thermostat will have some other properties that I could tweak?

I'll be eagerly waiting for EMS-ESP updates to support it! Please let me know if I can help in any way.

MichaelDvP, 2021-09-06:
as mentioned the current room temperature is taken from the thermostat 0x16F, 0x170 pos 4, which is 32.8°C (0x148) for both hc. With the remote FB10 we have to take pos6, but without reading the settings telegram that is not possible. 

In the [full log from here](https://github.com/emsesp/EMS-ESP32/issues/105#issuecomment-913133721) the thermostat sems to me very quiet. I think there is something wrong in configuration. 

I don't understand the system, there are hc1 and hc2, hc1 have a remote roomcontroller FB10, hc2 not, On IPM channel 1 (hc1) there is nothing connected at all, hc2 is mixed, but have no remote. Why there is a IPM2 if channel 1 is not used?  

Without the possibility to read the settings via emsesp, you have to check and note  all expert-settings in the FW200 and post it.  Also the hydraulic plan would be helpfull.

MichaelDvP, 2021-09-06:
Hm, i guess the FW200 is builtin in the boiler (high value of internal sensor) and in factory state. It is configured as dumb configuration panel for the boiler and does not control any heatingcircuit. 
This fits to:
- it does not send boilertemp to the boiler
- it does not send pump/temperature commands to mixer. 
both is needed to control a circuit. And therefore we can't set/query any value from thermostat.

simasjoris, 2021-09-06:
> FW200 is builtin in the boiler (high value of internal sensor) 
@MichaelDvP Yes, that is most likely the case. So in order to control the heating in the house I should change the boiler values, right?
Other option would be to control the target temperature for FP10 remotely, triggering the heater to turn on or off.

Here's the plan of the heating system: 
[Heating system plan.pdf](https://github.com/emsesp/EMS-ESP32/files/7116370/Heating.system.plan.pdf)

As for the expert-settings, do you mean the settings available in FW200 console after entering the service menu?

MichaelDvP, 2021-09-06:
The ipm2 is missing in the plan. hc2 is a mixed circuit and pump/mixinvalve is connected to ipm2-hc2. Where is the pump for hc1 connected, is it builtin the boiler? Is hot water managed by charge pump or 3-way-valve?

I only find german junkers-manuals for FW200, The hc configuration is [here, page 45, 8.1.1](https://www.manualslib.de/manual/44138/Junkers-Fw-200.html?page=45#manual). Systemconfiguration: press menu button for > 3 seconds.

simasjoris, 2021-09-06:
Pump 1 and 2 are separate from the boiler, here are the pics:
[image]
[image]
[image]
[image]

Configuration parameters from FW200:
DHW configuration - combi heater
Circulation pump - present
Heating circuit 1 configuration - unmixed, with IPM
Heating circuit 1 remote control - FB10
Heating circuit 2 configuration - mixed
Heating circuit 2 remote control - no
ISM 1 - no
ISM 2 - no

MichaelDvP, 2021-09-07:
Look ok for me, but i don't understand how hc1 IPM works without pump connected to IPM-P1. I also don't understand why the FW200 does not send any command to the devices and do not answer to requests. 
Sorry, but i give up, i have no further ideas what to test.
@proddy Do you have other ideas or do you know someone with a FW200, who can cross-check settings.

bbqkees, 2021-09-07:
Perhaps HC1 is just the whole water circuit connected to the boiler (using the boiler pump) and HC2 is switched in and out the water circuit with the 3-way valve.

MichaelDvP, 2021-09-07:
I also thought of that, but hc1 is configured as "unmixed, with IPM" and for this the connection should be [like this](https://www.libble.de/junkers-ipm-2/p/550443/?page=0027).

bbqkees, 2021-09-07:
The Wilo pump in the pictures is for drinking water only, it seems to be in the boiler circuit?
So maybe we are assuming the installer did a good job instead of just messing around?

proddy, 2021-09-07:
> Look ok for me, but i don't understand how hc1 IPM works without pump connected to IPM-P1. I also don't understand why the FW200 does not send any command to the devices and do not answer to [...]

The only other FW200 user I know is @FredericMa

FredericMa, 2021-09-07:
Over here everything seems to work fine.
I changed the mode from `nofrost` to `heat`. A few seconds later, I see the change being received via MQTT in home assistant.
Afterwards, the heater fires up, the pump of the heating circuit is activated and all goes as plan. Changing the mode back to `nofrost` afterwards also works as expected; everything is turned of again. Also changing the heating circuit selected temperature works without issues.

My TX error count is at 291 with an uptime of 24 days. Only 1 rx error occured.
Stats: 
```
Telegrams Received | 2,714,820 (quality 100%)
Telegrams Sent | 380,452 (quality 100%)
```

My pumps are connected to an IPM1 which came in a set of Junkers:
[image]

If there is anything else I can test to help, let me know.

simasjoris, 2021-09-08:
```
Perhaps HC1 is just the whole water circuit connected to the boiler (using the boiler pump) and HC2 is switched in and out the water circuit with the 3-way valve.
```
If I had to guess, Willo pump is the circulator for warm water, while HC1 and HC2 are two heating circuits actuated by the same Grundfos pump and Herz 3-way valve to switch between them.

@FredericMa , does your FW 200 have the physical switch to select between modes? If so, does web UI override the hardware setting?
[image]

I'll test a bit more to see if I can control the heating of the house with `selected flow temperature` for the boiler - setting it to a temp higher than current turns on the heater, but a question remains if that triggers the heating pump as well.

FredericMa, 2021-09-08:
> does your FW 200 have the physical switch to select between modes? If so, does web UI override the hardware setting?

@simasjoris Yes, my my thermostat has exact the same switch. EMS-ESP overrides the hardware settings.
As you can see below, the switch is on auto while the display shows 'vorst' (=`nofrost`). The language is set to dutch.
Also, from the moment EMS-ESP controls the system, 'netcom actief' (=`netcom active`) appears asswell as the NC icon on the bottom right.

My thermostat is mounted to the wall in our living room and is not integrated in the heater. Did you try to connect the thermostat to the bus instead using it integrated in the heater to check if that makes any difference?

[image]

norberts1, 2021-09-09:
I'll support you with my little acknowledge. 
@simasjoris please check the display on your FW200 showing that icon: 'NC' as it is visible at @FredericMa picture.
If this is not displayed than the reason for that faulty handling and not answered requests could be the 'FD-number', in German: 'FertigungsDatum'.
@simasjoris  your FW200 has 'FD790' -->> October 2007 and your gas-heater ZSBR28 'FD887' -->> Juli 2008. So the including software is a bit old for handling requests coming from external 'modems' like ESM-ESP, or MBLan2 etc.
But anyway, receiving and decoding data from your heater-system is possible without data-requests.
Attached you'll find two pictures for that FD-number handling and also that minimal required FD-number for controlling the heater-sytsem, mainly that 'thermostat'.

FD-number list:
[image]

Minimal FD-Requirements (FD889 -->> 09/2008):
[image]

That FD-numbers are required for that Fxyz-series of 'thermostats' for controlling the heater-system. They aren't required for receiving and decoding heater-data send by the heater-bustypes 'HT3, HT4i'.
For Cxyz thermostats (EMS2) the FD-number is not important, they all an be controlled by modems.

@proddy That list above could be placed in your wiki for Junkers and Fxyz - thermostats

norberts1, 2021-09-09:
@simasjoris your heater is a combiheater, so hotwater is created by the heater itself using that internal 3way-valve and the internal pump. The wilo-pump is used here as circulationpump.
For the heating that 3way-valve is switching back to the heater-position and using the internal pump as heatercircuit-pump, I'll thing mainly for that HC1.
For HC2 that Grundfos-pump and that Herz-mixer are used to handle the temperatur on your Floor mounted radiators. This is controlled by the IPM2 modul and the second switch on it. This is set to position 2.
For the HC1 the first switch on it is set to position1 as your configuration tells -->> HC1 configuration - unmixed with IPM. In this case only the themperatur-sensor input (VF on the blue block) is used and no pump is connected. 
I'm not sure where this themperatur-sensor is located, but in most cases it is used to measure the themperatur on a 'hydraulic switch' (Hydraulische Weiche) to get the correct temperatur for the HC1 and your Wall mounted radiators.
This themperatur is always above that one required for HC2 and the mixer and the IPM2 handle this.

@simasjoris  @bbqkees  The busconnection on an IPM2 must be allways at that white BUS BB port. 
At that orange BB port are slitley different signals (not identical to the white BB port).

MichaelDvP, 2021-09-09:
@norberts1 I'm always impressed of your knowledge, thank you very much for sharing.

proddy, 2021-09-09:
> @norberts1 I'm always impressed of your knowledge, thank you very much for sharing.

me too! amazing. You and Michael should set up your own consultancy company.

simasjoris, 2021-09-10:
Thanks everyone for sharing those details, especially @norberts1 - Now I know why I'm only able to read the data and not set any values for the thermostat!
Is there a way to do a firmware upgrade on FW200?
FB 10 has an even older software version, FD: 786

norberts1, 2021-09-10:
@simasjoris I have never seen any firmware-software for updating Fxyz-moduls. Perhaps it's very difficult to get and install it. Often there are moduls available on ebay etc. Years ago I soled my workable FB10, FB100 and FW100 and replaced them with one used CW400.
One information: You can take out your FW200 from the heater and replace the FB10 with it. Then the FW200 is located in your room, measuring the room-temperatur and controlling that HC1 and the rest of the system as before.
But you need then only to replace the FW200.