emsesp/EMS-ESP32 #106: Include Thermostat RC25 (0x17)

planckkonstante, 2021-09-06:


proddy, 2021-09-06:
do you know its product ID and brand?

planckkonstante, 2021-09-06:
The following is on the backside 

Buderus Rc25 EMS
Serie 00
Ver. V4. 01
2530-007-001230-8718577072

[image]

MichaelDvP, 2021-09-06:
@planckkonstante You should have a emsesp-device like:
`(99) Unknown: unknown (DeviceID:0x17, ProductID:xxx, Version:04.01)`
and we need the reported productID. 
Also it is very helpfull if you can log some messages from the RC25. This thermostat seems to be a RC20 with additional option for outdoor-temperature control. It's possible that it uses other telegram-types for controlling than RC20. 
Please set log-all, wait a bit more that a minute, download the log and post it here. We are interested in messages from 0x17 to all like: `?(0x17) -> All(0x00), ?(0xAE), data: 80 12 2E 00 D0 00 00 64`
If you give some information about your system (room or outdoorcontrolled, actual reported outside temperature, etc.) it is possible to identify some values.

planckkonstante, 2021-09-06:
Well, thanks a lot already:

At  the end if the log I played a little around on the Buttons of Rc25, disconnected and reconnected it  to the  bus 
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/7117681/log.1.txt)

Product ID is 151

{
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:13.06)",
      "handlers": ""
    },
    {
      "type": "Unknown",
      "name": "unknown (DeviceID:0x17, ProductID:151, Version:04.01)",
      "handlers": ""
    }
  ]
0x17 is the only temperature sensor connected to the boiler..

MichaelDvP, 2021-09-07:
Thanks, i've added RC25 to device-library. Please try the dev.

RC25 is compatible to the newer RC20 thermostats,and uses 0xAE and 0xAD telegrams, but the configuration telegram 0xAD is 9 bytes longer than for RC20 (23 instead of 14 bytes). 
`Unknown(0x17) -> All(0x00), ?(0xAD), data: 01 27 2D 00 44 05 01 FF 28 19 0A 07 00 00 F6 12 5A 11 00 28 05 05 00`
I guess pos14 `F6`= -10°C is the minimum outdoor temperature. pos 17  `!!` = 17°C could be the summer/winter switchtemperature. But also for RC20 a lot of stuff in this telegram are unknown.

planckkonstante, 2021-09-11:
Well... I am on a windows machine and im not sure if I can compile the dev brach from source.

Is there I Binary?

MichaelDvP, 2021-09-11:
The dev-bin is [here](https://github.com/emsesp/EMS-ESP32/releases). Please check the new (guessed) parameters and values and give feedback if all fits.

For compiling check if node.js is proper installed, check `node -v`and `npm -v` from commandline. The file `wwwdata.h` is generated from script by calling `npm install` and  `npm run build` in the interface path.

planckkonstante, 2021-09-15:
I am still struggling to compile from source...
However, I flashed the dev-binary a few days ago. Device is now listed, parameters are plausible. Everything fine!!

proddy, 2021-09-15:
Great!

planckkonstante, 2021-09-16:
Great work thanks!

proddy, 2021-12-28:
emcek reported on Discord

"I'm using the gateway with Buderus RC25 and by default the modes available in EMS-ESP are "auto/manual/off" and in reality it's "auto/day/night". Can I adjust that myself somehow or it would need a source code change?"

proddy, 2021-12-28:
I've added it to v3.4, needs testing though

closed 2022-01-19 by proddy