emsesp/EMS-ESP32 #107: Manually Controlling Solar Circuit

TheTrueRandom, 2021-09-07:
Hi there,

At my home I have a bosch/junkers device equipped with a solar pump (SM200 I suppose).

I do have a switch built in to the solar return pipes to a heat exchanger directly connected to a pool. However the system does not know about this switch. (There is some option to buy from bosch (ms 200) which is supposed to include this switch to the control system but I don't want to buy this expensive device for something I could easily do in software myself).

The issue is that the solar pump modulation is directly based on the temperature of the water tank. If for example the water tank is 55°C and the solar panel is 60°C, the pump would be switched off (too little gain). However in such a case I would like to override the pump and switch the valve to get the temperature to my pool. Also if the water tank reaches 60°C the pump would always be off (which I also cannot figure out how to change :thinking:). 

When researching I found this project which seems to offer quite some control regarding solar ([Example](https://github.com/emsesp/EMS-ESP32/blob/7eb1f061b7d325149857e4d8879ea087e0984351/src/devices/solar.h#L65)). 

Right now I'm reading values (and also changing some) with the HTTP-API exposed by the bosch/junkers device itself. ([Example endpoints](https://www.smarthomeng.de/developer/plugins/buderus/URLs.html)) Also some solar http endpoints exist already which can be queried. Example: 
- `/solarCircuits/sc1/pumpModulation` `{"id":"/solarCircuits/sc1/pumpModulation","type":"floatValue","writeable":0,"recordable":0,"value":0.0,"unitOfMeasure":"%","state":[{"off":0.0}]}`
- `/solarCircuits/sc1/dhwTankTemperature` `{"id":"/solarCircuits/sc1/dhwTankTemperature","type":"floatValue","writeable":0,"recordable":0,"value":57.1,"unitOfMeasure":"C","state":[{"open":-3276.8},{"short":3276.7}]}`

Unfortunately those values are not adjustable. 

What I tried so far and works quite good is to physically detach the temperature sensor from the water tank, so the system thinks the temperature is falling and enables the solar pump. So it would work to (physically) emulate the temperature sensor and inject the temperature of the pool when switching the pipe valve. 

I wonder if someone has an idea to override the pump modulation or dhwTankTemperature by software? Maybe even by HTTP, which I think should work as the proprietary device (ms 200) also can do it and is not physically connected to the solar device. I have no clue about EMS protocol used in this project.

Thank you

MichaelDvP, 2021-09-08:
Is it Bosch or Junkers? Junkers call their modules ISM1, ISM2, Bosch MS100, MS200 and Buderus SM10, SM100,, SM200. Function is mainly the same, but Bosch protects its brands by maken them incompatible. Because of this also the state of reverse engineering is different. We can set the maximum tank temperature for all, but the commands for turn on/off differences for Junkers are unknown.

Can you make a little scheme how it is connected, is the extra 3-way-valve manual or electric? I guess it is connected like that:
[image]

The maximum tank temperature is normally 60°C to avoid skin burning, if you raise it (max. setting is 90) you should install a thermostatic mixing valve (in german "Brauchwassermischer" for googeling). 

Controlling the pump in solar modules is not possible, it is only controller by internal logic, but we can set the paramters for this logic (like tank max. temp). Without knowing exactly your system i suggest: connect the 3-way-valve to a wifi plug, read the tank temperature and if DHW is warm enough (>50/55°C) open the valve. You can also add a temperature sensor to read the pool temperature and open the valve only it if solar reflow is higher than pool. This logic you have to do in your home automation system, emsesp is not a plc, you can't add own logic.

TheTrueRandom, 2021-09-08:
Hi @MichaelDvP thank you. 
The label on the solar pump reads "AGS10 MS100-2" so I assume it's Bosch? The central heating device is "Condens 9000i W". The installation was done end of last year, so relatively new. 
Your scheme looks right, just the 3-way-valve is connected to the solar return pipe. 
There is also a thermostatic mixing valve installed. 

Do you know how the tank temperature can be increased? When reading all values from the HTTP-Api, I can only find one value at 60°: /dhwCircuits/dhw1/singleChargeSetpoint. I can even increase it but solar will still stop when reaching 60°

> connect the 3-way-valve to a wifi plug, read the tank temperature and if DHW is warm enough (>50/55°C) open the valve. You can also add a temperature sensor to read the pool temperature and open the [...]

That's exactly what I have done already and works perfectly. Just the issue that that the solar pump will be off when the water tank reaches 60°, so switching the valve will not enable the solar pump. In this case I could override the tank temperature to the temperature of the pool. Do you know how to do that?

MichaelDvP, 2021-09-08:
I'm not familiar with the Bosch-API and can not comment it. 
But the buffer temperature can be changed in the settings of your control-panel/thermostat. The [MS100-manual lists it](https://www.manualslib.de/manual/468639/Bosch-Ms-100.html?page=14#manual). If MS100 is compatible with SM100 (buderus), than it also should work with emsesp.

PhillyGilly, 2021-09-14:
1.  I run my Solar to 65C at the bottom of the cylinder which heats the top up to above 60C.
2. Here are the reported values of on/off delta (times 10 but never mind) in my HA
 [image]
3. In my MS200 you can set the delta down to 3C
[image]
4. In line with @bbqkees I don't use my Raspberry PI and ESP32 combo to run anything (apart from table lamps) directly. Instead I monitor in HA and keep the primary config and control in the local equipment. With my SM200 there is an option P see below, but this recommends interfacing the Solar system to a separate pool control system.
[image]

Bartekn86, 2021-09-16:
hi, im not sure that im writing in right place, 
i get
2021-09-16 15:46:26.392   ERROR 671:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 62
2021-09-16 15:55:18.997   ERROR 672:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 58
2021-09-16 15:58:03.506   ERROR 673:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B 88 19 03 20
2021-09-16 15:58:40.901   ERROR 674:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 63
2021-09-16 16:10:30.787   ERROR 675:    [telegram]   Last Tx Read operation failed after 3 retries. Ignoring request: 0B B0 FF 03 20 02 62

i have replaced esp8266 to esp32, and added a sm200 module,  im not sure what device generate this frames,  could be sth wrong with solar module ?

bbqkees, 2021-10-05:
Probably not the best solution but if you want to override the solar pump, you could just add a relay on the pump in parallel with the MS100 so you can turn it on via your home automation independent of what the MS100 is doing.

Bartekn86, 2021-10-05:
you have wrong settings 
 pump turn on difference should be 10 degree
 pump turn off diference should be 5 degere
 
 remember that is a differentil temperature, not the real values