emsesp/EMS-ESP32 #108: Summer mode on heat Circuit

sfbia2m, 2021-09-11:
Hi 
I have a Bosch Compress7000 AW heat pump with 3 heat circuit, and non of them is showing the right state of Set summer mode or summer mode. If I try to change the value from the webinterface nothing is happing. I have checked the log and I can see something is happening.

000+00:45:27.194    INFO 5539:   [command]    Calling thermostat command 'hc3/summersetmode', value winter, id is default
000+00:45:27.194    INFO 5540:   [thermostat] Setting summer mode to winter for heating circuit 3
000+00:45:27.276   DEBUG 5541:   [telegram]   Sending write Tx [#170], telegram: 8B 10 FF 07 01 B1 02
000+00:45:27.296   TRACE 5542:   [telegram]   Rx: 8B 10 FF 07 01 B1 02 F9
000+00:45:27.300   DEBUG 5543:   [emsesp]     Last Tx write successful
000+00:45:27.300   DEBUG 5544:   [telegram]   Sending post validate read, type ID 0x2B1 to dest 0x10
000+00:45:27.600   DEBUG 5545:   [telegram]   Sending read Tx [#171], telegram: 8B 90 FF 00 20 01 B1
000+00:45:27.621   TRACE 5546:   [telegram]   Rx: 8B 90 FF 00 20 01 B1 06
000+00:45:27.661   DEBUG 5547:   [emsesp]     Last Tx read successful
000+00:45:27.661   TRACE 5548:   [telegram]   Rx: 90 0B FF 00 01 B1 00 28 00 00 3C 24 00 00 1E 1E 00 E4

[image]

MichaelDvP, 2021-09-11:
Looks like sumermode and summertemp is set to zero and is not writable. Do you have a outdoor sensor attached? You do not have a summermode if in roomtemperature-controlled state.

sfbia2m, 2021-09-11:
I can control the summer mode on the heat pump. I can change between, "summer, vinter and auto" and I also uses this to turn off the circuit.

MichaelDvP, 2021-09-12:
Ok, if the heatpump manages the summermode, this is a unknown setting, and the thermostat does not responde.
Please post your system-settings and system info. Use `log all` to catch the telegrams and change (while logging) the summermode and summertemperature on your heatpump. Post the log and describe what exactly you have set, than we try to identify the telegram and setting and can add it to emsesp.

sfbia2m, 2021-09-12:
Ok, here you have the requested 
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7149376/log.txt)
[image]
[image]
files.
Before doing anything Heat circuit 3 was in sommermode, then I changed it to automatic and then to winter and the back to sommer. I hope you can find it

MichaelDvP, 2021-09-12:
>  I hope you can find it :-)

Yes, it's from thermostat these telegrams:
```
000+20:41:09.851 T 397593: [telegram] Rx: 90 00 FF 00 03 73 01 F3
000+20:41:17.094 T 397675: [telegram] Rx: 90 00 FF 00 03 73 02 F0
000+20:41:24.370 T 397706: [telegram] Rx: 90 00 FF 00 03 73 00 F2
```
If `03 73` is for hc3 that the other hc should have telegrams hc1`03 71`, hc2:`03 72` ff., but better you repeat this logging for hc1 or hc2. 
Please also change the summertemp while logging (could be the same log or an extra log).
Also helpful would be if you can go to terminal and type `read 10 473` and post the reply. I think this telegram is longer than one byte.

For settings and info i've meant the two nice buttons if you go to `Dashboard/EMSESP-HELP`. I want to know what devices are connected. Especially the boiler and thermostat-type with product-id and version.

sfbia2m, 2021-09-12:
you are fast :-)  And I think you are right about circuit 1 & 3 I tried to change the summer temp to 19 from 17 and back to 19 

Here you have the result of the read command:
-esp:# read 10[Kems-esp:# read 10 [Kems-esp:# read 10 4[Kems-esp:# read 10 42[Kems-esp:# read 10 423
[Kems-esp:# [Kems-esp:# [K000+23:24:08.093 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x423), data: <empty>

[log.txt](https://github.com/emsesp/EMS-ESP32/files/7149561/log.txt)

[emsesp_system_info (2).txt](https://github.com/emsesp/EMS-ESP32/files/7149564/emsesp_system_info.2.txt)

MichaelDvP, 2021-09-12:
Ok, the summertemp is in the same telegram in offset 1 `Rx: 90 00 FF 01 03 73 11 EB`, i've expected that this telegram is longer. 

The read command should be `read 10 473` not 423.  But for now we know that this telegram is only published on change and contains mode in offset 0 and temp in offset 1.  But to read the full telegram is helpfull.

You have thermostat product-ID 158, that's sad because others have the same product-ID but summermode working, we have to check for the extra telegram and take the right value. Hopefully the 0x471 is empty if 0x2AF have valid summermode.

@proddy I think best way is to set the new telegram to fetch and check the `is_fetch`-status. I'll make a pr, please check if i miised something and my thinking is wrong.

@sfbia2m There are some unknown telegrams from heatpump, not changing while logging: 
```
Rx: 88 00 FF 00 03 9F 00 00 0A 8C 08 34 00 00 07 08 00 00 07 08 08
Rx: 88 00 FF 00 03 A0 0E 10 01 50 BA
Rx: 88 00 FF 00 03 A3 00 02 00 00 00 00 00 00 00 00 00 13 00 00 00 00 00 00 00 00 08 00 00 00 FB
```
The `01 50` in second telegram looks like a temperature 33.6°C, do you miss values in emsesp you have in shown in the control panel?

sfbia2m, 2021-09-12:
Ok, I have looked in the control to find a temperature around that and now the telegram returns 33.5 "Rx: 88 00 FF 00 03 A0 0E 10 01 4F A5" but I can't find any value with that in the control. I will try to follow this telegrams to see if I can find anything to pare it with! 
I have done a new read and here is the result:
[Kems-esp:# read 10 473
[Kems-esp:# [Kems-esp:# [K001+04:01:24.421 N 1: [emsesp] Thermostat(0x10) -> Me(0x0B), ?(0x473), data: 00 11 04 01 01 1C 08 04

MichaelDvP, 2021-09-13:
Thanks for the complete telegram, it's a bit longer, but i have no idea what the bytes 2-7 represents. We stay with summermode/summertemp in bytes 0, 1.
You can check the addition with bin file [here](https://github.com/MichaelDvP/EMS-ESP32/releases)

It would be fine it someone with thermostat productID:158 and working summermode can also check that this change does not affect him. 
@tp1de, would you test this? You know the parameters and have a good look if something get's wrong.

sfbia2m, 2021-09-13:
Super Thanks a lot! It is working perfectly :-)  I will try to follow some of the other telegrams you mentioned to see if it gives any sens compared to the control.

tp1de, 2021-09-15:
> @tp1de, would you test this? You know the parameters and have a good look if something get's wrong.

@MichaelDvP  
Michael, I installed the a.m. firmware and do not see any changes in repect to summemode and summertemp / summersettemp. They are working as before.

Anything special to test?

MichaelDvP, 2021-09-15:
@tp1de Thank you for testing. If 0x471, ff are empty the code works as before with telegrams 0x2AF, ff. 
I expected this, but was not sure. (The RC310 v13.06 from @sfbia2m have both telegrams, but 0x2AF have zeros for summermode/summertemp and 0x471 have the data).

The 0x473 to km200 is strange, i think you do not have hc3? Is it the same for 0x471, 0x472?  Emsesp will read only one time for the existing hc (you have only hc1 and hc2 right?), and after emty answer not again.
 
@proddy Please merge.

tp1de, 2021-09-17:
> The 0x473 to km200 is strange, i think you do not have hc3? Is it the same for 0x471, 0x472?

0x471 and 0x472 are the same empty telegrams.... and yes   I have only hc1 and hc2

proddy, 2021-11-10:
this should be fixed now.

closed 2021-11-10 by proddy