emsesp/EMS-ESP32 #110: Thermostat hc1 with Homeassistant: changing values (Operation, Target Temperature) doesn't work

NFis, 2021-09-14:
I just changed to EMS32 (thanks!), Sw v3.2.1, using Homeassistant (with mqq discovery) as a frontend to change values in Thermostat hc1.
My Config: 
(1) Boiler: Topline/GB162 (DeviceID:0x08, ProductID:115, Version:05.05)
(2) Controller: BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)
(3) Thermostat: RC35 (DeviceID:0x10, ProductID:86, Version:01.16) ** master device **
(4) Thermostat: RC20RF (DeviceID:0x18, ProductID:93, Version:02.00)
Homassistant 2021.9.4
Before changing to v3 I deleted all mqtt topics in my broker and deleted all EMS-devices in HA.

Thermostat hc1 is autodected bei HA and added as climate.thermostat_hc1. I can see the correct values for actual temp, mode, target temp. But when I change i. e. the mode or the target temperature in HA it will not work. 
In mqtt Explorer I see a topic ems-esp/thermostat_hc1 with payload corresponding to the selction in HA (auto, heat, off, temperature, one of those).

Changing mode, the log shows: 
WARNING:     [thermostat] Set mode: Heating Circuit 1 not found or activated
ERROR:          [mqtt]          MQTT error: invalid payload heat for this topic thermostat_hc1

Changing target temperature, the log shows: 
WARNING:     [thermostat] Set temperature: Heating Circuit 1 not found or activated for device ID 0x18

It looks like the values cannot be adressed to my devices. How can I fix that? In v2 everything worked fine.

proddy, 2021-09-14:
horray! Welcome to V3.

At first glance, it looks like when setting the temperature it's sending the command to the RC20RF (ID 0x18) and not the RC35 thermostat. Which is odd since the RC35 is defined as the "Master Device" in EMS-ESP. Let's fix that first and see what else is not working.

NFis, 2021-09-14:
ok, looking at RC20RF is a good hint. I made some tests: 
I disconnected the RC20RF from the bus, restarted EMS-ESP gateway (RC20RF is not in the list of EMS devices), and now Thermostat hc1 is working! But current temperature remains at the last value received from RC20RF. 
Then I reconnected the RC20RF to the bus (EMS-ESP detects it later on), Thermostat hc1 is still working, current temperatur is valid and comes from the RC20RF.

proddy, 2021-09-18:
I've added the lines. @NFis can you test with the latest dev build? https://github.com/emsesp/EMS-ESP32/releases/tag/latest

NFis, 2021-09-18:
Thanks, I'll be happy to try that out, but won't get to it until Monday.

NFis, 2021-09-19:
I have just tried version v3.2.2b5 / 28bf7c3 (after deleting all mqtt topics in my broker and all ems devices in homeassistant) and it works perfectly. Thank you very much!

proddy, 2021-09-19:
great, all thanks to Michael. Can I close this issue?

NFis, 2021-09-20:
Yes, the issue can be closed.

closed 2021-09-20 by proddy